verilator/verilator #5068: Verilator finds a latch where there is no one

algrobman, 2024-04-25:
getting LATCH waring on following code:

```
     13 module DW02_mult (A, B, TC, PRODUCT);
     14 parameter A_width = 8;
     15 parameter B_width = 8;
     16 // translate_off
     17 initial begin
     18   if (A_width<1) $display ("ERROR - DW02_mult.  Bad parameter, A_width=%d, which is <1.", A_width);
     19   if (B_width<1) $display ("ERROR - DW02_mult.  Bad parameter, B_width=%d, which is <1.", B_width);
     20   if (A_width<1) $finish;
     21   if (B_width<1) $finish;
     22 end
     23 // translate_on
     24 input [A_width-1:0] A;
     25 input [B_width-1:0] B;
     26 input TC;
     27 output [A_width+B_width-1:0] PRODUCT;
     28 reg [A_width+B_width-1:0] PRODUCT, PRODUCT2;
     29 reg [A_width-1:0] A2;
     30 reg [B_width-1:0] B2;
     31 always @(A or B or TC) begin
     32   if (TC == 1'b0)
     33     PRODUCT = A * B;
     34   else begin
     35     A2 = A[A_width-1] ? -A : A;
     36     B2 = B[B_width-1] ? -B : B;
     37     PRODUCT2 = A2 * B2;
     38     PRODUCT = (A[A_width-1]^B[B_width-1]) ? -PRODUCT2 : PRODUCT2;
     39   end
     40 end
     41 endmodule
```
DW02_mult.v:31:1: Latch inferred for signal 'A2' (not all control paths of combinational always assign a value)
                                                                                                                          : ... note: In instance 'tb_top.rvtop.swerv.exu.mul_e1.i_DW02_multp.U1'
                                                                                                                          : ... Suggest use of always_latch for intentional latches
   31 | always @(A or B or TC) begin
      | ^~~~~~

A2 is a function of A, which is in the sensitivity list.

Strangely, this warning pops up only with -O0 verilator option ...

closed 2024-04-25 by wsnyder

algrobman, 2024-04-25:
One more comment  - verilator version is 5.024.

Another interesting thing is what our design produces wrong arithmetic results when these modules are used, vs logically equivalent code.
-O0 does not change the failures.

wsnyder, 2024-04-25:
Well, logical incorrectness is obviously a different more serious issue, if that turns out that is a Verilator bug please open a different ticket, thanks.

algrobman, 2024-04-26:
Wilson, thanks for fast response, you/tool are right about the latch inference, I missed this 'if !TS' statement .