verilator/verilator #5070: `$readmemb` and `+verilator+rand+reset` incompatible

sifferman, 2024-04-26:
Hello.

I discovered that `+verilator+rand+reset` will change the `x` values in `$readmemb` files to hex digits instead of binary digits. I would be happy to look into this in the next couple weeks.

```systemverilog
// test.sv
module test;
    logic m [1];
    initial begin
       $readmemb("x.memb", m);
       $finish;
    end
endmodule
```
```systemverilog
// x.memb
x
```
```bash
$ verilator test.sv --binary --x-initial unique

$ ./obj_dir/Vtest +verilator+rand+reset+0

$ ./obj_dir/Vtest +verilator+rand+reset+1
%Error: x.memb:0: $readmemb (binary) file contains hex characters

$ ./obj_dir/Vtest +verilator+rand+reset+2
%Error: x.memb:0: $readmemb (binary) file contains hex characters
```

wsnyder, 2024-04-26:
Probably this line in verilated.cpp instead to vary on which base is used.  If you'd like to make a pull it would be appreciated.

                const int value
                    = (c >= 'a' ? (c == 'x' ? VL_RAND_RESET_I(4) : (c - 'a' + 10)) : (c - '0'));

wsnyder, 2024-05-21:
Appreciate your work on this!

closed 2024-05-21 by wsnyder