verilator/verilator #5071: Dynamically set the stack size on Linux

poemonsense, 2024-04-27:
For large designs, Verilated models may consume more stack space than the default stack size the OS provides. On Ubuntu 22.04, the default stack size we have (`ulimit -s`) is 8192, which could be easily exceeded with a fair size CPU design. If the verilated model requires a larger stack, the system will have a segmentation fault immediately. We have seen multiple Verilator users to report this as an issue.

While #4799 introduces the estimated stack size and run-time warning, the user must manually set the stack size after the segmentation fault.

To simplify the issue, we need a dynamic stack size expander for different Verilated models with `setrlimit`.

1) According to https://stackoverflow.com/questions/2279052/increase-stack-size-in-linux-with-setrlimit, it works.

2) According to ChatGPT, the reason why it works is that the limitation takes effect only at allocation (instead of the time when the process is created).

> In C/C++, when you use the setrlimit API to set the stack size of a process, you are typically doing so before the process starts. The setrlimit function is usually called at the beginning of your program, typically in the initialization phase, to establish resource limits for the process.
>
> However, it's important to note that setting resource limits with setrlimit affects the current process and its child processes. So, if you set the stack size limit in the parent process before creating any child processes, the child processes will inherit the stack size limit set by the parent.
>
> Once the stack size limit is set, it remains in effect for the duration of the process unless it is explicitly changed again with another call to setrlimit. So, it doesn't update the stack size of the process immediately while it's running; instead, it sets the limit that will be enforced for subsequent stack allocations.
> 
> In summary, setrlimit establishes resource limits before the process starts and affects subsequent stack allocations within the process and its child processes.

3) According to this example I attached, it works. You may try enabling/disabling `setrlimit` via the `SETRLIMIT` macro.

```C++
#include <sys/resource.h>
#include <stdio.h>

#define SIZE_MB  15
#define SIZE     SIZE_MB * 1024 * 1024U

#define STACK_MB 16
#define STACK    STACK_MB * 1024 * 1024U

#define SETRLIMIT

void huge_stack() {
  char c[SIZE];
  for (int i = 0; i < SIZE; i++) {
    c[i] = 1;
  }
}

int main() {
  struct rlimit rlim;
  getrlimit(RLIMIT_STACK, &rlim);
  printf("current stack size = %luMB\n", rlim.rlim_cur / 1024 / 1024);

#ifdef SETRLIMIT
  rlim.rlim_cur = STACK;
  if (!setrlimit(RLIMIT_STACK, &rlim)) {
    printf("set stack size = %dMB\n", STACK_MB);
  }
#endif

  printf("huge_stack: asking for %dMB\n", SIZE_MB);
  huge_stack();
  printf("success\n");
}
```

So, to summarize, we can use `setrlimit` to increase the stack size at runtime dynamically.

If you think this is too aggresive, we can provide a C++ variable for the estimated stack size (instead of the hardcoded value in Syms.cpp).

BTW, we observe that the estimated stack model size will be bigger than that even for a fairly small CPU design. So, how much stack size we request from the OS may be a serious issue. Should it be 1.5x or 2x as #4799 said? I think it would be too much for large designs. I don't know whether it will bring side effects if we request too much.

I'll be happy to make a PR if you agree with this proposal. Thanks

wsnyder, 2024-04-30:
It looks like on recent linux this only changes the error limit, not program layout, so that's why this works.

Can you see if the localize change (#5072) fixes your issue?

Presuming not, I'm open to a pull request to do this.  I'd suggest make a Verilated function to set the size, and try to resize right before the current function which checks the model's size.  So main() likely doesn't need to do anything.

poemonsense, 2024-05-01:
> Can you see if the localize change (#5072) fixes your issue?
> 

I'm sorry but I'm currently off the office because of the public holidays here in China. I will update the test results in 3 or 4 days. 

> 
> Presuming not, I'm open to a pull request to do this.  I'd suggest make a Verilated function to set the size, and try to resize right before the current function which checks the model's size.  So main() likely doesn't need to do anything.

It's great. That's the best option for us Linux users. Thanks for your work

poemonsense, 2024-05-06:
> Can you see if the localize change (#5072) fixes your issue?

I test the option on the XiangShan design generated by Chisel. Prior to the `--localize-max-size` commit, the stack checker asks for 43972753. On that commit, it asks for 43972753 (exactly the same). On that commit with `--localize-max-size 1`, it asks for 43942989 (slightly smaller but still huge).

Thus, it's true that it does not ultimately fix the huge stack issue. We still need a dynamic resizer for the stack on different verilated designs, though I believe the stack checker is asking for too much stack size.

closed 2024-05-09 by wsnyder