verilator/verilator #5072: Avoid stack overflow in models by not  localising large variables

gezalore, 2024-04-27:
I see there are some conversations going on about stack overflow with large models (#5071, #4799), so I just want to record some information here.

In my experience, stack overflows happens when V3Localize converts some huge array to a function local:

```verilog
initial begin
  bit [7:0] memoryImage [LARGE];  // A statically allocated variable
  $readmem("test.bin", memoryImage);
  // ... copy memoryImage into the right SV memories ...
end
```

V3Localize can figure it out that `memoryImage` is only used in one CFunc, and makes it local to that CFunc, but it's huge so we segfault in the prologue setting up the stack frame.

While #5071 is very useful on Linux, it is not a portable solution. e.g. I think MacOS has a ~64M hard limit.

One alternative we could consider is to teach V3Localize to not localize huge variables, hence limiting the stack usage. The emitted C++ otherwise has a pretty shallow call stack, so this is likely to fix the most common problems. This of course doesn't fix the case when the user typed in large automatic variables, but that is the same case as the example code in #5071, which is more of a "user error" than a tool problem. 

I think this is unlikely to cause much performance degradation (in fact locals are always zeroed out at the moment - they really shouldn't have to be - so might even help a little).

Personally I don't have a hard feeling about which way to go, so feel free to discuss or close as desired.

wsnyder, 2024-04-27:
I think this is a good point, I'll prepare a change to limit this.  It's easier to hard code it, but suspect someone will need a tweak so will add an option.

I'm thinking 1KB.  We don't want wide temporaries to not get localized; an 8192+ bit wire is rare so seems reasonable.

>in fact locals are always zeroed out at the moment

Note that's true of only users' signals, not temporaries created by Verilator.

poemonsense, 2024-04-27:
> In my experience, stack overflows happens when V3Localize converts some huge array to a function local:
>
> ```verilog
> initial begin
>   bit [7:0] memoryImage [LARGE];  // A statically allocated [...]

Is this the only reason? I think we don't have this local huge array in our design (XiangShan). We have large arrays defined in the C++ side instead of the RTL. In the RTL, we have only like ~MB SRAM (single-port, dual-port) primitives. However, our gate count is huge, though I assume they should be `new`'ed on the heap instead of the stack.

wsnyder, 2024-04-27:
No, definitely not the only reason, but this would help some designs remove some pressure, for little cost.

closed 2024-04-30 by wsnyder