verilator/verilator #5074: $random sequences with the same seed do not equal

punzik, 2024-04-27:
Generating random numbers using $random and $urandom with the same seed produces different sequences. At the same time, $dist_* functions with the same seed produces identical numbers.

Test code:
```verilog
`timescale 1ps/1ps

module testbench;
  initial begin
    int seed;
  
    $display("\nFirst sequence");
    seed = 0;
    repeat(10) $display("%0d", $random(seed));
  
    $display("\nSecond sequence");
    seed = 0;
    repeat(10) $display("%0d", $random(seed));

    $finish();
  end
endmodule
```

Output:
```
First sequence
3881442
-2253583
-2144416136
1075275580
-540298338
-270149169
2012409063
1006204531
-499336391
186539420

Second sequence
32234930
2134240985
-1076168852
1607631945
803815972
1748279570
-874884983
-1708316604
-1294560222
1512786449
```

NixOS 23.11

Verilator 5.025 devel rev v5.024-9-g1315aa31e (from oss-cad-suite-linux-x64-20240414 distribution)

punzik, 2024-04-27:
This behavior is observed only when seed is zero. With other seed values sim works correctly.

wsnyder, 2024-04-27:
Verilator follows C's seeding conventions where seed 0 means "pick a new seed".  For compatibility reasons we are unlikely to change this.  Don't use zero.

closed 2024-04-27 by wsnyder