verilator/verilator #5079: Unnecessary `WIDTHEXPAND` warning/error when the LHS is wider in a binary operation

goekce, 2024-04-28:
### Example

```sv
module m (
    input  [1:0] i1,
    i2,
    output [3:0] o
);
  assign o = i1 + i2;
endmodule
```

Compile using:

```sh
verilator --binary m.sv
```

### Current Output

```
%Warning-WIDTHEXPAND: calculator_simple.sv:9:56: Operator ADD expects 8 bits on the LHS, but LHS's VARREF 'i1'
...
%Warning-WIDTHEXPAND: calculator_simple.sv:9:56: Operator ADD expects 8 bits on the RHS, but RHS's VARREF 'i2'
...
%Error: Exiting due to 2 warning(s)
```

### Expected output

No warning/errors. I tried the example above on Edaplayground using all four available commercial simulators and they don't issue any warning/error.

### Workaround

```
/* verilator lint_off WIDTHEXPAND */
...
/* verilator lint_on WIDTHEXPAND */
```
### Version

verilator 5.024 on Archlinux

wsnyder, 2024-04-28:
Verilator is completely correct, there is zero extension, your output signal is one bit too wide, or input one bit too narrow.  Use a concat to make your intent clear.

`assign o = i1 + {3'b0, i2};`

None of the edaplayground options do linting, so that's not relevant.

closed 2024-04-28 by wsnyder

goekce, 2024-04-28:
@wsnyder I believe you misunderstood the example. You probably mean that the output is not one but *two* bits too narrow, right?

> None of the edaplayground options do linting, so that's not relevant.

Indeed I could not activate linting on VCS and Questa

Automatic zero extension is a feature of the standard and it should not be a warning or error:

From LRM 6.11.1 Integral types:

> Automatic type conversions from a smaller number of bits to a larger number of bits involve zero extensions if unsigned or sign extensions if signed.

If we have a statement like `logic[3:0] = logic[1:0]`, I would expect the aforementioned automatic type conversion in the LRM. What do you think?

---

FYI: `Current Output` was from another test so it was slightly different. I updated it.

wsnyder, 2024-04-28:
Verilator does the automatic type conversion that IEEE says. The point of the lint warning is this is a common mistake in code, and this message has avoided many many silicon bugs.  That's the point of having a "lint tool".  If you don't like the warning you are free to disable it.

goekce, 2024-04-28:
I understand that automatic type conversion is bad practice. 

1. Would you share an example when this avoided a silicon bug?
2. `assign o = $bits(o)'(i1 + i2);`  does output the same warning. Is this behavior also correct?
3. Should I expand every operand individually as follows?: `assign o = $bits(o)'(i1) + $bits(o)'(i2) + ...;`

wsnyder, 2024-04-28:
1. As one example, we had an old design with a structure of a certain size, say 8 entries using your example with:

    wire [2:0] memory_index = csr[1:0] + 1;

all's fine.  In a redesign we increased the size to 32, so had effectively

    wire [4:0] memory_index = csr[1:0] + x;

when the WIDTHEXTEND showed up we realized the CSR forgot to get widened by a bit.

Generally though WIDTHEXTEND is less bug-indicitive than WIDTHTRUNC.

2. With this cast it warns about i2, which is not the exact same warning; the cast correctly suppresses the assignment's warning.  I think do want your item 3, which does not warn:

    assign o = $bits(o)'(i1) + $bits(o)'(i2);

goekce, 2024-05-01:
> 2.

So your solution would be `assign o = $bits(o)'(i1) + $bits(o)'(i2);` right?

> 1. ...

This problem could also have been noticed in verification, couldn't it? If the linter can notice it, it is of course better, however.

The reason why I am pedantic about this kind of warnings is that HDL languages tend to be very unflexible. I teach FPGA programming to students who already know high-level programming. One of the reasons why I teach SystemVerilog is its flexibility, and one example is the automatic type conversion. As far as I remember VHDL is more pedantic regarding typing. But if I have to write something like this to do a simple addition:

```sv
assign o = $bits(o)'(i1) + $bits(o)'(i2) + $bits(o)'(i3) + $bits(o)'(i4);
```

This makes the language more awful — this should not be the default behavior of the linter IMHO.

I once heard from an experienced hardware engineer that SystemVerilog is only a tool, however this kind of idiosyncrasies make the language harder to learn in the beginning and thus demotivates CS students. I wish to have at some point a *Python* for hardware design. Maybe [Veryl](https://veryl-lang.org/) becomes an intermediate solution.

TL;DR: I hope that this kind of linter warnings (which become errors) are not default in the future to keep the flexibility of SystemVerilog regarding typing. These could be activated in a strict mode, for example.

wsnyder, 2024-05-01:
>So your solution would be `assign o = $bits(o)'(i1) + $bits(o)'(i2);` right?

That's the portable way to avoid the warning, yes, but more likely most designers and myself would just turn off the warning around that line.

>This problem could also have been noticed in verification, couldn't it? If the linter can notice it, it is of course better, however.

Certainly.  As you note though, it's far more expensive in time and potential risk to wait until verification hits it (or misses it).

Also in this specific example resizing CSRs is something a verif engineer may well missis unlikely to think of, as they typically automate CSR-setting in verification, relying on designers to spec correctly.

Note the cost of fixing a mistake in 3nm silicon is > $10 million.  That pays for a lot of engineer time in having rigorous design philosophies including checking lint warnings ;)

>I teach FPGA programming to students who already know high-level programming. One of the reasons why I teach SystemVerilog is its flexibility, and one example is the automatic type conversion. As far as I remember VHDL is more pedantic regarding typing.

Your teaching efforts are appreciated!  BTW Verilator tends to lint assuming fairly experienced users, students often make "creative" mistakes e.g. doing non-synthesizable things, that Verilator might not be good at warning about. As you come across those feel free to file bugs so Verilator can help them out better - though we'd appreciate some help fixing them - maybe you have some grad students?

>But if I have to write something like this to do a simple addition:
> `assign o = $bits(o)'(i1) + $bits(o)'(i2) + $bits(o)'(i3) + $bits(o)'(i4);`

Note an exception Verilator understands and suppresses is using or not using the carry in an addition, so this doesn't warn.

`wire [3:0] = sig[2:0] + 1;`

The less common case is designers are adding different sized things.  The common case is i1/i2/i3/i4 have identical sizes, which is why it flags this.  The warning disable is a way of the designer saying "I acknowledge thinking about the consequence of this thing I'm doing that's slightly unusual".

>however this kind of idiosyncrasies make the language harder to learn in the beginning and thus demotivates CS studen

Fair point.  There's good reasons for these rules, likewise for most reasons different languages do different things.  Verilog does things to make RTL design easier and less bug-intensive.  Every language and tools makes its appropriate choices - e.g. Rust certainly has things that make it harder to learn for those that come from Python, but it's for good security reasons.

All the more so for students, which are likely to come in using languages e.g. C where "int" is sort of the default, and not much thought goes into widths.  Widths matter a lot in HDL coding - e.g. often determine speed and cost (gate count).

>I wish to have at some point a *Python* for hardware design.

There's Chisel, which is close-ish.  If you want your engineers to understand closer to the "real world" of design, and ultimately potentially get HDL/FPGA jobs I'd stick with Verilog.

>TL;DR: I hope that this kind of linter warnings (which become errors) are not default

You're free to use -Wno-lint.  However, I think that's likely a disservice to their learning to code. IMHO I think it's good to teach linting in general  - e.g. require pylint/ruff clean when you get Python assignments from students -- and as part of that show them how to disable the extraneous lint warnings that are in those tools, too.

goekce, 2024-05-02:
> There's Chisel, which is close-ish. If you want your engineers to understand closer to the "real world" of design, and ultimately potentially get HDL/FPGA jobs I'd stick with Verilog.

Oh, I never used Chisel. I should dedicate a class to Chisel in the end and I will forward your suggestion about Verilog — thanks!

> You're free to use -Wno-lint. However, I think that's likely a disservice to their learning to code. ...

Linters are just great for learning due to the fast code-fix loop. This is the reason why I still encourage them to code using Code editor + a Plugin that uses (Verilator + Verible). I just found `WIDTHEXPAND` leading to an error somewhat unsexy for the beginners, so searched for a way to switch only this warning off like you can do in pylint etc. `Wno-lint` would just not make any sense for the students. I will suggest that we use `lint_off`.

> The less common case is designers are adding different sized things. The common case is i1/i2/i3/i4 have identical sizes, which is why it flags this. The warning disable is a way of the designer saying "I acknowledge thinking about the consequence of this thing I'm doing that's slightly unusual".

I understand that adding different sized things is uncommon, so this is the reason why Verilator flags this. However `i1` and `i2` are both two bits in my example. Is there a misunderstanding?

---

I really appreciate your thorough response Wilson, this motivates me to continue contributing to this project 😊