verilator/verilator #5080: A specific pattern is verilated into an infinite loop when optimizations are enabled

ArthurRosaMelexis, 2024-04-29:
I found a specific pattern that triggers a verilation bug.

From my initial testing, having:
- optimizations enabled
- an always block triggered by an inout signal
- with 2 nested conditionals 
- that lead to a task call
- in which a delay is present

makes Verilator not put the `co_await` operator on the `VTrigSched` corresponding to this `always` block, leading to an infinite loop at the first eval.

(I am not too familiar with Verilog or Verilator, so I may do some terminology errors)

An example of this pattern is given below.

```Verilog
// DESCRIPTION: Verilator: Verilog Test module
//
// This file ONLY is placed under the Creative Commons Public Domain, for
// any use, without warranty, 2024.
// SPDX-License-Identifier: CC0-1.0

module test;

  wire inoutSignal;

  sunmod foo(inoutSignal);

  initial begin
`ifdef TEST_VERBOSE
    $display("The top module is starting");
`endif
    #2 $display("*-* All Finished *-*");
    $finish();
  end

endmodule

module submod (
    inoutSignal
);

  inout inoutSignal;

  reg cond1;
  reg cond2;

  initial begin
`ifdef TEST_VERBOSE
    $display("The module's initial is called before the infinite loop");
`endif
    // These conditions could have any value
    cond1 = 1;
    cond2 = 0;
  end

  task testTask;
    begin
      #1;
`ifdef TEST_VERBOSE
        $display("This task block is never actually called");
`endif
    end
  endtask  // pumpErr

  always @(inoutSignal) begin
    // The infinte loop is in the generated timing block for this block
    if (cond1) begin
      if (cond2) begin
        testTask;
      end
    end
  end

endmodule
```

The following command will trigger the bug :

```verilator --binary --timing -DTEST_VERBOSE test.v```

Simply adding the `-O0` Verilator flag will avoid it.

The bug was discovered on Verilator v5.020, and is still present in the git master version.

I am using Pop!_OS 22.04 LTS (based on Ubuntu 22.04 LTS).

I am used to working with C++, but not very familiar with the Verilator code base, so I would need some help to fix this.

wsnyder, 2024-04-29:
Thanks for a clean simple test case.

@kbieganski might you take a look?  If difficult can we short-term get a UNSUPPORTED error?

kbieganski, 2024-05-02:
The issue is that `-fconst` [converts the sensitivity list](https://github.com/verilator/verilator/blob/25ca1af8b3212a6c7088777e36fc5bc596957d22/src/V3Const.cpp#L2853-L2860) of this `always` into an "initial" sentree (effectively, it converts the `always` into an `initial`). Later, any `always` with delays [is being converted to a `while (true)` loop](https://github.com/verilator/verilator/blob/25ca1af8b3212a6c7088777e36fc5bc596957d22/src/V3Sched.cpp#L434-L439). Usually, the `always`' sentree serves as a timing control that breaks the loop. But in this case we don't have a real sentree, so we get an infinite loop.

I think this can be fixed by a tweak to that condition. If it works, I'll open a PR.

PR #5349 (2024-08-08): Fix optimized out sentrees with `--timing`
Fixes #5080.
closed 2024-08-08 by wsnyder