verilator/verilator #5081: Deferred assertions have the same behavior as the immediate assertion

goekce, 2024-04-29:
Deferred assertions defined in LRM 16.4: `assert final` and `assert #0`. Are useful for waiting for signals to settle like `$strobe`.

#### Example

```sv
module tb;
  logic i, o;
  assign o = i;
  initial begin
    i = 0;

    #1 $display("\nAfter #1");
    i = 1;
    $display("%d: o: %d", $time, o);
    $strobe("%dstrobe: o: %d", $time, o);
    assert (o == 1)
    else $display("immediate assert failed");
    assert final (o == 1)
    else $display("final assert failed");
    assert #0 (o == 1)
    else $display("#0 assert failed");

    #0 $display("\nAfter zero delay");
    $display("%d: o: %d", $time, o);
    $strobe("%dstrobe: o: %d", $time, o);
    assert (o == 1)
    else $display("immediate assert failed");
    assert final (o == 1)
    else $display("final assert failed");
    assert #0 (o == 1)
    else $display("#0 assert failed");

    #1 $display("\nAfter #1");
    $display("%d: o: %d", $time, o);
    assert (o == 1);

    $finish;
  end
endmodule
```

#### Command

```sh
verilator \
                -MAKEFLAGS '--quiet' \
                --quiet \
                --assert \
                --binary \
                --timing \
                -o sim \
                m.sv
./obj_dir/sim
```

#### Output

All assertions fail:

```

After #1
                   1: o: 0
immediate assert failed
final assert failed
#0 assert failed

After zero delay
                   1: o: 0
immediate assert failed
final assert failed
#0 assert failed
                   1strobe: o: 1
                   1strobe: o: 1

After #1
                   2: o: 1
```

#### Expected

Only the first immediate assertion should fail. For example Questa output:

```
# 
# After #1
#                    1: o: 0
# immediate assert failed
# 
# After zero delay
#                    1: o: 1
#                    1strobe: o: 1
#                    1strobe: o: 1
# 
# After #1
#                    2: o: 1
```

#### Version

Verilator 5.024 2024-04-05

#### More info

1. I saw [this issue about deferred assertions](https://github.com/verilator/verilator/issues/1449) however I don't know whether they are only implemented in the linter.
2. `$strobe` has a similar function and it seems to work correctly (compared to deferred assertions).
3. According to the docs
    https://github.com/verilator/verilator/blob/72b96d506953fe9f86c5687411a8975b89a7a14a/docs/guide/languages.rst?plain=1#L125

    I believe the doc was not updated.

wsnyder, 2024-04-30:
I don't get an assertion failure on your example, is some different code needed?

wsnyder, 2024-05-01:
ccache certainly should not matter, but tried it, same:

```
After #1
                   1: o: 1

After zero delay
                   1: o: 1
                   1strobe: o: 1
                   1strobe: o: 1

After #1
                   2: o: 1
- tb.v:32: Verilog $finish
```

wsnyder, 2024-05-01:
Make sure to paste in the exact code here, not something from your disk which might be different.

goekce, 2024-05-03:
Thanks for the heads up 👍. I indeed use a makefile and I forgot to include the argument `--trace-fst`. I modified the issue above. Can you please try again?

I see two problems:

1. `--trace-fst` changes the behavior of the simulation.
2. Without `--trace-fst`, the output seems *almost* correct. I would at least expect `immediate assert failed` after the first `#1` like Questa's output. Does Verilator do further optimizations so that the reaction to the combinational signal is computed in the same simulation time slot?

wsnyder, 2024-08-24:
I presume this was resolved by the ordering problem described in #5177, if not please reopen and post an example that fails on master, thanks.

closed 2024-08-24 by wsnyder