verilator/verilator #5083: $typename doesn't return correct strings from LRM

AndrewNolte, 2024-04-29:
The LRM Specifies this in 20.6.1:
```
// source code // $typename would return
typedef bit node; // "bit"
node [2:0] X; // "bit [2:0]"
int signed Y; // "int"
package A;
enum {A,B,C=99} X; // "enum{A=32'sd0,B=32'sd1,C=32'sd99}A::e$1"
typedef bit [9:1'b1] word; // "A::bit[9:1]"
endpackage : A
import A::*;
module top;
typedef struct {node A,B;} AB_t;
AB_t AB[10]; // "struct{bit A;bit B;}top.AB_t$[0:9]"
endmodule
```

Verilator currently gets the Struct, enum and package reference examples wrong.
See test case here https://github.com/verilator/verilator/pull/5082/files

wsnyder, 2024-04-30:
Is this something you think you might make a pull to fix?

wsnyder, 2024-06-01:
Fixed, while not exact, close enough given variations across simulators.

closed 2024-06-01 by wsnyder