verilator/verilator #5086: MacOS x86 compilation failure

neeldugar, 2024-04-30:
Effectively there are a couple of lines of non-portable code for MacOS x86, see [here](https://github.com/verilator/verilator/blame/0da7ecf753a69aa35f991f0873d764b6ebae6a14/include/verilatedos_c.h#L54) and [here](https://github.com/verilator/verilator/blame/0da7ecf753a69aa35f991f0873d764b6ebae6a14/include/verilatedos_c.h#L69) which is preventing the conda-forge/verilator-feedstock automatic distribution, seems to have been introduced in 840970e for v5.024. Strangely enough seems like MacOS arm builds are not impacted by this.

wsnyder, 2024-04-30:
Can you please submit a pull request to fix this with appropriate ifdefs?

neeldugar, 2024-04-30:
Sure, will look at it a little later.

wsnyder, 2024-11-25:
@neeldugar did you find a work around for this - can you submit a patch?

wsnyder, 2024-11-26:
Ok, given that and lack of a patch I'm closing this out, if a Mac x86 user wants to submit a pull later, great.

closed 2024-11-26 by wsnyder