verilator/verilator #5106: compile error for --timing using clang 15.0.0

mxg, 2024-05-08:
Apparently, in clang-15.0.0, the `<coroutine>` header file is part of the regular set of header files and is no longer in the `experimental` subdirectory.  Building a design using --timing, which requires the `<coroutine>` header fails.

I am running on MacOS Sonoma version 14.4.1 using clang-15.0.0

The following code in `verilated_timing.h` always attempts to include `experimental/coroutine` instead of `coroutine` when versions of clang newer than 13.

```
// clang-format off
// Some preprocessor magic to support both Clang and GCC coroutines with both libc++ and libstdc++
#if defined _LIBCPP_VERSION  // libc++
# if __clang_major__ > 13  // Clang > 13 warns that coroutine types in std::experimental are deprecated
#  pragma clang diagnostic push
#  pragma clang diagnostic ignored "-Wdeprecated-experimental-coroutine"
# endif
# include <experimental/coroutine>
  namespace std {
      using namespace experimental;  // Bring std::experimental into the std namespace
  }
#else
# if defined __clang__ && defined __GLIBCXX__ && !defined __cpp_impl_coroutine
#  define __cpp_impl_coroutine 1  // Clang doesn't define this, but it's needed for libstdc++
# endif
# include <coroutine>
# if __clang_major__ < 14
   namespace std {  // Bring coroutine library into std::experimental, as Clang < 14 expects it to be there
       namespace experimental {
           using namespace std;
       }
   }
# endif
#endif
// clang-format on
```

wsnyder, 2024-05-08:
See #5031

mxg, 2024-05-09:
I look forward to PR#5031 being merged to main

wsnyder, 2024-05-10:
Note #5031 still needs changes to check using configure.ac (see the ticket), perhaps you can help that.

closed 2024-05-10 by wsnyder

mxg, 2024-05-11:
> Note #5031 still needs changes to check using configure.ac (see the ticket), perhaps you can help that.

I'm happy to help, but I need to figure out what you expect.  The PR has code in configure.ac to check for `<coroutine>` and changes in `verilated_timing.h`.  I'm not an autoconf expert, but if you could help me with what you're looking for, I can do the heavy lifting.