verilator/verilator #5107: Unable to compile with Verilator 5.008

jordankrim, 2024-05-08:
I am using Verilator 5.008.  I get this when trying to compile:
verilator --cc --exe --trace -CFLAGS " -O3 -std=c++17 -I/mnt/data/projects/project/jkrim/chipname/.conda-env/riscv-tools/include -I/mnt/data/projects/project/jkrim/chipname/tools/DRAMSim2 -I/mnt/data/projects/project/jkrim/chipname/sims/verilator/generated-src/chipyard.harness.TestHarness.CEPVerilatorRocketConfig/gen-collateral   -DVERILATOR -include /mnt/data/projects/project/jkrim/chipname/sims/verilator/generated-src/chipyard.harness.TestHarness.CEPVerilatorRocketConfig/chipyard.harness.TestHarness.CEPVerilatorRocketConfig.plusArgs" -LDFLAGS " -L/mnt/data/projects/project/jkrim/chipname/.conda-env/riscv-tools/lib -Wl,-rpath,/mnt/data/projects/project/jkrim/chipname/.conda-env/riscv-tools/lib -L/mnt/data/projects/project/jkrim/chipname/sims/verilator -L/mnt/data/projects/project/jkrim/chipname/tools/DRAMSim2 -lriscv -lfesvr -ldramsim "   --threads 1 --threads-dpi all -O3 --x-assign fast --x-initial fast --output-split 10000 --output-split-cfuncs 100 --assert -Wno-fatal --timescale 1ns/1ps --max-num-width 1048576 +define+CLOCK_PERIOD=1.0 +define+RESET_DELAY=777.7 +define+PRINTF_COND=TestDriver.printf_cond +define+STOP_COND=!TestDriver.reset +define+MODEL=TestHarness +define+RANDOMIZE_MEM_INIT +define+RANDOMIZE_REG_INIT +define+RANDOMIZE_GARBAGE_ASSIGN +define+RANDOMIZE_INVALID_ASSIGN +define+VERILATOR --top-module TestHarness --vpi -f /mnt/data/projects/project/jkrim/chipname/sims/verilator/generated-src/chipyard.harness.TestHarness.CEPVerilatorRocketConfig/sim_files.common.f  -o /mnt/data/projects/project/jkrim/chipname/sims/verilator/simulator-chipyard.harness-CEPVerilatorRocketConfig -Mdir /mnt/data/projects/project/jkrim/chipname/sims/verilator/generated-src/chipyard.harness.TestHarness.CEPVerilatorRocketConfig/chipyard.harness.TestHarness.CEPVerilatorRocketConfig -CFLAGS "-include /mnt/data/projects/project/jkrim/chipname/sims/verilator/generated-src/chipyard.harness.TestHarness.CEPVerilatorRocketConfig/chipyard.harness.TestHarness.CEPVerilatorRocketConfig/VTestHarness.h"
**%Warning-REDEFMACRO: Redefining existing define: 'VERILATOR', with different value: ''
                     ... Location of previous definition, with value: '1'
                     ... For warning description see https://verilator.org/warn/REDEFMACRO?v=5.008
                     ... Use "/* verilator lint_off REDEFMACRO */" and lint_on around source to disable this message.
%Error: /mnt/data/projects/project/jkrim/chipname/sims/verilator/generated-**src/chipyard.harness.TestHarness.CEPVerilatorRocketConfig/gen-collateral/prim_assert.sv:74:11: Cannot find include file: prim_assert_dummy_macros.svh
   74 |  `include "prim_assert_dummy_macros.svh"
      |           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
%Error: /mnt/data/projects/project/jkrim/chipname/sims/verilator/generated-src/chipyard.harness.TestHarness.CEPVerilatorRocketConfig/gen-collateral/prim_assert.sv:74:11: This may be because there's no search path specified with -I<dir>.
   74 |  `include "prim_assert_dummy_macros.svh"
      |           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
        ... Looked in:
             prim_assert_dummy_macros.svh
             prim_assert_dummy_macros.svh.v
             prim_assert_dummy_macros.svh.sv
             /mnt/data/projects/project/jkrim/chipname/sims/verilator/generated-src/chipyard.harness.TestHarness.CEPVerilatorRocketConfig/chipyard.harness.TestHarness.CEPVerilatorRocketConfig/prim_assert_dummy_macros.svh
             /mnt/data/projects/project/jkrim/chipname/sims/verilator/generated-src/chipyard.harness.TestHarness.CEPVerilatorRocketConfig/chipyard.harness.TestHarness.CEPVerilatorRocketConfig/prim_assert_dummy_macros.svh.v
             /mnt/data/projects/project/jkrim/chipname/sims/verilator/generated-src/chipyard.harness.TestHarness.CEPVerilatorRocketConfig/chipyard.harness.TestHarness.CEPVerilatorRocketConfig/prim_assert_dummy_macros.svh.sv
%Error: /mnt/data/projects/project/jkrim/chipname/sims/verilator/generated-src/chipyard.harness.TestHarness.CEPVerilatorRocketConfig/gen-collateral/llki_pp_wrapper.sv:30:10: Cannot find include file: prim_assert.sv
   30 | `include "prim_assert.sv"
      |          ^~~~~~~~~~~~~~~~
%Error: /mnt/data/projects/project/jkrim/chipname/sims/verilator/generated-src/chipyard.harness.TestHarness.CEPVerilatorRocketConfig/gen-collateral/llki_pp_wrapper.sv:100:3: Define or directive not defined: '`ASSERT_INIT'
                                                                                                                                                                            : ... Suggested alternative: '`ASSERT_IF'
  100 |   `ASSERT_INIT(srot_slaveTlSzw, top_pkg::TL_SZW >= SLAVE_TL_SZW)
      |   ^~~~~~~~~~~~
%Error: /mnt/data/projects/project/jkrim/chipname/sims/verilator/generated-src/chipyard.harness.TestHarness.CEPVerilatorRocketConfig/gen-collateral/llki_pp_wrapper.sv:100:15: syntax error, unexpected '('
  100 |   `ASSERT_INIT(srot_slaveTlSzw, top_pkg::TL_SZW >= SLAVE_TL_SZW)
      |               ^
%Error: /mnt/data/projects/project/jkrim/chipname/sims/verilator/generated-src/chipyard.harness.TestHarness.CEPVerilatorRocketConfig/gen-collateral/llki_pp_wrapper.sv:101:3: Define or directive not defined: '`ASSERT_INIT'
                                                                                                                                                                            : ... Suggested alternative: '`ASSERT_IF'
  101 |   `ASSERT_INIT(srot_slaveTlAiw, top_pkg::TL_AIW >= SLAVE_TL_AIW)
      |   ^~~~~~~~~~~~
%Error: /mnt/data/projects/project/jkrim/chipname/sims/verilator/generated-src/chipyard.harness.TestHarness.CEPVerilatorRocketConfig/gen-collateral/llki_pp_wrapper.sv:102:3: Define or directive not defined: '`ASSERT_INIT'
                                                                                                                                                                            : ... Suggested alternative: '`ASSERT_IF'
  102 |   `ASSERT_INIT(srot_slaveTlAw, top_pkg::TL_AW >= SLAVE_TL_AW)
      |   ^~~~~~~~~~~~
%Error: /mnt/data/projects/project/jkrim/chipname/sims/verilator/generated-src/chipyard.harness.TestHarness.CEPVerilatorRocketConfig/gen-collateral/llki_pp_wrapper.sv:103:3: Define or directive not defined: '`ASSERT_INIT'
                                                                                                                                                                            : ... Suggested alternative: '`ASSERT_IF'
  103 |   `ASSERT_INIT(srot_slaveTlDbw, top_pkg::TL_DBW == SLAVE_TL_DBW)
      |   ^~~~~~~~~~~~
%Error: /mnt/data/projects/project/jkrim/chipname/sims/verilator/generated-src/chipyard.harness.TestHarness.CEPVerilatorRocketConfig/gen-collateral/llki_pp_wrapper.sv:104:3: Define or directive not defined: '`ASSERT_INIT'
                                                                                                                                                                            : ... Suggested alternative: '`ASSERT_IF'
  104 |   `ASSERT_INIT(srot_slaveTlDw, top_pkg::TL_DW == SLAVE_TL_DW)
      |   ^~~~~~~~~~~~
%Error: /mnt/data/projects/project/jkrim/chipname/sims/verilator/generated-src/chipyard.harness.TestHarness.CEPVerilatorRocketConfig/gen-collateral/llki_pp_wrapper.sv:110:24: syntax error, unexpected '[', expecting IDENTIFIER or randomize
  110 |     slave_tl_h2d.a_size[SLAVE_TL_SZW-1:0]       = slave_a_size;
      |                        ^
%Error: /mnt/data/projects/project/jkrim/chipname/sims/verilator/generated-src/chipyard.harness.TestHarness.CEPVerilatorRocketConfig/gen-collateral/llki_pp_wrapper.sv:111:49: syntax error, unexpected '=', expecting IDENTIFIER or randomize
  111 |     slave_tl_h2d.a_source                       = '0;
      |                                                 ^
%Error: /mnt/data/projects/project/jkrim/chipname/sims/verilator/generated-src/chipyard.harness.TestHarness.CEPVerilatorRocketConfig/gen-collateral/llki_pp_wrapper.sv:112:26: syntax error, unexpected '[', expecting IDENTIFIER or randomize
  112 |     slave_tl_h2d.a_source[SLAVE_TL_AIW-1:0]     = slave_a_source;
      |                          ^
%Error: /mnt/data/projects/project/jkrim/chipname/sims/verilator/generated-src/chipyard.harness.TestHarness.CEPVerilatorRocketConfig/gen-collateral/llki_pp_wrapper.sv:113:49: syntax error, unexpected '=', expecting IDENTIFIER or randomize
[... cut ...]
%Error: /mnt/data/projects/project/jkrim/chipname/sims/verilator/generated-src/chipyard.harness.TestHarness.CEPVerilatorRocketConfig/gen-collateral/prim_fifo_sync.sv:160:3: Define or directive not defined: '`ASSERT_KNOWN'
                                                                                                                                                                           : ... Suggested alternative: '`ASSERT_KNOWN_IF'
  160 |   `ASSERT_KNOWN(WreadyKnown_A, wready_o)
      |   ^~~~~~~~~~~~~
%Error: Cannot continue

wsnyder, 2024-05-09:
This error is saying the code going into Verilator is bad - the file is missing. It is not a problem with Verilator.  Ask the people that provided this code how to solve it.

```
%Error: /mnt/data/projects/project/jkrim/chipname/sims/verilator/generated-**src/chipyard.harness.TestHarness.CEPVerilatorRocketConfig/gen-collateral/prim_assert.sv:74:11: Cannot find include file: prim_assert_dummy_macros.svh
   74 |  `include "prim_assert_dummy_macros.svh"
```

closed 2024-05-09 by wsnyder

jordankrim, 2024-05-09:
There is a -I for where that file exists:
generated-src/chipyard.harness.TestHarness.CEPVerilatorRocketConfig/gen-collateral
so why is it not finding it there?

ekallal, 2024-05-10:
what was incorrect about the includes?

wsnyder, 2024-05-10:
I took your message to mean you figured it out.  Did you mean to provide a -I, are the files readable in the path Verilator says?  You'll need to debug it.  If this is for a class project, ask a TA or professor that knows the project for assistance, if for a company ask a coworker likewise.