verilator/verilator #5109: Compilation error on classes with names differing in case

kozdra, 2024-05-09:
On current master, as well as older versions, Verilator runs into a compilation error when compiling any code like the following:
```sv
class Cls;endclass
class CLS;endclass
```

The primary reason for that is that the class names are reflected in the file names of the headers that need to be both included from `V*__Syms.h`:
```c
#include "Vt___024unit__03a__03aCls__Vclpkg.h"
#include "Vt___024unit__03a__03aCLS__Vclpkg.h"
```
The above would be a problem in itself already on case-insensitive filesystems (including NTFS), but on binary-safe filesystems like ext4 it also manifests itself, because the 'include guard' defines in the headers are normalized to upper-case, and therefore become exactly the same:
```c
#ifndef VERILATED_VT___024UNIT__03A__03ACLS__VCLPKG_H_
#define VERILATED_VT___024UNIT__03A__03ACLS__VCLPKG_H_  // guard
```

Ideally, Verilator should detect the situation and change parts of the file name(s), or just the guard variables, but other options include modifying all the file names proactively (adding a hash or escaping uppercase letters like underscores etc.), or an early warning.

(loosely related: #2062)

wsnyder, 2024-05-09:
This seems a case where the code is doing something so "evil" that whoever coded it gets what is deserved ;) 

But seriously, would an error be acceptable, or must we make this work?

kozdra, 2024-05-10:
I can think of it as similar to SYMRSVDWORD — a rare coincidence that can be easily fixed by the user, but is relatively time-consuming to track down.  This actually happened in a real-world project (RISCV-DV), and only requires changing one identifier, but it would still be great to see the cause outright rather than after extensive debugging.

closed 2024-07-14 by wsnyder