verilator/verilator #5111: VexRiscv based simulation design failing compilation starting from v5.022

Sarmad-Salman, 2024-05-10:
This is a Linux Regression that is used to boot a linux kernel on a VexRiscv CPU, with verilator as the simulation tool.

The testcase can be found at this link: [https://github.com/os-fpga/Jira_Testcase/tree/main/EDA-2828](https://github.com/os-fpga/Jira_Testcase/tree/main/EDA-2828)

**Steps to Reproduce: -**

- Move into the /sim directory and hit make for the compilation to begin.

This was passing on earlier verilator versions starting from 5.002 up till 5.020 and began failing on 5.022, 5.024 and the current master branch.

Tested on Ubuntu 22.04.

wallento, 2024-05-10:
Thanks a lot. Can you please paste the error message you receive?

Sarmad-Salman, 2024-05-10:
Upon some debugging and research, i found that Verilator starting from v5.022 requires the C++14 standard instead of the previously used C++11 standard. Using a newer compiler fixes this.

closed 2024-05-10 by Sarmad-Salman