verilator/verilator #5120: Parameterized SV interface array not handled correctly when only one item

paul-demo, 2024-05-15:
When I have an array of SV interfaces like this, Verilator works as expected for N=4.

```
AXIS_IF.master m_axis_if[N-1:0]
```

However, when I change to N=1, I get errors where Verilator has apparently mixed up master/slave modports and thinks I am assigning to an sv signal with the wrong direction. Basically Verilator seems to have confused master/slave modports.

```
  AXIS_IF #(
      .DATA_W(AXIS_DATA_WIDTH),
  ) axis_if (
      .aclk   (clk),
      .aresetn(~reset)
  );
  assign axis_if.tdata = tdata;
```

This produces errors like:
```
Attempt to drive input-only modport: 'tdata'
  144 |   assign axis_if.tdata = tdata;
```

wsnyder, 2024-05-15:
We will need a complete, small example, thanks.

One thing to try is change toe default parameter value for N in the interface definition.

paul-demo, 2024-05-15:
I will try to reproduce with something less than my full design.

The interface definition is not arrayed and is basically just what you would expect:
```
/*
  AXI-Stream Interface
*/
interface AXIS_IF #(
    parameter int DATA_W = 128,
    parameter int KEEP_W = DATA_W / 8,
    parameter int STRB_W = DATA_W / 8,
    parameter int USER_W = 16,
    parameter int ID_W   = 8,
    parameter int DEST_W = 16,
    parameter int LAST_W = 1
) (
    input logic aclk,
    input logic aresetn
);
  logic [DATA_W-1:0] tdata;
  logic [ID_W-1:0] tid;
  logic [USER_W-1:0] tuser;
  logic [DEST_W-1:0] tdest;
  logic [LAST_W-1:0] tlast;
  logic [KEEP_W-1:0] tkeep;
  logic [STRB_W-1:0] tstrb;

  logic tvalid, tready;

  modport master(
      input aclk, aresetn,
      output tdata, tvalid, tkeep, tstrb, tlast, tid, tuser, tdest,
      input tready
  );

  modport slave(
      input aclk, aresetn,
      input tdata, tvalid, tkeep, tstrb, tlast, tid, tuser, tdest,
      output tready
  );

  modport monitor(
      input aclk, aresetn,
      input tdata, tvalid, tkeep, tstrb, tlast, tid, tuser, tdest, tready
  );

endinterface : AXIS_IF
```

Declarations/instances of interfaces of this type are declared as [N-1:0], since everything works fine when N=4.

paul-demo, 2024-05-16:
By continuously deleting things and linting the module using Verilator, I found that the problem is that Verilator doesn't handle names correctly when interfaces have the same name in different layers of hierarchy in the design!

```
interface MYIF ();
  logic tvalid, tready;
  modport master(output tvalid, input tready);
  modport slave(input tvalid, output tready);
endinterface : MYIF

module tb ();
  MYIF myif ();
  a a_i (.myif);
endmodule

module a (
    MYIF.slave myif
);
  b b_i ();
endmodule

module b ();
  // The following produces an error:
  //  Attempt to drive input-only modport: 'tvalid'
  //  xx |   assign myif.tvalid = '0;
  //     |               ^~~~~~
  // https://github.com/verilator/verilator/issues/5120

  MYIF myif ();
  assign myif.tvalid = '0;

  // Renaming myif in this module to anything else makes the error go away.
  // This shows there is a name-overloading bug: Verilator is getting confused
  // by the fact that there is a myif instance in module `tb` and a myif in module
  // `b`. Since module b has no ports, they cannot be referring to the same thing
  // in Verilog; but Verilator thinks they are the same and since module `a` myif
  // is a slave port, it would be a direction mismatch to drive it to '0 here.

  // Comment out the two lines above in this module, and uncomment these lines,
  // and the error goes away!
  // MYIF myif2 ();
  // assign myif2.tvalid = '0;

endmodule

```

paul-demo, 2024-05-16:
This seems like a pretty major flaw -- it means Verilator could be mixing up SV interfaces across levels of design hierarchy, just by chance if they happen to have the same name in their local block of code.

wsnyder, 2024-05-19:
This is because the module's parent has a interface of the same name, and the child and parent are both inlined.

As a workaround add to the child module

```
// verilator no_inline_module
```

Need more time as simple fixes all broke other things. Interfaces are complicated.

paul-demo, 2024-05-30:
No worries. I just avoid name reuse in the hierarchy for now when they don't actually refer to the same thing.

paul-demo, 2024-08-02:
I just saw another confusing case of this same thing. An interface at some point in the hierarchy dies out and all its fields are assigned to some other interface. Then further down, that same name is brought back to life. Verilator gets the width wrong -- since it thinks they're the same thing one of them ends up with the default width instead of the width it is parameterized to use.

We think that we see similar issues with VCS and also suspect "SV interface name reuse" as the culprit.