verilator/verilator #5121: Sometimes verilator produces invalid VCD, cadence simvision or Synopsis DVE can't display

algrobman, 2024-05-15:
Cadence/Synopsys tools could not read VCD produced by Verilator from one of our testbenches.

$version Generated by VerilatedVcd $end
$timescale 1ps $end
 **$scope module  $end
  $scope module $unit $end**
   $var wire 32 @G$ UVM_LOW [31:0] $end
  $upscope $end

Cadence tool did not show waves, Synopsys DVE complained about line 4 illegal syntax:

Error: [VCD2VPD-PRINT-ERROR] VCD syntax error.
A fatal error occurred at line 4:
Syntax error - $end expected while reading $scope.
The input VCD file contains unsupported syntax that cannot be converted to
VPD.
It could be that the input VCD was generated by a different simulator.
Please check the VCD file and try again.
Error: Conversion from VCD to VPD failed

I guess it was because no name in the $scope statements..

I had to patch it as :

$version Generated by VerilatedVcd $end
$timescale 1ps $end
 **$scope module aaa  $end
  $scope module bbb $end**
   $var wire 32 @G$ UVM_LOW [31:0] $end
  $upscope $end
  
 and the tools could display it  then..

I can't say what exactly causes this, but I didn't have this problem with the same design and verilator 4.211 ...
I also had no this problem with another small example. 

Could it be due usage of SV package / endpackage constructs as something new for VCD?

wsnyder, 2024-05-15:
It's unlikely related to a package, but possible.

Please create a test case for us to proceed on this - I suspect you can start with your design and rip out modules and signals, while making sure the issue persists.  Alternatively, debug Verilator, I'd suggest trace backwards the code that prints the code to declare scopes in the generated code.

algrobman, 2024-05-16:
>  Alternatively, debug Verilator,

Are you kidding ? I'm not programmer

wsnyder, 2024-06-11:
Closing until get a test case.

closed 2024-06-11 by wsnyder