verilator/verilator #5125: Vtrigprevexpr type do not match signal type, generated code fails to compile

pawel256, 2024-05-19:
version:
Verilator 5.025 devel rev v5.024-56-g5f85c8942

Please find attached a test in test_regress format.

[t_type_match.zip](https://github.com/verilator/verilator/files/15370455/t_type_match.zip)

Type used for __Vtrigprevexpr (in generated C++ code) do not match the type used for the signal itself

VL_IN8(c[1],0,0); // CData [1]

VlUnpacked<CData/*0:0*/, 1> __Vtrigprevexpr___TOP__t__DOT__i_top__DOT__i_unit__c__1;

And following operations fail to compile:

vlSelf->__VactTriggered.set(0U, vlSymsp->TOP__t__DOT__i_top__DOT__i_unit.c.neq(vlSelf->__Vtrigprevexpr___TOP__t__DOT__i_top__DOT__i_unit__c__1));

Vt_type_match___024root__DepSet_hb7a96300__0.cpp:18:80: error: request for member 'neq' in '((Vt_type_match__Syms* __restrict__)vlSymsp)->Vt_type_match__Syms::TOP__t__DOT__i_top__DOT__i_unit.Vt_type_match_unit::c', which is of non-class type 'CData [1]' {aka 'unsigned char [1]'}
     vlSelf->__VactTriggered.set(0U, vlSymsp->TOP__t__DOT__i_top__DOT__i_unit.c.neq(vlSelf->__Vtrigprevexpr___TOP__t__DOT__i_top__DOT__i_unit__c__1));

vlSelf->__Vtrigprevexpr___TOP__t__DOT__i_top__DOT__i_unit__c__1.assign(vlSymsp->TOP__t__DOT__i_top__DOT__i_unit.c);

In file included from Vt_type_match__ALL.cpp:4:
Vt_type_match___024root__DepSet_hb7a96300__0.cpp:21:118: error: no matching function for call to 'VlUnpacked<unsigned char, 1>::assign(CData [1])'
     vlSelf->__Vtrigprevexpr___TOP__t__DOT__i_top__DOT__i_unit__c__1.assign(vlSymsp->TOP__t__DOT__i_top__DOT__i_unit.c);

Thanks,
Pawel

wsnyder, 2024-05-19:
Thanks for the good test. Looks like it's not de-arraying the unpacked array "c".  Perhaps you'd be willing to look into why?

pawel256, 2024-05-20:
Hi Wilson,
thanks a lot for quick answer
I am not familiar with Verilator code, but I can give it a try
Any hints where to start looking?
Thanks,
Pawel

gezalore, 2024-05-20:
This is an unfortunate effect of keeping top level I/O as a native C arrays instead of `VlUnpacked` (which is the type we use for all other arrays, and is what the generated code assumes - it has a `neq` method, which is just a not-equals predicate).

One fix is to swap the operands of the `neq` (that is: change `a.neq(b)` into `b.neq(a)`, so 'this' is always a VlUnpacked, and then provide an overload of `neq` that takes a native C array. The `a.neq(b)` is an `AstCMethodHard` created in V3SenExprBuilder.cpp. The `neq` implementation lives in include/verlated_types.h (`class VlUnpacked`)

pawel256, 2024-05-20:
Thanks Geza,
I've followed your advice and changed the operands order on neq
after adding overloads on neq(), assign() and operator=() my test case is passing
Thanks,
Pawel

pawel256, 2024-05-20:
please find attached a patch with the changes

[0001-fix-for-issue-5125.patch.gz](https://github.com/verilator/verilator/files/15380481/0001-fix-for-issue-5125.patch.gz)

this is "git format-patch"
please let me know if other format would be more convenient

gezalore, 2024-05-20:
Would you mind raising a pull requests including the test this fixes please?

gezalore, 2024-05-20:
> Thanks Geza,
> I've followed your advice and changed the operands order on neq
> after adding overloads on neq(), assign() and operator=() my test case is passing
> Thanks,
> Pawel
> 

I think operator= should not be necessary (we use .assign instead

pawel256, 2024-05-20:
> I think operator= should not be necessary (we use .assign instead)

for the test case Verilator generates code like that:

void Vrl___024root___eval_initial(...)
[...]
vlSelf->__Vtrigprevexpr___TOP__v__DOT__i_unit__c__0
        = vlSymsp->TOP__v__DOT__i_unit.c;

where:
VlUnpacked<CData/*0:0*/, 1> __Vtrigprevexpr___TOP__v__DOT__i_unit__c__0;
and ...unit.c is
 VL_IN8(c[1],0,0);

clearly .assign() is not used here
and this code would not compile without operator=() overload

gezalore, 2024-05-20:
The intention is that for unpacked arrays, it should be generating a call to `.assign` (In general we try to avoid operator overloading so we don't accidentally generate inefficient code - the `.assign` is an understood special case where this is OK).

I think we need to change this:
https://github.com/verilator/verilator/blob/5f85c89425c1256faacf4736c1ae5fec2044cf1a/src/V3SenExprBuilder.h#L123-L125

to something like this:
https://github.com/verilator/verilator/blob/5f85c89425c1256faacf4736c1ae5fec2044cf1a/src/V3SenExprBuilder.h#L142-L149

commit 913679f2 (2024-05-22): Fix output C++ type error on change detect of I/O arrays (#5125) (#5131)

operand order reversed for AstCMethodHard "neq"
interface between C-style arrays and VlUnpacked
overloads added to VlUnpacked::neq(), VlUnpacked::assign()
VlUnpacked::operator=() added

Fixes  #5125