verilator/verilator #5127: Optimizer removes side-effecting assignment in initial block

sethp, 2024-05-20:
Deep in IceNet, there's a snippet of verilog that looks like this:

```verilog
    string devname = "";
    int rlimit_gbps = 64;
    byte rlimit_inc = 1;
    byte rlimit_period = 1;
    byte rlimit_size = 8;
    int dummy;

   // ...

    initial begin
        dummy = $value$plusargs("netbw=%d", rlimit_gbps);
        dummy = $value$plusargs("netburst=%d", rlimit_size);
        dummy = $value$plusargs("netdev=%s", devname);
        network_init(devname, rlimit_gbps, rlimit_inc, rlimit_period);
    end
```
—https://github.com/firesim/icenet/blob/7c17985a46108c8ff8d424b1b3af6eebc37c38a3/src/main/resources/vsrc/SimNetwork.v#L68-L73

When verilated,  `+netdev=...` wasn't getting picked up on the command line, and I couldn't find any writes to the `devname` variable anywhere. On a hunch, I changed the block to:

```
    initial begin
        dummy = $value$plusargs("netbw=%d", rlimit_gbps);
        dummy = $value$plusargs("netburst=%d", rlimit_size);
        if ($value$plusargs("netdev=%s", devname))
            $display("using devname=%s", devname);
        network_init(devname, rlimit_gbps, rlimit_inc, rlimit_period);
    end
```

And the devname was indeed assigned from the plusarg. To me, this looks very similar to https://github.com/verilator/verilator/issues/3131 —the optimizer isn't "seeing" that the plusargs call changes devname, and so can't be removed without changing the meaning of the program.

Verilator version:
```
$ verilator --version
Verilator 5.022 2024-02-24 rev conda-forge build 0
```
I haven't tried the latest from git; if you think it's likely to make a difference I'll give it a shot.

What OS and distribution are you using? Arch Linux

May we assist you in trying to fix this in Verilator yourself?

Sure, I'd be happy to give it a shot: I assume the fix would be somewhat similar to 08c8b0d7d655d2c8afe91114818e2e943bc8a601 , if you can point me towards the rough area of the optimizer that's deleting the AST nodes when they're writing to a never-read variable.

wsnyder, 2024-05-20:
Thanks for the good bug report, if you can look at this it's likely straightforward to make a pull to fix it.
1. Please update the tests, or make a new one to show the issue.
2. Run with --debug and look at the .tree files to see where the value setting (VALUEPLUSARGS) disappears.
3. Debug the source file corresponding to that stage (e.g likely V3Const), to see why it isn't honoring the isPure() that should be set on the AstValuePlusArgs.

sethp, 2024-05-22:
Thanks, that's very helpful! I'll see what I can do with your guidance here.

FWIW, if someone else happens along and fixes the issue before I get to it, I'd be perfectly fine with that outcome too!

wsnyder, 2024-07-02:
Thanks for the pull!  I made some other changes related to this.

closed 2024-07-02 by wsnyder