verilator/verilator #5135: `$fatal` does not call `$finish` and leads to coredump

goekce, 2024-05-24:
```sv
module m;
  always #1 $display($time);

  final $display("final");

  initial begin
    #3;
    $fatal("Timeout");
  end
endmodule
```
Compilation & running:
```
verilator --binary -j 0 -MAKEFLAGS '--silent' --quiet m.sv && obj_dir/Vm +verilator+quiet
```
Output:
```
                   1
                   2
[3] %Fatal: m.sv:8: Assertion failed in TOP.m
%Error: m.sv:8: Verilog $stop
Aborting...
Aborted (core dumped)
```

v5.024-56-g5f85c8942 5f85c8942 on ArchLinux

Problems:
- According to LRM 20.10 `$fatal` implicitly calls `$finish`, which should execute the `final` block.
- It leads to coredump somehow

Other observations:
- Verilator allows `+verilator+error+limit+`, this also overrides `$fatal`. Is this intended? IMHO `+verilator+error+limit+` is great for continuing the simulation a bit to see the context. However `$fatal` should override error limit.
- Also `$error` does not execute the `final` block. `final` should be executed at the end of the simulation. How would you interpret *the end of simulation*?

wsnyder, 2024-05-24:
As to evaluating final blocks, duplicate of #4980.

The core dump is (currently until #4980 fixed) intentional, that's why it says aborting.

I agree error limit shouldn't affect fatal.

All are historical simplifications from when there was no event loop in Verilator.

Pull(s) welcome to clean all this up.

goekce, 2024-05-24:
Thanks, I revised the title. The #4980 is about catching the exception and executing `final` I believe. I still leave the keyword `coredump` here so that this issue can be easily found by others.

wsnyder, 2025-04-05:
Closing, as the error limit was resolved, and rest covered by #4980.

closed 2025-04-05 by wsnyder