verilator/verilator #5136: Enabling SystemC multi-thread simulation?

mbalboni07, 2024-05-24:
Hello,
I would like to know if someone has some suggestions/hints to parallelize on multi-thread a SystemC simultaion
Here a brief description of the scenario I would like to tackle.
Let's say I have a complex component with sub-modules (verilated with option thread=1).
If I have a SystemC sc_main instantiating 2 (up to N) instances of the same verilated module, do you know some way to make the simulation kernel mapping on different physical threads/cores of the machine the sim execution?
This way getting a parallelization?

Any kind of pointers, hints here would be great.

Big thanks in advance to all

Marco

wsnyder, 2024-05-28:
I personally haven't done any research into SystemC kernel multithreading.  I did turn up this paper that might be of interest [Standard-compliant parallel SystemC simulation of
loosely-timed transaction level models](https://hal.science/hal-03487607/document).

Another possibility without hacking the kernel, if you model supports it, would be to e.g. add an input FIFO between the SystemC inputs and the Verilated model inputs, and another FIFO between Verilated outputs and SystemC outputs.  These FIFOs allow the Verilated model to execute in parallel on another thread(s).  This is similar to the technique often used to allow multiple systems to share simulation while hiding the latency between the systems.

Maybe others have additional ideas.

wsnyder, 2024-06-15:
Closing - others feel free to comment even though closed.

closed 2024-06-15 by wsnyder