verilator/verilator #5138: `VerilatedVpiPutHolder` kinda chunky

ktbarrett, 2024-05-24:
Inertial writes can queue up a number of `VerilatedVpiPutHolder` objects. Each of these objects is 160 bytes on a 64-bit Linux machine. It might be worthwhile to find a way to reduce the size of these.

Ideally we only need a reference to the object being assigned and the value to be assigned (should be a tagged union). This could be as small as 48 bytes on such a system (pointer to handle: 8 bytes, union for value: 32 bytes (string is 32), tag: 8 bytes (align is 8)).

wsnyder, 2024-05-24:
Probably the easiest thing would be separate the common case of only needing m_value versus needing a string/vector.

Pull request to shrink this welcome.

wsnyder, 2024-05-24:
Probably the easiest thing would be separate the common case of only needing m_value versus needing a string/vector into two separate classes.

Pull request to shrink this welcome.

ktbarrett, 2024-05-24:
Not sure if putting these in separate classes is the best play. They become heap allocated, dynamically dispatched, and only 32-56 bytes smaller. Most of the overhead comes from `VerilatedVpioVar` as there are 80 bytes of unused fields in that object.

wsnyder, 2026-08-28:
Agree with points and no objections to cleanup PRs.  Don't see need to leave issue open on this.

closed 2026-08-28 by wsnyder