verilator/verilator #5139: Unable to -include headers in Cflags due to precompile header

ngraybeal, 2024-05-25:
I believe https://github.com/verilator/verilator/pull/4580 broke the ability to `-include` header files in the `-CFLAGS` option. 

Here are a few issues filed that all share the same symptom of the problem. They appear to all be closed as they didn't actually need the header files, but didn't address the case where the header file was necessary for a provided `cpp` file. This was working before the PR.
https://github.com/verilator/verilator/issues/4839
https://github.com/verilator/verilator/issues/4785
https://github.com/verilator/verilator/issues/4730

The issue manifests itself as g++ complaining it cannot find the `*__pch.h.fast` file. The make flow does generate the precompiled header `*__pch.h.fast.gch` and compiles the user provided `-include` header into the precompiled header. The issue arises when it compiles the `.cpp` files, it adds `-include` for both the user header `foo.h` and the precompile header `*__pch.h.fast`. g++ ignores the existence of the precompiled header (it does check automatically for files with `.gch`) and appears to expect `*__pch.h.fast` to exist since it needs to compile both header files. I [see](https://gcc.gnu.org/onlinedocs/gcc/Precompiled-Headers.html) that gcc only supports one precompiled header per compilation, but didn't see anything about mixing headers with precompiled headers.

Does verilator need a new option to pass `-includes` outside of `-CFLAGS` so it can be included in the precompiled header, but not in the cpp compilation? Or must users now compile their cpp files with the dependencies beforehand and link them in?
I don't believe verilator generates a precompiled for every compiled design, I didn't look into what causes the fast/slow files. 

Build Information
Verilator: 5.024 (and tried on latest on master)
OS: Rhel8.7
GCC: 13.2.0

Here's what I did to recreate this problem.
```
$ cd test_regress
$ cp -p t/t_dpi_open_vecval.v t/t_include.v 
$ cp -p t/t_dpi_open_vecval.pl t/t_include.pl
$ cp -p t/t_dpi_open_vecval_c.cpp t/t_include_c.cpp
$ touch t/t_include.h
$ t/t_include.pl
```

`t/t_include.pl`
```
#!/usr/bin/env perl
if (!$::Driver) { use FindBin; exec("$FindBin::Bin/bootstrap.pl", @ARGV, $0); die; }
# DESCRIPTION: Verilator: Verilog Test driver/expect definition
#
# Copyright 2003 by Wilson Snyder. This program is free software; you
# can redistribute it and/or modify it under the terms of either the GNU
# Lesser General Public License Version 3 or the Perl Artistic License
# Version 2.0.
# SPDX-License-Identifier: LGPL-3.0-only OR Artistic-2.0

scenarios(simulator => 1);

compile(
    v_flags2 => ["t/t_include_c.cpp"],
    verilator_flags2 => ["-Wall -Wno-DECLFILENAME -unroll-count 1 -CFLAGS '-include $Self->{t_dir}/t_include.h'"],
    );

execute(
    check_finished => 1,
    );

ok(1);
1;
```

ERROR
```
ccache g++ -O0  -I.  -MMD -I/scratch/verilator/test_regress/../include -I/scratch/verilator/test_regress/../include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-shadow -Wno-sign-compare -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable    -include /scratch/verilator/test_regress/t/t_include.h   -DVERILATOR=1 -DVL_DEBUG=1 -DTEST_OBJ_DIR=obj_vlt/t_include -DVM_PREFIX=Vt_include -DVM_PREFIX_INCLUDE="<Vt_include.h>" -DVM_PREFIX_ROOT_INCLUDE="<Vt_include___024root.h>" -DT_INCLUDE   -DVL_LOCK_SPINS=10000 -include Vt_include__pch.h.fast -c -o Vt_include.o Vt_include_c.cpp
cc1plus: fatal error: Vt_include__pch.h.fast: No such file or directory
compilation terminated.
```

wsnyder, 2024-05-25:
Great analysis. Likely a long wait before clang fixes this, so might you make a pull to add e.g. `--compiler-include` flag?

b-chmiel, 2024-06-24:
@wsnyder I have implemented `--compiler-include` flag that resolves this issue. Can you assign me to it?

wsnyder, 2024-06-27:
Thanks for working on this.

>Sidenote: Should we warn users that use -CFLAGS -include '<header-path>' to use --compiler-include instead?

Hmm, I'm worried about false positives, let's wait and see if this becomes more a problem.

closed 2024-06-27 by wsnyder

zzhwarrior, 2025-02-20:
I tried to include two header files then the error occurred: `cc1plus: error: one or more PCH files were found, but they were invalid
cc1plus: note: use -Winvalid-pch for more information
<command-line>: fatal error: VTestHarness__pch.h.fast: No such file or directory`
`	--compiler-include $(GEN_COLLATERAL_DIR)/verilator.h \
	--compiler-include $(model_header) \`

wsnyder, 2025-02-20:
@zzhwarrior  please file a new issue with a small self-contained example showing the problem and we will discuss there.