verilator/verilator #5140: Installation of architecture specif binaries isn't FHS compatible

tijuca, 2024-05-25:
Hello,

after packaging the latest release 5.024 for Debian the Lintian QS tooling discovered that now the installation process is installing architecture specific files into the folder `/usr/share/` which isn't respecting the Linux File Hierarchy System. These files need to go into `/usr/bin` or maybe `/usr/libexec`.

https://refspecs.linuxfoundation.org/FHS_3.0/fhs/ch04s11.html
https://refspecs.linuxfoundation.org/FHS_3.0/fhs/ch04s07.html

In detail the following files are at least in the wrong folder placed.
```
root@i7:/tmp/buildd/verilator-5.024/debian/verilator/usr/share/verilator/bin# file verilator_bin verilator_bin_dbg verilator_coverage_bin_dbg 
verilator_bin:              ELF 64-bit LSB pie executable, x86-64, version 1 (GNU/Linux), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, BuildID[sha1]=e657dbf47183fbce8e8519dfb6098fed075a61df, for GNU/Linux 3.2.0, stripped
verilator_bin_dbg:          ELF 64-bit LSB pie executable, x86-64, version 1 (GNU/Linux), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, BuildID[sha1]=a8e3f8a4e4d774f0825e238fa1c28ed93be0fc27, for GNU/Linux 3.2.0, stripped
verilator_coverage_bin_dbg: ELF 64-bit LSB pie executable, x86-64, version 1 (GNU/Linux), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, BuildID[sha1]=b093c8887f6e45a0f2fafca9df5098f94a66519a, for GNU/Linux 3.2.0, stripped
```
As the other files here also intended to be callable and somehow executable I think it's fine to install all files again (as before) into `/usr/bin`.

wsnyder, 2024-05-25:
@gezalore?

gezalore, 2024-05-26:
As we discussed in #4893, the current way was the least intrusive to enable the installation to be relocatable.

See option 'b' here, for the implications if we insist on the binaries being in a different place, while maintaining testability of the relocated install:
https://github.com/verilator/verilator/issues/4893#issuecomment-1948442069

To respect the FSH, we have a couple options:

1. Revert the #4927 and go back to the old ways
2. Do work in 'option b' from the previous discussion (if VERILATOR_ROOT is defined, it overrides both VERILATOR_BINDIR - what points to the directory with the binaries - and VERIALTOR_GUTS - what points to the directory with the internal parts, like `include` and the private binaries)
3. Add a configure option `--enable-relocatable`, without which it installs in the old way, and the installation is not relocatable.

I would say 2. is the "cleanest" but most complicated and intrusive. I'm happy with either option, or any other ideas, do you have a preference?

tijuca, 2024-05-28:
@wsnyder 
Yes, exactly. Place all the files and binaries which are currently installed into `/usr/share/verilator/bin` into /`usr/bin` and it's fine. Don't mind about Perl/Python scripts that then are going into /`usr/bin`, that's quite common practice and done by other projects too.

gezalore, 2024-05-28:
> we wouldn't need a major change as I think "b" suggests.

The problem is testing. driver.pl et al. picks up components through VERILATOR_ROOT, but if the binaries don't live under `$VERILATOR_ROOT/bin`, then we need a new path to access them during testing. The bulk of the required changes are limited to `test_regress/`

wsnyder, 2024-05-28:
@tijuca is anything breaking?  I understand why this standard was made, it was important for NFS mounted drives across multi architecture clusters in the 1990s, but I with RPM it is still relevant in 2020s to avoid breakage of something (what)?

nickelpro, 2024-10-11:
Someone needs to blink here, because this is starting to break containment with the release of Ubuntu 24.10 shipping a totally broken Verilator package traced to this dispute. Ran into this from a massive spike in students being unable to configure and tests their projects. (Not anyone here's fault, Ubuntu's fault for unquestioningly shipping unstable packages)

There's some unconventional terminology at play here, the normal meaning of `<PACKAGE>_ROOT` is where the package is installed to. That should typically be `/usr` for a typical system install, and Verilator should thus be expecting its binaries and scripts in `/usr/bin`, architecture intendent data at `/usr/share` etc. If the user has a custom Verilator install, such as `/opt/verilator` it would be the same story (/opt/verilator/bin, etc).

A script being run _outside_ the `<PACKAGE>_ROOT` (ie, /usr/bin/verilator) and redirecting to some unrelated "package root" (/usr/share/verilator) is outside the normal expectations of downstream packagers.

I ran into the strange assumptions of driver.pl while [trying to make testing work out of a CMake install folder](https://github.com/verilator/verilator/pull/5098) to support a more conventional CMake workflow but gave up over it only supporting in-source builds.

My two cents is that driver.pl shouldn't make these kinds of assumptions, and that the conventional installation tree should be respected.

wsnyder, 2024-10-11:
Please see the notes in the other comments.  I'd suggest contributing a pull along the lines of option "b" with some extensions:
1. VERILATOR_ROOT is defined, then that is used to intelligently default the other variables below.  (The above suggests that VERILATOR_ROOT overrides the below - that doesn't feel right to me, but maybe I missed something.)
2. VERILATOR_INCLUDE has the path to includes, if not set, VERILATOR_ROOT/include. If VERILATOR_ROOT not set, then hardcoded from $pkgdatadir)/include at config.
3. VERILATOR_BINDIR has path to binaries.  If not set VERILATOR_ROOT/bin, If VERILATOR_ROOT not set, then hardcoded from $bindir at config
4. VERILATOR_PKGDATADIR/bin has path to relocatable scripts.  If not set VERILATOR_ROOT, if VERILATOR_ROOT not set, then hardcoded from $pkgdatadir at config

The current DEFENV_VERILATOR_ROOT used in src/Makefile would extend to have DEFENV_VERILATOR_INCLUDE etc as indicated above.  Likewise `verilator -V` should print all these variables so we can debug when things go bad.

Note driver.pl isn't part of the install, and were not designed to be used against a system-installed verilator.  However @gezalore did a lot of work to make many of the tests able to support running against builds outside the default git tree, so likewise if you want to make pulls to improve this they are welcome.

gezalore, 2024-10-11:
I will fix this according to what you just said

closed 2024-10-18 by gezalore

gezalore, 2024-10-18:
Fixed by #5544. Please let me know if there are more issues @tijuca @nickelpro.

wsnyder, 2024-10-18:
@gezalore thanks. @tijuca @nickelpro if you need a version release also to get this out there, let me know.

PR #5544 (2024-10-18): Install public executables into bindir instead of pkgdatadir (#5140)
I found a minimally invasive way to respect the FHS, by swapping the redirect scripts and actual executables. Remember the reason we needed the redirect scritps, is because we want pkgdatadir to have the same layout as the source tree for testing purposes. With the swap (so in the installation pkdatadir/bin/ contains redirects to bindir/), it works just fine.

To do this, we need to do one install-time patch to the `verilator` wrapper script, to tell it the relative path between bindir and pkgdatadir.

VERILATOR_ROOT, just li [... cut]
commit 55661e7f (2024-10-18): Install public executables into bindir instead of pkgdatadir (#5140) (#5544)

Instead of installing redirect scripts in to bindir, and real
executables in to pkgdatadir/bin, we now put the real executables into
bindir, and add redirect scripts into pkgdatadir/bin. (That is, we
redirect in the opposite direction as before).

This enables the same relocatable and testable installations as before
without putting architecture specific binaries into pkgdatadir, so
hopefully we now conform to the Linux Filesystem Hierarchy Standard.

This requires a small fixup in the installed `verilator` script to [... cut]