verilator/verilator #5148: %Error: Value too wide for 32-bits expected in this context

Dercury, 2024-05-29:
I used verilator in a big project, and got an error: Value too wide for 32-bits expected in this context.
So, i checked it's verilog code, and simplifed an example and still got this error:
`
%Error: ../expand_32bits.v:12:44: Value too wide for 32-bits expected in this context 35'h100000990
                                : ... note: In instance 'expand_32bits'
-node: CONST 0x13958a0 <e1131#> {d12bs} @dt=0x13a6290@(G/w35)  35'h100000990
   12 |     vlan <= pkt_data[ { (WIDTH-12), 3'b0 } - 1 -: 16];
      |                                            ^
`
# example
`
// expand_32bits.v
module expand_32bits
#(
    parameter WIDTH = 320
)
(
    input clk,
    input [2559:0] pkt_data,
    output reg [15:0] vlan
);

always @(posedge clk) begin
    vlan <= pkt_data[ { (WIDTH-12), 3'b0 } - 1 -: 16];
end 

endmodule
`
# command line
--sc --debug --gdbbt --Wno-fatal ../expand_32bits.v

# verilator --version
Verilator 5.024 2024-04-05 rev UNKNOWN.REV
I didn't try it with the git master version.

#  OS distribution 
CentOS Linux release 7.9.2009 (Core)

# May we assist you in trying to fix this in Verilator yourself?
Yes, please.

Dercury, 2024-05-29:
btw, i modified verilog like below, and it's ok
`//vlan <= pkt_data[ { (WIDTH-12), 3'b0 } - 1 -: 16];
vlan <= pkt_data[ ((WIDTH-12) << 3)  - 1 -: 16];`

Dercury, 2024-05-29:
I think there is no way to get an '35'h100000990' in the verilog, so i checked verilator's source, added some track UINFOs. I got this output:
```
- V3Number.cpp:1894:  opSub lhs:?32?sh140 rhs:?32?shc
- V3Number.cpp:1861:  opNegate lhs:?32?shc
- V3Number.cpp:1874:  opAdd lhs:32'hfffffff3 rhs:32'h1
- V3Number.cpp:1890:  opAdd result:32'hfffffff4
- V3Number.cpp:1870:  opNegate result:32'hfffffff4
- V3Number.cpp:1874:  opAdd lhs:?32?sh140 rhs:32'hfffffff4
- V3Number.cpp:1890:  opAdd result:32'h134
- V3Number.cpp:1902:  opSub result:32'h134
- 
- V3Number.cpp:1894:  opSub lhs:35'h9a0 rhs:35'h1
- V3Number.cpp:1861:  opNegate lhs:35'h1
- V3Number.cpp:1874:  opAdd lhs:35'h7fffffffe rhs:35'h1
- V3Number.cpp:1890:  opAdd result:35'h7ffffffff
- V3Number.cpp:1870:  opNegate result:35'h7ffffffff
- V3Number.cpp:1874:  opAdd lhs:35'h9a0 rhs:35'h7ffffffff
- V3Number.cpp:1890:  opAdd result:35'h99f
- V3Number.cpp:1902:  opSub result:35'h99f
- 
- V3Number.cpp:1894:  opSub lhs:35'h99f rhs:32'hf
- V3Number.cpp:1861:  opNegate lhs:32'hf
- V3Number.cpp:1874:  opAdd lhs:32'hfffffff0 rhs:32'h1
- V3Number.cpp:1890:  opAdd result:32'hfffffff1
- V3Number.cpp:1870:  opNegate result:32'hfffffff1
- V3Number.cpp:1874:  opAdd lhs:35'h99f rhs:32'hfffffff1
- V3Number.cpp:1890:  opAdd result:35'h100000990
- V3Number.cpp:1902:  opSub result:35'h100000990
```

wsnyder, 2024-05-29:
>35'h99f - 32'hf produced 35'h100000990

Note the width of this is 35 bits.  The current code assumed 32 bit math, which was the reason for that error, as you found.

The workaround is to keep indices to 32-bit math, as you discovered.

Yes, the fix seems reasonable, please create a pull along with the test case, thanks!

wsnyder, 2024-06-07:
Thanks for the pull - merged.

closed 2024-06-07 by wsnyder