verilator/verilator #5154: Read & write (AXI-Lite) register using DPI-C imported function/task

abarajithan11, 2024-06-03:
Verilator does not allow DPI-C imported tasks cannot pass simulation time, else they get deadlocked as per this issue: #4225, unlike other simulators (Vivado xsim...).

### My use case:
I am building an SV design that runs on the FPGA & controlled by the host ARM processor running a custom C firmware. I am using Verilator and Vivado Xsim to test them together in simulation. I have built the firmware as a set of C functions, and I call them from a main SystemVerilog testbench via DPI-C.

Everything works fine, except writing into an AXI-lite register bank from the C firmware. Since the register bank is in SystemVerilog and not memory mapped to C's memory space, I need to export an SV function/task, import it in C, and use it to read & write to the register. But **writing to AXI-Lite register takes one or more clock cycles**, and the SV task I import in C cannot pass time, due to the above Verilator issue.

### Workaround

My current workaround is to directly update the memory in SV function, bypassing AXI-Lite as follows:

#### SV testbench
```sv
export "DPI-C" function get_config;
export "DPI-C" function set_config;
function integer get_config(input integer offset);
  return dut.axi_lite_controller.cfg[offset];
endfunction

function void set_config(input integer offset, input integer data);
  dut.axi_lite_controller.cfg[offset] = data;
endfunction
```

#### C firmware
```c
#ifdef SIM
  extern int32_t get_config(int32_t offset);
  extern void set_config(int32_t offset, int32_t data);
#else // when running on FPGA
  int32_t get_config(int32_t offset){
    return *((int32_t*)(CONFIG_BASEADDR + offset));
  }
  void set_config(int32_t offset, int32_t data){	
    *((int32_t*)(CONFIG_BASEADDR + offset)) = data;
  }
#endif

void runtime() {
 // ...
 int conf3 = get_config(3);
 set_config(3, 27);
 // ...
}
```

### Problem:

But this does not verify the AXI-Lite pipeline, which makes me uncomfortable. Is there a way I can call `get_config(offset)` and `set_config(offset, data)` from C, to read & write from AXI-Lite register bank?

udif, 2024-06-03:
Let me add my 2 cents...
As the one who opened #4225 I ended up solving the memory read/write problems by adding a simple FIFO in front, delegating the bus access to verilog code, and splitting the read function into two parts, on that sends the request by writing it into a FIFO (using zero time) and one that checked for response.

1. The the request C part checks if the request FIFO is not full and adds a request. If it is full it returns an error code, so you can retry later (all in zero-time).
2. The verilog request part checks if there is anything in the FIFO and executes the transaction, taking one or more cycles if the bus is not busy.
3. The response verilog part checks if a response if available, saving it in a response FIFO.
4. The response C part checks if the FIFO is not empty, and returns the result if available. If none available, it returns an error code so it can be retried later.

The solution worked (with verilator), but I got x4 worse performance on an 8-core machine that a single core license of our commercial tool (using the **same** split DPI/Verilog code), so I had postponed using it. It also stopped working at some point in time, and I couldn't figure out what broke it. I plan coming back to this when I have time and investigate the issue (don't worry, the commercial tool has its stability issues as well).

In my case, I had **two** buses to worry about, one narrow for control, and one wide to load data.
loading the code+data through the wide bus took too long, and I ended up loading it via backdoor:
I used `slang-hier` which is a very simple tool I wrote (shameless plug) using [slang](https://github.com/MikePopoloski/slang), that produced for me a flattened list of all instances, from which I filtered out all the memories I needed. Some of the memories were interleaved, some had more then one instance for providing a deeper memory, and some had both.
I also had an external table with addresses for each memory.
The `slang-hier` is now part of the `slang` distribution, under the `tools` directory, and built by default.

My script eventually produced a C function that reads and writes the memory using backdoor VPI calls.
The speed boost was huge, with memory loading almost instantaneous, even when the memory was loaded word-by-word, without $readmemh/$writememh.
My specific script cannot be shared as it is too project specific and part of my employer's proprietary code.

abarajithan11, 2024-06-03:
Hi, thanks for the detailed response!

I'm not sure how I can add fifos to the above get and set functions. If I want to get a value from SV memory to C, how should these be to structured?

abarajithan11, 2024-06-03:
C code reading the status register and waiting would need to pass simulation time right?

I have something like

```c

void process(){
// some code 1
int c1 = 
}

udif, 2024-06-03:
In my case, we initiate memory reads and writes from a standalone client, and the Verilog simulator is spawned as a separate process and communicates via TCP/IP. The client converts the read/write requests into TCP/IP messages, and a thin C DPI code in the Verilog simulation listens to the messages.

My suggestion above assumed your top level control is from a host ARM processor as you wrote. I assumed you either have a SW model of the CPU or compile its firmware directly to the simulator host C compiler. In both cases, your top level can be C based, and can call the Verilator kernel every tick, as did all pre 5.x simulations (look at the Verilator example sim_main_c).

Reading your original issue again, I see your main testbench is SV, and you call your firmware functions as DPI. You could work around this if you had write-only calls, but if your code reads from your Verilog model, it blocks until it gets an answer. I can't suggest any method unless you do one of the following:

1. Change to a C top where you can independently call the Verilator eval() function yourself in the middle of your C call.
2. Separate the Verilog simulator to a separate process and communicate with your main process using sockets/files/shared mem/ other OS mechanism.
3. Change your firmware code somehow so that when it needs to read or write a value it will save it state, return to the caller with expected values, and let you get back to the saved state after your verilog bus model will perform the read/write operation.

abarajithan11, 2024-06-03:
Since I want to use the same testbench for XSIM and VCS, I think (3) is the only option. Thanks a lot

wsnyder, 2024-06-11:
Think this was answered, but feel free to follow up here.

closed 2024-06-11 by wsnyder