verilator/verilator #5159: [DFG optimization pass] Internal error on out-of-range slice

wkkuna, 2024-06-06:
Hi, working on the project I stumbled upon an internal error.
Both module package and the module itself had a address width parameter & I forgot to ensure that both of them carry the same value.
The error, in my case, is cause by a bit-slice but only if the slice is excluding the least significant bits.

Verilator version I'm using:
```
❯ verilator --version
Verilator 5.024 2024-04-05 rev v5.024
```

I was able to reproduce it on a smaller example:

```
\\ top.sv file
package top_pkg;
  parameter integer AW = 12;
  typedef struct packed {logic [AW-1:0] addr;} ctx_t;
endpackage

module top
  import top_pkg::*;
#(
    parameter integer AW2 = 32  // Address Width greater than one defined in mod_pkg
);
  logic [AW2-1:0] addr;
  ctx_t ctx;
  always_comb addr = ctx.addr[AW2-1:1];  // ctx.addr[AW2-1:0] won't cause an error
endmodule
```

I ran Verilator with:

```
❯ verilator --binary -j 0 -Wall top.sv
```

The log:
```
%Warning-DECLFILENAME: top.sv:1:9: Filename 'top' does not match PACKAGE name: 'top_pkg'
    1 | package top_pkg;
      |         ^~~~~~~
                       ... For warning description see https://verilator.org/warn/DECLFILENAME?v=5.024
                       ... Use "/* verilator lint_off DECLFILENAME */" and lint_on around source to disable this message.
%Warning-SELRANGE: top.sv:13:30: Extracting 31 bits from only 12 bit number
                               : ... note: In instance 'top'
   13 |   always_comb addr = ctx.addr[AW2-1:1];
      |                              ^
%Warning-SELRANGE: top.sv:13:30: Selection index out of range: 31:1 outside 11:0
                               : ... note: In instance 'top'
   13 |   always_comb addr = ctx.addr[AW2-1:1];
      |                              ^
%Warning-WIDTHEXPAND: top.sv:13:20: Operator ASSIGN expects 32 bits on the Assign RHS, but Assign RHS's SEL generates 31 bits.
                                  : ... note: In instance 'top'
   13 |   always_comb addr = ctx.addr[AW2-1:1];
      |                    ^
%Warning-UNUSEDSIGNAL: top.sv:11:19: Signal is not used: 'addr'
                                   : ... note: In instance 'top'
   11 |   logic [AW2-1:0] addr;
      |                   ^~~~
%Warning-UNDRIVEN: top.sv:12:9: Signal is not driven: 'ctx'
                              : ... note: In instance 'top'
   12 |   ctx_t ctx;
      |         ^~~
%Error: Internal Error: top.sv:13:26: ../V3DfgPeephole.cpp:753: OOPS
   13 |   always_comb addr = ctx.addr[AW2-1:1];
      |                          ^~~~
```

Note:
If `-fno-dfg-peephole` flag is passed the `Internal Error` is not displayed.

wsnyder, 2024-06-09:
@gezalore?   Not sure if the thing to do is fix the internal error, or add an (non-internal) error earlier.

wsnyder, 2024-06-09:
Thanks Geza, I'm fine with the warning change.

closed 2024-06-09 by wsnyder