verilator/verilator #5167: segfault during tracing - repeated runs with default VerilatedContext

pieter3d, 2024-06-10:
Seen at ToT today, linux (ubuntu).

The verilated design is created on the stack as a local variable (technically a class member of a googletest fixture), then a loop runs the design calling `eval()` as per usual. Afterwards the test exits and everything is cleaned up (`trace_file->close()` is called too).
If the test is run again, with the same design, then I am seeing segfaults when tracing is enabled. In particular, a heap-use-after-free determined by gcc's `-fsanitize=address` at:
```
    #0 0x55c81f60458c in VerilatedModel::contextp() const /home/pieter/work/verilator/include/verilated.h:284                                                              
    #1 0x55c81f758bd1 in VerilatedTrace<VerilatedFst, VerilatedFstBuffer>::addModel(VerilatedModel*) /home/pieter/work/verilator/include/verilated_trace_imp.h:632         
```
There seems to be some cleanup happening of the static global default VerilatedContext when a design is destructed (in my case, when the stack variable goes out of scope), when tracing is enabled. If I create a new VerilatedContext and pass it to the design's constructor, there is no issue and things run fine.

I am verilating the design with CMake, the `verilate` macro, with no additional args or flags 9other than `--trace-fst`. VCD tracing behaves the same way. And no issues using the default context when not tracing.

wsnyder, 2024-06-10:
I don't see any obvious places where the default VerilatedContext could be deleted, it's a singleton with no deletion.  Do you have an example you can post?  Or of course a pull request to fix it...

pieter3d, 2024-06-10:
Alright, let me work on a small repro case.

pieter3d, 2024-06-11:
CMakeLists.txt
```cmake
cmake_minimum_required(VERSION 3.14)
project(cmake_example)
find_package(verilator HINTS $ENV{VERILATOR_ROOT})
add_executable(test test.cc)
verilate(test SOURCES test.sv TRACE)

```
Design file, `test.sv`
```SystemVerilog
module test (
  input logic in,
  output logic out
);

assign out = !in;

endmodule
```

Testbench

```c++
#include "obj_dir/Vtest.h"
#include "verilated_vcd_c.h"

void run() {
#ifdef FIX
  auto ctx = std::make_unique<VerilatedContext>();
  Vtest test(ctx.get());
#else
  Vtest test;
#endif
  Verilated::traceEverOn(true);
  auto tf = std::make_unique<VerilatedVcdC>();
  test.trace(tf.get(), 99);
  tf->open("sim.vcd");
  for (int i = 0; i < 4; ++i) {
    test.in = i & 1;
    test.eval();
    tf->dump(i * 100);
    printf("out = %d\n", test.out);
  }
  tf->close();
}

int main(int argc, char *argv[]) {
  run();
  run();
  return 0;
}
```

When running this without `FIX` defined, the output is:
```
out = 1
out = 0
out = 1
out = 0
%Error: /home/pieter/work/verilator/include/verilated_trace_imp.h:636: The same model has already been added to this trace file
Aborting...
Aborted (core dumped)
```
With `FIX` defined, it works as expected.

This is a little different from the free-after-use segfault I saw, but I'm guessing it's the same underlying issue.

wsnyder, 2024-06-11:
Are you using git master?  I get:

`%Error: include/verilated_trace_imp.h:636: The same model has already been added to this trace file`

which now I see it makes sense - a VerilatedContext is never intended to be reused.  So my inclination is to close this as not a bug, ok?

pieter3d, 2024-06-11:
Well, it's kind of unintuitive that the default context can't be reused, and it's also weird that the way this shows up is with mystery crashes when tracing (and works fine when not tracing).

wsnyder, 2024-06-11:
Fair point, I'll add a clearer error if adding a model after time has passed.

pieter3d, 2024-06-11:
Also - the message is technically incorrect right? It's a new model and new trace, just that on the second run the context state isn't clean. 

On the second run, the trace should not already think it has a model added to it?

closed 2024-06-11 by wsnyder