verilator/verilator #5168: Clocking Blocks inside Modports are not Supported

VigneshRavisekar, 2024-06-10:
```
verilator --cc --Mdir vbuild --prefix Vtop --binary -Wno-INITIALDLY -Wno-lint
e top 2>&1 | tee Vlt comp. log ; ./vbui1d/Vtop 2>&1 | tee Vlt comp. log
%Error-UNSUPPORTED: wb
Modport clocking
Unsupported:
- -top-module
top
--assert
cpu
to
v
top.
I. --assert
--trace
-top-modul
27 |
modport WB DRIV( clocking wb drv);
. note: In file included from itop.svi
top.sv:4:1:
. For error description see https://veri1ator.org/warn/UNSUPPORTED?v=5.@25
%Error-UNSUPPORTED: wb
Modport clocking
Unsupported:
29 |
Exiting
%Error:
See
modport WB MON( clocking wb mon ) ;
top.sv:4:1:
due to 2
error( s)
. note: In file included from itop.svi
the manual at https://verilator.org/verilator doc.html for more assistance.
```

Tried including clocking blocks inside the Modports in the interface file but Verilator does not support Modports with Clocking Blocks.

wsnyder, 2024-06-10:
Could you post a self checking example for this feature?   Or, ideally a pull request to implement it?  I'm not sure when this will get worked on otherwise, maybe @kbieganski has plans?

kbieganski, 2024-06-10:
We have no plans for this at the moment.

VigneshRavisekar, 2024-06-10:
The issue has been previously raised and attaching the link for reference:
https://github.com/verilator/verilator/issues/4555#issue-1937565792

wsnyder, 2024-06-10:
#4555

closed 2024-06-10 by wsnyder