verilator/verilator #5174: Poor performance in assigning large bit width signals

Dercury, 2024-06-12:
I encountered poor performance when using Verilator to convert assignments between large bit width signals. This is the example below. In the generated code, a large number of temporary variables, "__Vtemp_+<number>", will be used (the number will increase from 249 to 1913, and the bit width will increase from 2048 to 15360), and the VL_CONCAT_WIW function will be used extensively (approximately 1665 times) to continuously perform memcpy and concatenation from __ Vtemp_249, all the way to __Vtemp_1913, and finally to the final __Vcellout_ shift_exceed_ch_sig_merge. The intermediate memcpys are too time-consuming, resulting in poor execution performance of the entire program.

# example
```verilog
// shift_exceed.v
module shift_exceed
#(
    parameter CH_NUM = 640,
    parameter CN_W = 22
)(
    input [CH_NUM-1:0] pd_fp_ind,
    input [CH_NUM-1:0] ch_en,
    input [CH_NUM * CN_W-1:0] ch_in,
    output reg [CH_NUM * (CN_W+2) - 1:0] ch_sig_merge
);

genvar i1;
generate
for (i1 = 0; i1 < CH_NUM; i1=i1+1) begin: GEN_SIG_MERGE
    assign ch_sig_merge[i1 * (CN_W+2) +: (CN_W+2) ] = {
            pd_fp_ind[i1],
            ch_en[i1],
            ch_in[i1 * CN_W +: CN_W]
        };
end 
endgenerate

endmodule
```
# command line
`verilator --sc --debug --gdbbt --Wno-fatal ../shift_exceed.v`

# verilator --version
Verilator 5.024 2024-04-05 rev UNKNOWN.REV
I didn't try it with the git master version.

#  OS distribution 
CentOS Linux release 7.9.2009 (Core)

# May we assist you in trying to fix this in Verilator yourself?
Yes, please. I haven't found the root cause yet, I don't know how to fix it.

Dercury, 2024-06-12:
The problematic generated code snippets:
```C++
VL_INLINE_OPT void Vshift_exceed___024root___ico_sequent__TOP__0(Vshift_exceed___024root* vlSelf) {
    (void)vlSelf;  // Prevent unused variable warning
    Vshift_exceed__Syms* const __restrict vlSymsp VL_ATTR_UNUSED = vlSelf->vlSymsp;
    VL_DEBUG_IF(VL_DBG_MSGF("+    Vshift_exceed___024root___ico_sequent__TOP__0\n"); );
    // Init
    VlWide<20>/*639:0*/ __Vcellinp__shift_exceed__ch_en;
    VL_ZERO_W(640, __Vcellinp__shift_exceed__ch_en);
    VlWide<20>/*639:0*/ __Vcellinp__shift_exceed__pd_fp_ind;
    VL_ZERO_W(640, __Vcellinp__shift_exceed__pd_fp_ind);
    VlWide<64>/*2047:0*/ __Vtemp_249;
    VlWide<65>/*2079:0*/ __Vtemp_250;
    VlWide<65>/*2079:0*/ __Vtemp_251;
    VlWide<65>/*2079:0*/ __Vtemp_252;
    VlWide<66>/*2111:0*/ __Vtemp_253;
    VlWide<66>/*2111:0*/ __Vtemp_254;

   // many lines

    VlWide<480>/*15359:0*/ __Vtemp_1909;
    VlWide<480>/*15359:0*/ __Vtemp_1910;
    VlWide<480>/*15359:0*/ __Vtemp_1911;
    VlWide<480>/*15359:0*/ __Vtemp_1912;
    VlWide<480>/*15359:0*/ __Vtemp_1913;                                                                                                                                                  
    // Body                 
    VL_ASSIGN_WSW(640,__Vcellinp__shift_exceed__pd_fp_ind, vlSelf->pd_fp_ind);
    VL_ASSIGN_WSW(640,__Vcellinp__shift_exceed__ch_en, vlSelf->ch_en);
    VL_ASSIGN_WSW(14080,vlSelf->__Vcellinp__shift_exceed__ch_in, vlSelf->ch_in);
    __Vtemp_249[0U] = (IData)((((QData)((IData)((1U
    // lines

    VL_CONCAT_WIW(2062,22,2040, __Vtemp_250, (0x3fffffU                                                                                                                                   
                                              & ((vlSelf->__Vcellinp__shift_exceed__ch_in[0x3bU]
                                                  << 0x12U)
                                                 | (vlSelf->__Vcellinp__shift_exceed__ch_in[0x3aU]
                                                    >> 0xeU))), __Vtemp_249);
    VL_CONCAT_WIW(2063,1,2062, __Vtemp_251, (1U & (
                                                   __Vcellinp__shift_exceed__ch_en[2U]
                                                   >> 0x15U)), __Vtemp_250);
    VL_CONCAT_WIW(2064,1,2063, __Vtemp_252, (1U & (
                                                   __Vcellinp__shift_exceed__pd_fp_ind[2U]
                                                   >> 0x15U)), __Vtemp_251);

   // many lines

    VL_CONCAT_WIW(15358,22,15336, __Vtemp_1912, (vlSelf->__Vcellinp__shift_exceed__ch_in[0x1b7U]
                                                 >> 0xaU), __Vtemp_1911);
    VL_CONCAT_WIW(15359,1,15358, __Vtemp_1913, (__Vcellinp__shift_exceed__ch_en[0x13U]
                                                >> 0x1fU), __Vtemp_1912);
    VL_CONCAT_WIW(15360,1,15359, vlSelf->__Vcellout__shift_exceed__ch_sig_merge,
                  (__Vcellinp__shift_exceed__pd_fp_ind[0x13U]
                   >> 0x1fU), __Vtemp_1913);                                                                                                                                              
    VL_ASSIGN_SWW(15360,vlSelf->ch_sig_merge, vlSelf->__Vcellout__shift_exceed__ch_sig_merge);
}
```

wsnyder, 2024-06-12:
Verilator is converting the part-wise assignment into one large assignment with a series of concats, which is reasonable and helps other optimization versus keeping the separate assignments.

Thus there are CH_NUM*3 (as you have 3 concats per loop), or 1920 concats in your code.

Verilator later gives up optimizing due to the signal being 15360 bits wide.  If you use --expand-limit 100000, it takes longer to verilate, but results in what seems to be nearly optimal code which computes an output 32-bit word at a time, and runs ~100x faster(!).

To do better, we could improve V3Expand so that when seeing concat trees it directly emits word-by-word assignments, which would remove the huge current code expansion and later V3Subst reduction to word assignments, to instead directly do word assignments.

Dercury, 2024-06-13:
' --expand-limit 100000' really works.
Btw, I remember that the old version (4.024) I used before did not have this issue.

wsnyder, 2024-11-26:
#5602 and related massively improve this.

closed 2024-11-26 by wsnyder