verilator/verilator #5177: Verilator simulation writes to memory in wrong module

davidharrishmc, 2024-06-12:
Verilator is failing a simulation that Questa and VCS are passing.  I eventually tracked the discrepancy down to a cache with four ways, each containing tag memories.  The Verilog writes to the memory in way 0. In Questa, the way 0 memory changes.  In Verilator, the way 2 memory changes.  This is quite strange!

The waveforms below show the issue.  Just before the rising clock edge at 376540, both CacheWays[0] and [2] receive addr = 00, ce = 1, and din = 08006F.  CacheWays[0] receives the write enable we = 1, while CacheWays[2] receives we = 0.  However, after the clock edge, entry 0 of the memory (ram0) changes in CacheWays[2] and not in CacheWays[0].

[image]

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

The failure is in CORE-V Wally
https://github.com/openhwgroup/cvw/issues/752

I can give detailed directions to reproduce this problem.  They are some effort to follow because they depend on building the test cases, which depend on the RISC-V GCC tool chain and other tools, which are a chore to install.  With some work, I can create a tarball of just the files needed, avoiding the tool chain.  Before I go through that process, let me know if there is any more clever solution to debugging this.

What 'verilator' command line do we use to run your example?

<set ulimit as recommended by verilator>
verilator \
--Mdir wkdir/rv32gc_arch32f -o Vtestbench \
--binary --trace \
 -DVERILATOR=1 --no-trace-top --stats \
-DMAKEVCD=1 \
--top-module testbench  --relative-includes \
"-I/home/harris/cvw/config/shared" "-I/home/harris/cvw/config/rv32gc" "-I/home/harris/cvw/config/deriv/rv32gc" \
/home/harris/cvw/sim/verilator/wrapper.c \
 \
/home/harris/cvw/src/cvw.sv /home/harris/cvw/testbench/testbench.sv /home/harris/cvw/testbench/common/*.sv /home/harris/cvw/src/*/*.sv /home/harris/cvw/src/*/*/*.sv

What 'verilator --version' are you using?  Did you try it with the git master version?

I recompiled just now.
harris@chips:/opt/riscv/verilator$ verilator --version
Verilator 5.025 devel rev v5.024-97-g320301940

What OS and distribution are you using?
Ubuntu 22.04.4 LTS

May we assist you in trying to fix this in Verilator yourself?
Delighted to collaborate, but I'm doubtful I could find the bug in Verilator source by myself.

wsnyder, 2024-06-12:
We're unfortunately unlikely to have the time to debug a complex test bench, or even something that is made into a large tarball.  Please reduce this to a small as possible test case - usually just the code around the problem signals is sufficient surrounded by a new test bench. Then check it fails on Verilator and passes on other simulators.  See docs/internals.rst for more information.

Alternatively, you can debug the code around the signals in the large example and see if you can isolate the problem.

Thanks.

davidharrishmc, 2024-06-14:
Verilator runs correctly on most of the riscv-arch-test suites and only fails on the larger cases, so unfortunately I have't been able to isolate to something small.I've made a significant effort to create as small of a standalone tarball as I can.  I hope this is sufficient.  

To reproduce the problem, extract the tarball and cd into 5177. Then run Verilator with make (which has a simple Makefile you can inspect).

```
$ make
verilator -o Vtestbench --binary -DVERILATOR=1 --no-trace-top --stats -DMAKEVCD=0 --top-module testbench --relative-includes -Iconfig/shared -Iconfig/rv32gc sim/verilator/wrapper.c src/cvw.sv testbench/testbench.sv testbench/common/*.sv src/*/*.sv src/*/*/*.sv
make[1]: Entering directory '/home/harris/5177/obj_dir'
...
make[1]: Leaving directory '/home/harris/5177/obj_dir'
- V e r i l a t i o n   R e p o r t: Verilator 5.025 devel rev v5.024-97-g320301940
- Verilator: Built from 296.237 MB sources in 248 modules, into 30.786 MB in 61 C++ files needing 0.014 MB
- Verilator: Walltime 5.751 s (elab=0.221, cvt=1.465, bld=3.826); cpu 0.000 s on 1 threads; alloced 185.051 MB
obj_dir/Vtestbench
-Info: testbench/testbench.sv:232: $dumpvar ignored, as Verilated without --trace
              **376550: CacheWay[2] RAM[0] changed even though we was pulsed on CacheWays[0] rather than [2]**
  Error on test rv32i_m/F/src/fadd_b11-01.S result        1020: adr = 8007c100 sim (D$) deadbeef signature = 00000000
%Error: testbench/testbench.sv:923: Verilog $stop
```

You'll see that testbench/testbench.sv has code starting line 274 to highlight the Verilator bug. The code checks when the write enable (we) to CacheWays[0] memory is asserted and to CacheWays[2] is not, and then when CacheWays[2] incorrectly gets written.  The Verilator simulation shows this happening at time 376550, as noted in the transcript above.

```
  always @(posedge clk)
    if (dut.core.lsu.bus.dcache.dcache.CacheWays[0].CacheTagMem.we == 1'b1 &
        dut.core.lsu.bus.dcache.dcache.CacheWays[2].CacheTagMem.we == 1'b0 &
        dut.core.lsu.bus.dcache.dcache.CacheWays[2].CacheTagMem.addr == 6'b0 &
        dut.core.lsu.bus.dcache.dcache.CacheWays[2].CacheTagMem.din == 22'h08006F &
        dut.core.lsu.bus.dcache.dcache.CacheWays[2].CacheTagMem.RAM[0] == 22'h08006E) begin
      @ (posedge clk);
      if ( dut.core.lsu.bus.dcache.dcache.CacheWays[2].CacheTagMem.RAM[0] == 22'h08006F)
        $display("%d: CacheWay[2] RAM[0] changed even though we was pulsed on CacheWays[0] rather than [2]", $time);
    end
```

The malfunctioning RAM module (src/generic/mem/ram1p1rwe.sv, used as CachTagMem) is simple.  As you can see, when we = 0, RAM should not change despite the testbench observing that it does change from 8006E to 8006F.

```
module ram1p1rwe import cvw::* ; #(parameter USE_SRAM=0, DEPTH=64, WIDTH=44) (
  input logic                     clk,
  input logic                     ce,
  input logic [$clog2(DEPTH)-1:0] addr,
  input logic [WIDTH-1:0]         din,
  input logic                     we,
  output logic [WIDTH-1:0]        dout
);

  bit [WIDTH-1:0]               RAM[DEPTH-1:0];
 
  // Combinational read: register address and read after clock edge
  logic [$clog2(DEPTH)-1:0] addrd;
  flopen #($clog2(DEPTH)) adrreg(clk, ce, addr, addrd);
  assign dout = RAM[addrd];

  logic [WIDTH-1:0] ram0;
  assign ram0 = RAM[0]; // just for gtkwave viewing purposes because arrays are hard to view

  always @(posedge clk)
    if (ce & we)
      RAM[addr] <= din;
endmodule
```

The Makefile also contains "make vcd" which produces a testbench.vcd suitable for examining in gtkwaves if that helps with the debug.

I wonder if there is some sort of Verilator pointer problem or memory overflow causing the wrong RAM to be written. Changing the size of the RAM (config/rv32gc/config.vh DCACHE_WAYSIZE_IN_BYTE = 32'd1024) changes which part of the test fails. Changing DCACHE_NUMWAYS from 4 to 1 makes the problem go away because there is no other CacheWay.

[5177.tar.gz](https://github.com/user-attachments/files/15836145/5177.tar.gz)

wsnyder, 2024-06-15:
I don't understand your assertion, the we should look at the previous cycle as the data is from a flop, I think this:

```
   reg past;
   always @(posedge clk) begin
     past <= (dut.core.lsu.bus.dcache.dcache.CacheWays[0].CacheTagMem.we == 1'b1 &
              dut.core.lsu.bus.dcache.dcache.CacheWays[2].CacheTagMem.we == 1'b0 &
              dut.core.lsu.bus.dcache.dcache.CacheWays[2].CacheTagMem.addr == 6'b0 &
              dut.core.lsu.bus.dcache.dcache.CacheWays[2].CacheTagMem.din == 22'h08006F);
     if (past &
         dut.core.lsu.bus.dcache.dcache.CacheWays[2].CacheTagMem.RAM[0] == 22'h08006E) begin
        @ (posedge clk);
        if ( dut.core.lsu.bus.dcache.dcache.CacheWays[2].CacheTagMem.RAM[0] == 22'h08006F) begin
           $display("[%0t] CacheWay[2] RAM[0] changed even though we was pulsed on CacheWays[0] rather than [2]", $time);
           $stop;
        end
     end
   end
```

With that this doesn't fire.

wsnyder, 2024-06-15:
P.S. what we really need is an assertion which checks for logical incirrectness directly, that is if the 'we' is wrong, the assertion computed 'we' from  other upstream signals and $stop if it mismatches, printing the expected and wrong results and intermediates.

davidharrishmc, 2024-06-16:
Wilson, I think my original assertion is correct and your modified assertion is off by one cycle.  The invisible delta delays in gtkwave are sneaky. Here's a timing diagram with delta delays exaggerated to make it easier to view.

[image]

At the positive edge of the clock at 376540, my if condition is satisfied.  My assertion then waits until the next clock edge at 376550, where it observes that ram0 way 2 has changed, and displays the failure.

At the positive edge of the clock at 376540, your & logic is satisfied.  After a delta delay, your past signal rises. At the positive edge at 376550, your if (past) is now true.  However, by this point, RAM is 6F, so your if body never fires.

I believe my assertion is as simple as possible to detect the inputs to a RAM setting up correctly and the output becoming incorrect after a clock edge.  I'm not able to parse what you mean by "the assertion computed we from other upstream signals".  we is correct.  The bug is that Verilator is writing into way 2 instead of way 0 even though we = 1 for way 0 and 0 for way 2.  This makes me suspect that Verilator garbled the pointer to the RAM.

wsnyder, 2024-06-16:
Thanks for the good diagram, appreciated.  Please ignore waves (don't trust them if simulation is suspect - the values they show are often computed separately from the signals themselves for speed reasons), also avoid the "@posedge(clk)" as timing is suspect. Instead use normal logic and use only $display and "normal logic". When I add a display to the ram:

```
  always @(posedge clk)
    if (ce & we) begin
       $display("[%0t] DBG %m.RAM[%x] <= %x (ce=%x we=%x)", $time, addr, din, ce, we);
       RAM[addr] <= din;
    end
```

I see:

`[376540] DBG TOP.testbench.dut.core.lsu.bus.dcache.dcache.CacheWays[2].CacheTagMem.RAM[00] <= 08006f (ce=1 we=1)
`

furthermore adding prints to the assertion you can see that the `we` was a 1 just before the '@posedge(clk);', but then goes to zero for your print, so it's misleading (or symptom of the larger issue).

Assuming the problem is still with EN, adding displays to look into the signals feeding further:

`[376540] DBG TOP.testbench.dut.core.lsu.bus.dcache.dcache.CacheWays[2] SetValidEN=1 SetValidWay=1 FlushStage=0`
further
`[376540] DBG TOP.testbench.dut.core.lsu.bus.dcache.dcache.CacheWays[2] SetValidEN=1=(SetValidWay=1=(SetValid=1 & SelectedWay=1) & ~FlushStage=0)`
further
`[376540] DBG TOP.testbench.dut.core.lsu.bus.dcache.dcache.CacheWays[2].flushlogic SelectedWay=1 VictimWay=1, FlushWay=0, FlushCache=0, SelectDirty=1 HitWay=0 SelVictim=1`

and all still seems fine from that logic working's standpoint.  

Anyhow I need more on what's specifically gone wrong.  It might be that the traces are wrong, but given the model fails more likely some signal feeding into this tree got scheduled wrong which is why tracing is also wrong.  Perhaps compare these displays to traces to see where things start mismatching?  Perhaps look at the C code to see where the logic goes bad?

goekce, 2024-06-17:
Could be related: In https://github.com/verilator/verilator/issues/5081 I found out that `--trace-fst` may change the simulation behavior related to assertions. You may also try whether deactivating `fst` has any effect.

davidharrishmc, 2024-06-17:
@goekce I am not familiar with --trace-fst and don't see it in my build command.  Is there a change I should make to the following build command to address your #5081 concern?

	verilator -o Vtestbench --binary -DVERILATOR=1 --no-trace-top --stats -DMAKEVCD=0 --top-module testbench --relative-includes -Iconfig/shared -Iconfig/rv32gc sim/verilator/wrapper.c src/cvw.sv testbench/testbench.sv testbench/common/*.sv src/*/*.sv src/*/*/*.sv

davidharrishmc, 2024-06-17:
@wsnyder thank you for the example display statement.  I've incorporated it, along with another display on the negedge clk, and a display on posedge we to try to determine whether we rises before or after clk.  Call these PE, NE, and WE to distinguish them. Is using negedge reliable in this context? $display should print values immediately without waiting for any delta delays to pass; is that a reliable expectation as well?

```
  always @(posedge we)
    $display("[%0t] WE  %m.RAM[%x] is %x, we rises (din = %x ce=%x clk=%x)", $time, addr, RAM[addr], din, ce, clk);

  always @(negedge clk)
    $display("[%0t] NE  %m.RAM[%x] is %x (din = %x ce=%x we=%x)", $time, addr, RAM[addr], din, ce, we);

  always @(posedge clk)
    if (ce & we) begin
      $display("[%0t] PE  %m.RAM[%x] <= %x (ce=%x we=%x)", $time, addr, din, ce, we);
      RAM[addr] <= din;
    end
```

The negedge clk display shows the RAM contents and inputs at a time when they are stable, to avoid delta delay confusion.  I print a lot of stuff, but  "grep -B 5 -A 50 376540 log" extracts the interesting parts around the bad clock edge, and I've trimmed out the irrelevant parts to show the logs below.

The NE log shows that on the nedgedge at 376535 before the rising edge of the clock, din and ce are correct and that the RAMs hold their old values.  It also shows that we = 0 for way 2 and we = 1 for way 0, as expected.  The NE log also shows that on negedge 376545 after the rising edge of the clock, 8006f was written to way 2 instead of way 0, which is the bug we are trying to trace. 

At 376540, the PE log shows way 2 we=1, as you discovered in your post above. This explains why way 2 is getting written.  
At 376540, the WE log also shows we rising in way 2.  It indicates that clk is already 1 when we rises.

I don't understand the subtle parts of how Verilator optimizes for cycle-based simulations with delta delays instead of #delay, but this behavior seems inconsistent with the SystemVerilog spec.  On the rising edge of the clock, RAM[addr] should change after one delta delay.  we should also change after at least one delta delay.  The WE display statement is consistent with this, showing clk=1 when we rises.  However, the PE display statement is inconsistent, showing we = 1 already at the posedge clk.

I agree with your theory that this is a scheduling problem, not a memory leak.  Maybe there is a delta delay scheduling problem such that ram1p1we sees we = 1 at @posedge clk, instead of a delta delay later?

I'll see what else I can trace in the C code, but this is outside my expertise.  Please let me know if there are other experiments I can run that would help. 

```
[376535] NE  TOP.testbench.dut.core.lsu.bus.dcache.dcache.CacheWays[2].CacheTagMem.RAM[00] is 08006e (din = 08006f ce=1 we=0)
[376535] NE  TOP.testbench.dut.core.lsu.bus.dcache.dcache.CacheWays[0].CacheTagMem.RAM[00] is 08006d (din = 08006f ce=1 we=1)
[376540] PE  TOP.testbench.dut.core.lsu.bus.dcache.dcache.CacheWays[2].CacheTagMem.RAM[00] <= 08006f (ce=1 we=1)
[376540] WE  TOP.testbench.dut.core.lsu.bus.dcache.dcache.CacheWays[2].CacheTagMem.RAM[00] is 08006e, we rises (din = 08006f ce=1 clk=1)
[376545] NE  TOP.testbench.dut.core.lsu.bus.dcache.dcache.CacheWays[2].CacheTagMem.RAM[00] is 08006f (din = 08006f ce=1 we=0)
[376545] NE  TOP.testbench.dut.core.lsu.bus.dcache.dcache.CacheWays[0].CacheTagMem.RAM[00] is 08006d (din = 08006f ce=1 we=0)
              376550: CacheWay[2] RAM[0] changed even though we was pulsed on CacheWays[0] rather than [2]
```

goekce, 2024-06-17:
Replying to https://github.com/verilator/verilator/issues/5177#issuecomment-2173413638: No David, you are not generating any fst files, so no concern.

wsnyder, 2024-06-17:
As long as all events are scheduled with blocking assignment from the same clock (regardless of posedge/negedge), there should be no races.  As a hybrid scheduler Verilator doesn't (usually) have delta cycles.

I find it suspicious though you are turning off UNOPTFLAT on this signal - doesn't this feed the RAMs?  Can you try to not use UNOPTFLAT there perhaps by using split_var?  While Verilator should schedule ok even with this disabled it certainly indicates likely edge cases, all the more so combined with --timing (part of --binary).

```
  /*verilator lint_off UNOPTFLAT*/
  logic [NUMWAYS-2:0]                  LRUUpdate;

```

davidharrishmc, 2024-06-18:
Replacing the UNOPTFLAT with split_var in cacheLRU.sv also solves the lint error but doesn't change the failure.

  logic [LOGNUMWAYS-1:0] Intermediate [NUMWAYS-2:0] /* verilator split_var */ ;

davidharrishmc, 2024-06-18:
Interestingly, adding #1 to the memory in ram1p1rwe.sv doesn't solve the issue either, but changes the failing test vector from 1020 to 956.  I can't think of any reason that should happen. This is really strange.

  always @(posedge clk) 
    if (ce & we) begin
      RAM[addr] <= #1 din;
    end

wsnyder, 2024-06-18:
Try a negedge flop to feeding in din/addr/ce/we.

davidharrishmc, 2024-06-18:
Curiously, this moves the failure vector from 1020 to 956. Changing adrregd to use addrn instead of addr brought the failing vector back to 1020. A similar modification moved the failure to 1022. This makes no sense.

```
logic wen, cen;
logic [$clog2(DEPTH)-1:0] addrn;
logic [WIDTH-1:0] dinn;

always @(negedge clk) begin
wen <= we;
cen <= ce;
addrn <= addr;
dinn <= din;
end

always @(posedge clk)
if (cen & wen)
RAM[addrn] <= dinn; 
```

davidharrishmc, 2024-06-18:
The problem does seem to stem from cacheLRU.sv.  Hardwiring VictimWay or CacheLRU  makes the sim pass.  I'll look for clues here.

davidharrishmc, 2024-06-18:
I found the root cause.  We used blocking instead of nonblocking assignments for LRUMemory because Verilator 5.021 lint was complaining about the array initialization. 

The problem code was 
```

  // note: Verilator lint doesn't like <= for array initialization (https://verilator.org/warn/BLKLOOPINIT?v=5.021)
  // Move to = to keep Verilator happy and simulator running fast
  always_ff @(posedge clk) begin
    if (reset | (InvalidateCache & ~FlushStage)) 
      for (int set = 0; set < NUMSETS; set++) LRUMemory[set] = '0; // exclusion-tag: initialize
    else if(CacheEn) begin
      // Because we are using blocking assignments, change to LRUMemory must occur after LRUMemory is used so we get the proper value
      if(LRUWriteEn & (PAdr == CacheSetTag)) CurrLRU = NextLRU;
      else                                   CurrLRU = LRUMemory[CacheSetTag];
      if(LRUWriteEn)                         LRUMemory[PAdr] = NextLRU;
    end
  end
```

It appears that Verilator no longer generates lint problems with nonblocking assignments. 

Changing the = to <= in the four places in this code block fixed the simulation.  

Perhaps in the blocking form, Verilator misscheduled CurrLRU to change after LRUMemory?

My issue is resolved to the extent that I can make forward progress simulating in Verilator.  I'm not sure if you want to close the issue or leave it open because the Verilator bug is still here.

wsnyder, 2024-06-18:
Verilator now supports the nonblocking assignments, and unfortunately for you "correctly" interprets the blocking also, so this means the code indicated I think can legally by IEEE either race or not.  Verilator was making different optimization decisions when the usage changed, so the failure moved.

We should add a new lint warning on blocking variable assignment on LRUMemory, all the more so with the always_ff.

davidharrishmc, 2024-06-18:
I thought blocking assignments cannot race.  In any event, a lint warning sounds good.

wsnyder, 2024-06-18:
Actually you didn't have "-Wall" - sorry I forgot to recommend that earlier I usually do that first thing. This is correctly caught already:

```
%Warning-BLKSEQ: inputs.v:5219:62: Blocking assignment '=' in sequential logic process
                                 : ... Suggest using delayed assignment '<='
 5219 |       for (int set = 0; set < NUMSETS; set++) LRUMemory[set] = '0;
      |                                                              ^

```

closed 2024-06-18 by wsnyder