verilator/verilator #5182: Generating getters/setters for ports

rzig, 2024-06-16:
Given some Verilog code like
```
module mymodule(input bit a, output bit b);
    // do something with a to get b
endmodule
```
After `verilator -CFLAGS -fPIC mymodule.sv`, I'm able to get a `Vmymodule` C++ class that I can interact with, i.e. `dut->a = 0`. I'm trying to interact with this from Rust, and `autocxx` does not support setting member variables on non-POD classes ([reference](https://google.github.io/autocxx/cpp_types.html#pod-and-non-pod)), so it would be useful if Verilator could generate a `set_a`/`get_b` method, since I could call that from Rust. 

Right now I'm hacking around it by redefining the macros `VL_IN8` and `VL_OUT8` and etc, but it would be great to not have to do this. Ideally, this would be something I could enable with a flag when I run `verilator`.

I'm happy to try and implement this myself if it is something that would be a good idea to include into Verilator.

wsnyder, 2024-06-17:
Yes, please contribute a pull which adds a new flag to add these accessors.  I'd prefer to follow Verilator naming where e.g. for signal sig we have:

```
     VL_IN8  __Vm_sig;
     void sig(const Vl_IN8& value) { __Vm_sig = value; }
     VI_IN8& sig() const { return __Vm_sig; }
     const VI_IN8& sig() const { return __Vm_sig; }
```

I think we should use references to avoid extra copies when possible.

wsnyder, 2024-06-17:
BTW the __Vm_sig renaming should be done in V3Name.cpp.

rzig, 2024-06-17:
Cool, thanks, will get working on this. Is there any reason for having both the `const VL_in8&` and `VL_IN8&` overloads? Is the idea that there would no longer be a `VL_IN8 &a;` member?

wsnyder, 2024-06-17:
I had a typo should be:

     VI_IN8& sig() { return __Vm_sig; }
     const VI_IN8& sig() const { return __Vm_sig; }
     
That is ok to get constant information on constant pointer passed in.

rzig, 2024-06-17:
Thanks, and is this in addition to a member variable called `sig`?

wsnyder, 2024-06-17:
Member for Verilog "sig" would be __Vm_sig so as not to conflict with "sig()".  I'd think with this option __Vm_sig should be moved to "private:".

rzig, 2024-06-17:
I see, so when the generate getters/setters flag is on, then using `model->sig = 3` would not work correct?

wsnyder, 2024-06-17:
Correct, I presume that works for your purpose.

rzig, 2024-06-17:
Yep that works, just wanted to clarify my understanding. Will get started on this, thanks for answering all my questions :grinning:

wsnyder, 2024-06-17:
Great, BTW separately would welcome a pull to docs/guide, or just general web article you can to point to from an issue here, to describe how to use with rust.

rzig, 2024-06-17:
Have realized that bindgen will call these methods sig() and sig1() when generating rust bindings to the verilator module because it doesn’t handle overloads… I think this is probably fine, maybe there is a case for an option to rename the getters/setters? What are your thoughts?

wsnyder, 2024-06-17:
For my reference docs on the Rust naming of overloads: https://google.github.io/autocxx/cpp_functions.html#overloads---and-identifiers-ending-in-digits

It's not stated but I presume the order of the overloads determines the number suffix.

I'd prefer to follow the existing Verilator convention and so settle for the "1" suffixes - which seems rust's current choice.  Note the VerilatedContext will also need importing to rust and already also needs overloads/the "1" suffixes.

It might be we're inconsistent about ordering of the accessors, please make a pre-pull for VerilatedContext that puts all of the get methods first, sets second, so then all of the setters will be "1" suffix (a easy mnemonic will be the setters take ONE argument).  For extra credit make a new t_dist test that has some hacky read-source checks to enforce this ordering, otherwise we'll botch it again at some point.

wsnyder, 2024-06-17:
Definitely is currently inconsistent:

```
    void randSeed(int val) VL_MT_SAFE;
    int randSeed() const VL_MT_SAFE { return m_s.m_randSeed; }
    ...
    inline uint64_t time() const VL_MT_SAFE;
    void time(uint64_t value) VL_MT_SAFE { m_s.m_time = value; }
```

rzig, 2024-06-22:
Question regarding renaming in V3Name - it seems like once I rename `sig` to `__Vm_sig` in here, then there is no way to recover the original name without string slicing hacks. This led me down the path of thinking maybe I need to add a member to `AstVar` to store this, but at that point it seems easier to just add something like a `privateName` method to AST var. It's then straightforward to deal with codegen - change `emitVarDecl` to `emitPublicVarDecl` which either generates the existing decl or getters/setters based on options, and add `emitPrivateVarDecl` which (if option enabled) emits the private declarations.

What are your thoughts on this? Is there something easier that I missed?

wsnyder, 2024-06-22:
The name unfortunately may get used for other purposes later, so I'd prefer not to have (another - as there's an origName) member for this.  I'd suggest simply asserting there's the _Vm and stripping it.

rzig, 2024-07-01:
Ok, I've implemented this for the most part, but the verilator naming convention causes some issues. Since names beginning with `__` are technically implementation reserved, the parameter to the setter can't be called `__Vm_sig_a` when autocxx generates bindings. So, the following would not work:
```cpp
void a(VL_IN8(&__Vm_sig_a,0,0)) {this->__Vm_sig_a = __Vm_sig_a;}
```
The natural way to solve this would be to rename the setter parameter something else, although currently the only way to emit the type of a node is using the `VL_` macros via `emitVarDecl`. This leaves us with a few options:
- Extract the logic from `emitVarDecl` to compute the type (but this gets hairy fairly quickly because parens are required for cc but not sc, etc). Once we do this, it's easy to emit the type + a new parameter name
- Create a new dummy node in the method where I'm emitting getters/setters (this also becomes a bit tricky when considering what needs to be copied, seems like a lot of overhead). The nummy node would be used to emit the parameter
- Just make the `__Vm_sig_` variables public, and use `decltype(__Vm_sig_a)` as the parameter type, i.e. `void a(decltype(__Vm_sig_a) newA) {...}`. This seems the simplest, and would require the least modification to existing code. I'm not a *huge* fan of the `__Vm_sig_` being public, but given they're dunder prefixed I think it's probably fine?

What are your thoughts on this? There may very well be a simpler idea I'm missing :smile:

wsnyder, 2024-07-02:
>Just make the __Vm_sig_ variables public

I think that's ok and probably needed anyways as DPI exports and other generated code might need to access those members from other classes anyways.

closed 2024-07-06 by wsnyder