verilator/verilator #5183: Possibility to waive rules with source code matcher (spyglass integration)

kaddkaka, 2024-06-17:
Verilator has great linting support and several ways to configure what rules to ignore (lint_off waiver), see https://veripool.org/guide/latest/exe_verilator.html#cmdoption-arg-lint_off

We already have a lot of inline comments to waive errors in the commercial lint tool `spyglass`, which means the source code might look like:

```verilog
module mem #(
   parameter integer DATA_WIDTH = 16  // spyglass disable W175
)
(
   wire in,
   wire out
);
   assign out = in;
endmodule
```

It would make it easier to start using verilator as lint tool if it support some configuration to waive rules depending on source code match, something like

```
`verilator_config
lint_off -rule UNUSEDPARAM -source-match "spyglass disable W175"
```

Verilator can already ignore based on line numbers, but I haven't seen anything like above. It would be a great feature.

If it's already possible but I just missed it, I apologize.

wsnyder, 2024-06-17:
FWIW at my site we do

  `parameter integer DATA_WIDTH = 16  // company-disable W175`

and run a preprocessor (or verilator pipe filter script) when we do lint using any tool, then we can completely control what messages are waived, and also deal with cases where vendors change the messages between versions.

Anyhow this is an interesting idea.  The main logistical problem is the preprocessor will remove all comments, so the regexp won't be able to fire.

What about writing your own pipe filter to look for these messages and insert lint_off/lint_on around those lines?

kaddkaka, 2024-06-18:
> FWIW at my site we do
>
> `parameter integer DATA_WIDTH = 16 // company-disable W175`
>
> and run a preprocessor (or verilator pipe filter script) when we do lint using any tool, then we can completely [...]

Your flow sounds interesting, could you explain it a bit more? I'm not sure I understand. 

I think we might be doing something similar, now that we found out how to use the `-lines` option in the lint_off config file:

1. Scan for comments
2. Generate a `.vlt` file
   ```
   lint_off -rule UNUSEDPARAM -lines 2
   ```
3. lint the original code with this config file. 

> Anyhow this is an interesting idea. The main logistical problem is the preprocessor will remove all comments, so the regexp won't be able to fire.
> 

Oh, when verilator loads the code, an early step is the preprocessor which drops the comments? That does sound like a problem. I see. 

> What about writing your own pipe filter to look for these messages and insert lint_off/lint_on around those lines?
> 

Actually modifying the files before sending them to the lint seems a bit intrusive, risk of linting the wrong stuff if there's a mishap, and verilator would report line numbers that are off wrt the source code, right?

wsnyder, 2024-06-18:
Yes, your method is similar.  The complication with your method is knowing what source files to read, e.g. how to follow submodules which are not easy to determine (and includes which are easy to find).

The pipe filter would insert `line directives, so is not noticeable to the user.

kaddkaka, 2024-06-18:
Some stupid questions:

- What is a "verilator pipe filter script"? 
- Where/when do you apply this filter to add `line directives?
- What are line directives?

wsnyder, 2024-06-19:
>What is a "verilator pipe filter script"?
>Where/when do you apply this filter to add `line directives?

See https://veripool.org/guide/latest/exe_verilator.html#cmdoption-pipe-filter
and https://github.com/verilator/verilator/blob/master/test_regress/t/t_pipe_filter.pf

>What are line directives?

See the IEEE standard for "`line".

wsnyder, 2024-09-08:
Seems resolution is to use filter as described.

closed 2024-09-08 by wsnyder