verilator/verilator #5196: Can't find modules with very long names

Timmmm, 2024-06-21:
Test case:

```
// main.sv
module top();
  module_with_a_very_long_name_this_is_autogenerated_so_its_not_unreasonable_and_it_really_has_to_be_quite_long_to_trigger_this_bug u_thing();
  always begin
    assert(u_thing.foo == 1'b1);
  end
endmodule;
```

```
// module_with_a_very_long_name_this_is_autogenerated_so_its_not_unreasonable_and_it_really_has_to_be_quite_long_to_trigger_this_bug.sv

module module_with_a_very_long_name_this_is_autogenerated_so_its_not_unreasonable_and_it_really_has_to_be_quite_long_to_trigger_this_bug();
  logic foo;
endmodule;
```

Error:

```
❯ verilator --cc main.sv
%Error: main.sv:4:12: Dotted reference to instance that refers to missing module/interface: 'module_with_a_very_long_name___VhshBwuAn6bFacuI8XALZEoiBChJGz3PFYqNsH2YEUjb'
    4 |     assert(u_thing.foo == 1'b1);
      |            ^~~~~~~
%Error: main.sv:4:20: Can't find definition of 'foo' in dotted variable: 'u_thing.foo'
    4 |     assert(u_thing.foo == 1'b1);
      |                    ^~~
%Error: Exiting due to 2 error(s)
```

If you use a shorter name then it works fine.

Version (`master` from today):

```
❯ verilator --version
 rev v5.026-4-g833df6d9e
```

wsnyder, 2024-06-30:
Something in the long name (Vhsh) handling messes up dot resolution.

Might you look into a pull request to test and fix this?

Timmmm, 2024-06-30:
Yeah sure if you can give me some hints where to look?

wsnyder, 2024-07-02:
If you run with --debug I suspect you'll see it is failing in V3LinkDot, so then use `--debugi-V3LinkDot` and trace it back, and probably something in one of the name resolution calls that V3LinkDot is using (maybe AstNode::dedotNode?) doesn't properly understand how the hashed names (those with __Vhsh in them as made by V3String) are formed/resolved.

Timmmm, 2024-07-03:
Thanks, great logging capabilities! So the problem is `V3Options::filePath()` gets given the "hashed" string (with `__Vhsh` in it) and it doesn't dehash it.

If I add dehash calls like this:

```
string V3Options::filePath(FileLine* fl, const string& modname, const string& lastpath,
                           const string& errmsg) {  // Error prefix or "" to suppress error
    // Find a filename to read the specified module name,
    // using the incdir and libext's.
    // Return "" if not found.
    const string filename = V3Os::filenameCleanup(modname);
    if (!V3Os::filenameIsRel(filename)) {
        // filename is an absolute path, so can find getStdPackagePath()
        string exists = filePathCheckOneDir(filename, "");
        if (exists != "") return exists;
    }
    for (const string& dir : m_impp->m_incDirUsers) {
        string exists = filePathCheckOneDir(VName::dehash(filename), dir);   // <----
        if (exists != "") return exists;
    }
    for (const string& dir : m_impp->m_incDirFallbacks) {
        string exists = filePathCheckOneDir(VName::dehash(filename), dir);  // <-----
        if (exists != "") return exists;
    }
```

then it works fine. I dunno if that's really correct though. Maybe it should go here instead:

```
    const string filename = V3Os::filenameCleanup(VName::dehash(modname));
```

That also works. But I can't say if either will always work.

In general there seems to be quite a confusing conflation of filenames and module names, and also the lack of distinct types for hashed and unhashed strings doesn't help.

I looked into [the original commit](https://github.com/verilator/verilator/commit/d0653f72e2d812b892b11e144a7809899c99b819) that added hashing and based on [the issue it was added to fix](https://github.com/verilator/verilator/issues/937) it seems like hashing should only be applied to *output* filenames, so I feel like my fix above is incorrect. Presumably the hashing should happen in some output stage rather than when the modules are being parsed?

wsnyder, 2024-07-05:
> const string filename = V3Os::filenameCleanup(VName::dehash(modname));

Seems right, can you make into a pull?

>In general there seems to be quite a confusing conflation of filenames and module names, and also the lack of distinct types for hashed and unhashed strings doesn't help.

Agreed/Sorry, one of those cleanup tasks hasn't been gotten to.

PR #5246 (2024-08-09): Fix shortened module names when searching for files (#5196)
Verilator hashes very long module names (>=128 characters) so that when it writes out the .cpp files they don't hit OS path/file length limits.

However that is transformation is done before the final output step, so we need to account for it when using module names. In this case, when auto-searching for a module named `foo` in a file called `foo.sv`. If `foo` is very long then `modname` in `filePath()` will be the shortened (hashed) version, but we actually want to search for the unshortened file name.

Fixes #5196
closed 2024-08-09 by wsnyder