verilator/verilator #5198: Unsupported: Modport clocking

solomatnikov, 2024-06-21:
This works with VCS on https://edaplayground.com/ but verilator errors out:

```
%Error-UNSUPPORTED: t/t_interface1_modport.v:35:28: Unsupported: Modport clocking
   35 |    modport  tb (input clk, clocking cb);
      |                            ^~~~~~~~
```

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

```
// DESCRIPTION: Verilator: Verilog Test module
//
// This file ONLY is placed under the Creative Commons Public Domain, for
// any use, without warranty, 2013 by Wilson Snyder.
// SPDX-License-Identifier: CC0-1.0

// Very simple test for interface pathclearing

interface ifc;
   integer hidden_from_isub;
   integer value;
   modport out_modport (output value);
endinterface

interface bus_if(input logic clk);

    // Define signals
    logic [31:0] addr;
    logic [31:0] data;
    logic write_enable;
    logic read_enable;
    logic [31:0] read_data;

    // Define a clocking block
    clocking cb @(posedge clk);
        default input #1step output #0;
        input read_data;
        output addr, data, write_enable, read_enable;
    endclocking

    // Define modports for different roles
    modport master (input read_data, output addr, data, write_enable, read_enable);
    modport slave (output read_data, input addr, data, write_enable, read_enable);

   modport  tb (input clk, clocking cb);

endinterface

module t (/*AUTOARG*/
   // Inputs
   clk
   );

   input clk;
   integer cyc=1;

   ifc itop();
   bus_if bus(clk);

   sub  c1 (.isub(itop),
            .i_value(4));

   always @ (posedge clk) begin
      cyc <= cyc + 1;
      if (cyc==20) begin
         if (itop.value != 4) $stop;
         itop.hidden_from_isub = 20;
         if (itop.hidden_from_isub != 20) $stop;
         $write("*-* All Finished *-*\n");
         $finish;
      end
   end
endmodule

module sub
`ifdef NANSI  // bug868
  (
   isub, i_value
   );
   ifc.out_modport isub;   // Note parenthesis are not legal here
   input integer i_value;
`else
  (
   ifc.out_modport isub,
   input integer i_value
   );
`endif

   always @* begin
      isub.value = i_value;
   end
endmodule
```

What 'verilator' command line do we use to run your example?

```
verilator --prefix Vt_interface1_modport_trace ../obj_vlt/t_interface1_modport_trace/Vt_interface1_modport_trace__main.cpp --exe --make gmake --x-assign unique -cc -Mdir obj_vlt/t_interface1_modport_trace --fdedup --debug-check --comp-limit-members 10 --trace --clk clk  -f input.vc +define+TEST_OBJ_DIR=obj_vlt/t_interface1_modport_trace +define+TEST_DUMPFILE=obj_vlt/t_interface1_modport_trace/simx.vcd t/t_interface1_modport.v
```

What 'verilator --version' are you using?  Did you try it with the git master version?

```
Verilator 5.027 devel rev v5.026-2-gd5cfe1a37
```

What OS and distribution are you using?

```
Linux 5.14.0-427.20.1.el9_4.x86_64
```

solomatnikov, 2024-06-22:
Original test case from:
https://www.asic-world.com/systemverilog/interface5.html

```
//+++++++++++++++++++++++++++++++++++++++++++++++++
// Define the interface
//+++++++++++++++++++++++++++++++++++++++++++++++++
interface mem_if (input wire clk);
  logic        reset;
  logic        we_sys;
  logic        cmd_valid_sys;
  logic        ready_sys;
  logic  [7:0] data_sys;
  logic  [7:0] addr_sys;
  logic        we_mem;
  logic        ce_mem;
  logic  [7:0] datao_mem;
  logic  [7:0] datai_mem;
  logic  [7:0] addr_mem;
  //=================================================
  // Modport for System interface 
  //=================================================
  modport  system (input clk,reset,we_sys, cmd_valid_sys,
                   addr_sys, datao_mem, 
                   output we_mem, ce_mem, addr_mem, 
                   datai_mem, ready_sys, inout data_sys);
  //=================================================
  // Modport for memory interface 
  //=================================================
  modport  memory (input clk,reset,we_mem, ce_mem,
                   addr_mem, datai_mem, output datao_mem);
  //=================================================
  // Clockin in Modport for testbench 
  //=================================================
  clocking cb @ (posedge clk);
    input  ready_sys; 
    output reset,we_sys, cmd_valid_sys, addr_sys;
    inout  data_sys;
  endclocking
  //=================================================
  // Modport for testbench 
  //=================================================
  modport  tb (input clk, clocking cb);

endinterface

//+++++++++++++++++++++++++++++++++++++++++++++++++
//  Memory Model 
//+++++++++++++++++++++++++++++++++++++++++++++++++
module memory_model (mem_if.memory mif);
// Memory array
logic [7:0] mem [0:255];

//=================================================
// Write Logic
//=================================================
always @ (posedge mif.clk)
 if (mif.ce_mem && mif.we_mem) begin
   mem[mif.addr_mem] <= mif.datai_mem;
 end

//=================================================
// Read Logic
//=================================================
always @ (posedge mif.clk)
 if (mif.ce_mem && ~mif.we_mem)  begin
   mif.datao_mem <= mem[mif.addr_mem];
 end

endmodule
[... 147 lines cut ...]
g++  -o ../simv      -rdynamic  -Wl,-rpath='$ORIGIN'/simv.daidir -Wl,-rpath=./simv.daidir -Wl,-rpath=/apps/vcsmx/vcs/U-2023.03-SP2/linux64/lib -L/apps/vcsmx/vcs/U-2023.03-SP2/linux64/lib  -Wl,-rpath-link=./   objs/amcQw_d.o   _331_archive_1.so   SIM_l.o       rmapats_mop.o rmapats.o rmar.o rmar_nd.o  rmar_llvm_0_1.o rmar_llvm_0_0.o            -lvirsim -lerrorinf -lsnpsmalloc -lvfs    -lvcsnew -lsimprofile -luclinative /apps/vcsmx/vcs/U-2023.03-SP2/linux64/lib/vcs_tls.o   -Wl,-whole-archive  -lvcsucli    -Wl,-no-whole-archive          /apps/vcsmx/vcs/U-2023.03-SP2/linux64/lib/vcs_save_restore_new.o -ldl  -lc -lm -lpthread -ldl 
../simv up to date
CPU time: .554 seconds to compile + .600 seconds to elab + .299 seconds to link
Chronologic VCS simulator copyright 1991-2023
Contains Synopsys proprietary information.
Compiler version U-2023.03-SP2_Full64; Runtime version U-2023.03-SP2_Full64;  Jun 21 19:53 2024
@170ns Writing address 0 with data 24
@270ns Writing address 1 with data 81
@370ns Writing address 2 with data 9
@470ns Writing address 3 with data 63
@790ns Reading address 0, Got data 24
@890ns Reading address 1, Got data 81
@990ns Reading address 2, Got data 9
@1090ns Reading address 3, Got data 63
$finish called from file "testbench.sv", line 173.
$finish at simulation time                 1100
```

Verilator:

```
%Error-UNSUPPORTED: t/t_interface1_modport.v:39:27: Unsupported: Modport clocking
   39 |   modport  tb (input clk, clocking cb);
      |                           ^~~~~~~~
```

wsnyder, 2024-06-22:
See #4555.

closed 2024-06-22 by wsnyder