verilator/verilator #5201: clang>12 coroutine test fail while ./configure

jcteng, 2024-06-25:
[image]

wsnyder, 2024-06-25:
What OS?  If Mac see #5190.

wsnyder, 2024-06-25:
If not mac please convert the screen shot to text, thanks.

jcteng, 2024-06-25:
lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.4 LTS
Release:        22.04
Codename:       jammy

`./configure CC=clang-17 CXX=clang++-17
configuring for Verilator 5.027 devel
checking whether to perform partial static linking of Verilator binary... yes
checking whether to use tcmalloc... check
checking whether to build for coverage collection... no
checking whether to use hardcoded paths... yes
checking whether to show and stop on compilation warnings... no
checking whether to run long tests... no
checking for z3... yes
checking for cvc5... no
checking for cvc4... no
checking for SMT solver... z3 --in
checking for gcc... clang-17
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether clang-17 accepts -g... yes
checking for clang-17 option to enable C11 features... none needed
checking whether the compiler supports GNU C++... yes
checking whether clang++-17 accepts -g... yes
checking for clang++-17 option to enable C++11 features... none needed
checking for a BSD-compatible install... /usr/bin/install -c
compiler is clang++-17 --version = Ubuntu clang version 17.0.6 (++20231209124227+6009708b4367-1~exp1~20231209124336.77)
checking that C++ compiler can compile simple program... yes
checking for ar... ar
checking for perl... /usr/bin/perl
checking for python3... /usr/bin/python3
checking for flex... /usr/bin/flex
/usr/bin/flex --version = flex 2.6.4
checking for bison... /usr/bin/bison
/usr/bin/bison --version = bison (GNU Bison) 3.8.2
checking for ccache... ccache
objcache is ccache --version = ccache version 4.5.1
checking for stdio.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for strings.h... yes
checking for sys/stat.h... yes
checking for sys/types.h... yes
checking for unistd.h... yes
checking for size_t... yes
checking for size_t... (cached) yes
checking for inline... inline
checking whether clang++-17 accepts -pg... yes
checking whether clang++-17 accepts -std=gnu++17... yes
checking whether clang++-17 accepts -Wextra... yes
checking whether clang++-17 accepts -Wfloat-conversion... yes
checking whether clang++-17 accepts -Wlogical-op... no
checking whether clang++-17 accepts -Wthread-safety... yes
checking whether clang++-17 accepts -fcoroutines-ts... no
checking whether clang++-17 accepts -fcoroutines... no
checking whether clang++-17 accepts -fcoroutines-ts -Wno-deprecated-experimental-coroutine... no
checking whether coroutines are supported by clang++-17... no
checking whether clang++-17 accepts -Qunused-arguments... yes
checking whether clang++-17 accepts -faligned-new... yes
checking whether clang++-17 accepts -Wno-unused-parameter... yes
checking whether clang++-17 accepts -Wno-shadow... yes
checking whether clang++-17 accepts -Wno-char-subscripts... yes
[... 36 lines cut ...]
checking whether clang++-17 linker accepts -latomic... yes
checking whether clang++-17 linker accepts -lbcrypt... no
checking whether clang++-17 linker accepts -lpsapi... no
checking whether clang++-17 linker accepts -l:libtcmalloc_minimal.a... yes
checking whether clang++-17 accepts -fno-builtin-malloc... yes
checking whether clang++-17 accepts -fno-builtin-calloc... yes
checking whether clang++-17 accepts -fno-builtin-realloc... yes
checking whether clang++-17 accepts -fno-builtin-free... yes
checking whether clang++-17 supports C++14... yes
checking for clang++-17 precompile header include option... -include-pch
checking for struct stat.st_mtim.tv_nsec... yes
checking whether SystemC is found (in system path)... yes
configure: creating ./config.status
config.status: creating Makefile
config.status: creating src/Makefile
config.status: creating src/Makefile_obj
config.status: creating include/verilated.mk
config.status: creating include/verilated_config.h
config.status: creating verilator.pc
config.status: creating verilator-config.cmake
config.status: creating verilator-config-version.cmake
config.status: creating src/config_package.h

Now type 'make' (or sometimes 'gmake') to build Verilator.
`

wsnyder, 2024-06-25:
I checked and this works on the git master, I think due to #5200 being fixed.

closed 2024-06-25 by wsnyder