verilator/verilator #5203: waiting on virtual interface clocking causes Internal Error: ../V3Width.cpp:2818: MemberSel of non-variable

solomatnikov, 2024-06-26:
```
%Error: Internal Error: bfm.sv:33:16: ../V3Width.cpp:2818: MemberSel of non-variable
                                                               : ... note: In instance 'top'
   33 |          @(vif.cb);
      |                ^~
                        driver_if.sv:101:13: ... Location of found object
  101 |    clocking cb @(posedge clk);
      |             ^~
```

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

What 'verilator' command line do we use to run your example?

```
time verilator  --cc --exe -sv --assert --trace --trace-params --trace-structs --report-unoptflat --no-timing --timescale 1ps/1ps --x-assign unique --Mdir verilator_obj_dir --top-module top --timing -F src_files.f top.v
```

What 'verilator --version' are you using?  Did you try it with the git master version?

```
v5.026-gd5cfe1a37
```

What OS and distribution are you using?
```
Linux 5.14.0-427.20.1.el9_4.x86_64
```

solomatnikov, 2024-06-26:
Test case:

```
// DESCRIPTION: Verilator: Verilog Test module
//
// This file ONLY is placed under the Creative Commons Public Domain, for
// any use, without warranty, 2023 by Antmicro Ltd.
// SPDX-License-Identifier: CC0-1.0

interface Bus (input logic clk);
  logic [15:0] data;
   default clocking cb @(posedge clk);
      default input #1step output #0;
      output data;
   endclocking // cb
endinterface

module t (
    clk
);
  input clk;
  Bus intf1(clk), intf2(clk);
  virtual Bus vif1 = intf1, vif2 = intf2;

  task assign_to_vif2();
    if ($c("0")) return;
    #1 vif2.data = 'hfafa; #1;
  endtask

  initial forever begin
    intf1.data = 'hdead;
    if ($c("1")) begin
      #1 vif2.data = 'hbeef; #1;
    end
    intf1.data = 'hcafe;
    if ($c("0")); else begin
      #1 vif2.data = 'hface; #1;
    end
    intf1.data = 'hfeed;
    while ($time < 5) begin
      #1 vif2.data = 'hdeed; #1;
    end
    intf1.data = 'hdeaf;
    assign_to_vif2;
    intf1.data = 'hbebe;
     @(vif1.cb);
     @(vif2.cb);
    #1 $write("*-* All Finished *-*\n");
    $finish;
  end

  always_comb if ($time < 9) $write("vif1.data==%h\n", vif1.data);
  always_comb if ($time < 9) $write("intf2.data==%h\n", intf2.data);
endmodule
```

```
%Error: Internal Error: t/t_interface_virtual_timing.v:43:13: ../V3Width.cpp:2818: MemberSel of non-variable
                                                            : ... note: In instance 't'
   43 |      @(vif1.cb);
      |             ^~
                        t/t_interface_virtual_timing.v:9:21: ... Location of found object
    9 |    default clocking cb @(posedge clk);
      |                     ^~
                        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
%Warning: vlt/t_interface_virtual_timing: Exec of perl failed: %Error: Internal Error: t/t_interface_virtual_timing.v:43:13: ../V3Width.cpp:2818: MemberSel of non-variable
```

solomatnikov, 2024-06-26:
Another test case:

```
// DESCRIPTION: Verilator: Verilog Test module
//
// This file ONLY is placed under the Creative Commons Public Domain, for
// any use, without warranty, 2023 by Antmicro Ltd.
// SPDX-License-Identifier: CC0-1.0

interface Bus (input logic clk);
  logic [15:0] data;
   default clocking cb @(posedge clk);
      default input #1step output #0;
      output data;
   endclocking // cb
endinterface

module t (
    clk
);
  input clk;
  Bus intf1(clk), intf2(clk);
  virtual Bus vif1 = intf1, vif2 = intf2;

  task assign_to_vif2();
    if ($c("0")) return;
    #1 vif2.data = 'hfafa; #1;
  endtask

  initial forever begin
    intf1.data = 'hdead;
    if ($c("1")) begin
      #1 vif2.data = 'hbeef; #1;
    end
    intf1.data = 'hcafe;
    if ($c("0")); else begin
      #1 vif2.data = 'hface; #1;
    end
    intf1.data = 'hfeed;
    while ($time < 5) begin
      #1 vif2.data = 'hdeed; #1;
    end
    intf1.data = 'hdeaf;
    assign_to_vif2;
    intf1.data = 'hbebe;
    vif1.cb.data <= '0;
    vif2.cb.data <= '1;
    #1 $write("*-* All Finished *-*\n");
    $finish;
  end

  always_comb if ($time < 9) $write("vif1.data==%h\n", vif1.data);
  always_comb if ($time < 9) $write("intf2.data==%h\n", intf2.data);
endmodule
```

```
%Error: Internal Error: t/t_interface_virtual_timing.v:43:10: ../V3Width.cpp:2818: MemberSel of non-variable
                                                            : ... note: In instance 't'
   43 |     vif1.cb.data <= '0;
      |          ^~
                        t/t_interface_virtual_timing.v:9:21: ... Location of found object
    9 |    default clocking cb @(posedge clk);
      |                     ^~
                        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
%Warning: vlt/t_interface_virtual_timing: Exec of perl failed: %Error: Internal Error: t/t_interface_virtual_timing.v:43:10: ../V3Width.cpp:2818: MemberSel of non-variable
```

solomatnikov, 2024-06-28:
Another test case:

```
// DESCRIPTION: Verilator: Verilog Test module
//
// This file ONLY is placed under the Creative Commons Public Domain, for
// any use, without warranty, 2022 by Arkadiusz Kozdra.
// SPDX-License-Identifier: CC0-1.0

// See also t_interface_virtual_bad.v

interface PBus;
   logic req, grant;
   logic [7:0] addr, data;
   modport phy(input addr, ref data);
endinterface

typedef virtual PBus vpbus_t;
typedef vpbus_t vpbus2_t;

class Cls;
   vpbus2_t fa, fb;

   function logic [7:0] sample(int intf, int sig);
      sample = fa.phy.addr;
   endfunction // sample
endclass

class Clsgen#(type T = logic);
   T x[0:3];
endclass

module t (/*AUTOARG*/);

   PBus ia(), ib();
   virtual PBus va, vb;
   virtual PBus.phy pa, pb;
   Cls ca, cb;
   Clsgen#(virtual PBus) gen;

   initial begin
      va = ia;
      vb = ia;
      $display("va==vb? %b", va==vb);
      $display("va!=vb? %b", va!=vb);
      vb = ib;
      $display("va==vb? %b", va==vb);
      $display("va!=vb? %b", va!=vb);

      ca = new;
      cb = new;
      gen = new;

      va.addr = 8'haa;
      ia.data = 8'h11;

      vb.addr = 8'hbb;
      ib.data = 8'h22;

      $display("va.addr=%x", va.addr, " va.data=%x", va.data, " ia.addr=%x", ia.addr, " ia.data=%x", ia.data);
      $display("vb.addr=%x", vb.addr, " vb.data=%x", vb.data, " ib.addr=%x", ib.addr, " ib.data=%x", ib.data);

      ca.fa = ia;
      ca.fb = ib;
      cb.fa = ib;
      cb.fb = ia;
      gen.x[0] = va;
      gen.x[1] = vb;

      pa = va;
      pb = vb;

      pb.addr = 8'hb0;
      pa.addr = 8'ha0;

      $display("ca.fa.addr=%x", ca.fa.addr, " ca.fa.data=%x", ca.fa.data, " ca.fa.addr=%x", ca.fb.addr, " ca.fb.data=%x", ca.fb.data);
      $display("cb.fa.addr=%x", cb.fa.addr, " cb.fa.data=%x", cb.fa.data, " cb.fa.addr=%x", cb.fb.addr, " cb.fb.data=%x", cb.fb.data);
      $display("gen.x[0].addr=%x", gen.x[0].addr, " gen.x[1].addr=%x", gen.x[1].addr);
      $display("gen=%p", gen);

      $write("*-* All Finished *-*\n");
      $finish;
   end
endmodule
```

```
%Error: Internal Error: t/t_interface_virtual.v:22:19: ../V3Width.cpp:2818: MemberSel of non-variable
                                                     : ... note: In instance 't'
   22 |       sample = fa.phy.addr;
      |                   ^~~
                        t/t_interface_virtual.v:12:12: ... Location of found object
   12 |    modport phy(input addr, ref data);
      |            ^~~
```

kozdra, 2024-07-09:
Note that your last test case is not failing because of clocking blocks, but because of accessing virtual interface variables through modports.
A workaround would be to access them directly (w/o modports), but you can submit a PR for that as well if you want to help (basic support would involve replacing the deeper AstMemberSel with its fromp in V3Width, but full support would need to consider whether the variable is really present in modport argument list; inspiration for that for regular interfaces can be taken from V3LinkDot).

closed 2024-07-09 by wsnyder

solomatnikov, 2024-07-10:
> Note that your last test case is not failing because of clocking blocks, but because of accessing virtual interface variables through modports.

It doesn't fail after 2cfec0ecc39e5adbbd512632c8788d3f2eb70e16