verilator/verilator #5206: Help - unable to get $value$plusargs working

ajeethakv, 2024-06-28:
While trying to bring up some UVC on Verilator, I noticed plusargs do not seem to work as expected. Perhaps I am missing something very simple, please clarify.

```verilog
module test;
integer i, r;

initial begin
  r = $value$plusargs("myint=%d", i);
  $display("Value is %0d", i);
  #100;
  r = $value$plusargs("myint=%d", i);
  $display("%t Value is %0d", $time, i);
  $finish(2);
end

endmodule

```
Ran with another opensource sim, got expected results. Below are sample logs:

```log
./a.out 
Value is x
                 100 Value is x
test.sv:11: $finish(2) called at 100 (1s)
[ajeethakv@localhost tmp]$ ./a.out +myint=11
Value is 11
                 100 Value is 11
test.sv:11: $finish(2) called at 100 (1s)
```
Verilaotr log below:
```log
[ajeethakv@localhost tmp]$ ./obj_dir/Vtest +myint=11
Value is 0
                 100 Value is 0
- S i m u l a t i o n   R e p o r t: Verilator 5.027 devel
- Verilator: end at 100ps; walltime 0.000 s; speed 0.000 s/s
- Verilator: cpu 0.000 s on 1 threads; alloced 25 MB

```

What 'verilator' command line do we use to run your example?

> verilator --binary top.sv

What 'verilator --version' are you using?  Did you try it with the git master version?

> Verilator 5.027 devel

What OS and distribution are you using?
> Linux CentOS7

May we assist you in trying to fix this in Verilator yourself?

I believe it is more of usemodel as it should be working. Thanks

wsnyder, 2024-06-30:
You can try using "r" in your code later and it might work.  I thought we had just fixed a similar misoptimization earlier but needs more work I guess.

ajeethakv, 2024-06-30:
Thanks for the clue, below is what I tried just now and it seems to work.

```verilog
module test;
integer i, r;

initial begin
  `ifdef VERILATOR
  $value$plusargs("myint=%d", i);
  $display("Value is %0d", i);
  #100;
  r = $value$plusargs("myint=%d", i);
  $display("%t Value is %0d", $time, i);
  `else
  r = $value$plusargs("myint=%d", i);
  $display("Value is %0d", i);
  #100;
  r = $value$plusargs("myint=%d", i);
  $display("%t Value is %0d", $time, i);
  `endif 

  $finish(2);
end

endmodule
```
Log:
```log
./obj_dir/Vtest +myint=8
Value is 8
                 100 Value is 8
- test.sv:20: Verilog $finish

```
Is this misoptimization something that is easy for a newbie (to Verilator) attempt to delve into? 

Thanks

ajeethakv, 2024-06-30:
Maybe this one

https://github.com/verilator/verilator/pull/5137

?

ajeethakv, 2024-06-30:
Quick update - I picked the change done by @sethp from https://github.com/verilator/verilator/pull/5137 and re-built Verilator, ran it and the test works without any hack (meaning my original test). 

FWIW - I observed this from a UVM run, so will now try the UVM_VERBOSITY stuff, thanks a lot for your wonderful work here folks!

wsnyder, 2024-07-02:
See #5137, now fixed.

closed 2024-07-02 by wsnyder