verilator/verilator #5208: Internal Error: t/t_interface_class.v:29:9: ../V3Scope.cpp:73: Can't locate varref scope

solomatnikov, 2024-06-28:
```
%Error: Internal Error: t/t_interface_class.v:29:9: ../V3Scope.cpp:73: Can't locate varref scope
                                                  : ... note: In instance 'APB3_TB_top.apb3_intf.APB3_concrete_BFM'
   29 |         PENABLE <= '0;
      |         ^~~~~~~
```

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

From **"Abstract BFMs Outshine Virtual Interfaces for Advanced SystemVerilog Testbenches"**

```
// DESCRIPTION: Verilator: Verilog Test module
//
// This file ONLY is placed under the Creative Commons Public Domain, for
// any use, without warranty, 2024 by .
// SPDX-License-Identifier: CC0-1.0

package APB3_pkg_Fig4;
   virtual class APB3_BFM;
      pure virtual task write (int unsigned addr, data);
      pure virtual task read (int unsigned addr, output int unsigned data);
   endclass
endpackage

interface APB3_TB_intf(input bit PCLK);
   import APB3_pkg_Fig4::*;

   typedef logic [15:0] addr_t;
   typedef logic [15:0] data_t;

   logic PENABLE, PWRITE, PSEL, PREADY;
   addr_t PADDR;
   data_t PWDATA, PRDATA;

class APB3_concrete_BFM extends APB3_BFM;
   task write(int unsigned addr, data);
      @(posedge PCLK)
	PSEL <= '1;
        PENABLE <= '0;
        PWRITE <= '1;
        PADDR <= addr_t'(addr);
        PWDATA <= data_t'(data);
      @(posedge PCLK)
	PENABLE <= 1'b1;
      do @(posedge PCLK); while (!PREADY);
      PSEL <= 1'b0;
      PENABLE <= 1'b0 ;
   endtask

   task read (int unsigned addr, output int unsigned data);
      @(posedge PCLK)
	PSEL <= '1;
        PENABLE <= '0;
        PWRITE <= '0;
        PADDR <= addr_t'(addr);
      @(posedge PCLK)
	PENABLE <= 1'b1;
      do @(posedge PCLK); while (!PREADY);
      PSEL <= '0;
      PENABLE <= '0 ;
      data = int'(PRDATA);
   endtask
endclass

   APB3_concrete_BFM bfm = new;
endinterface

module APB3_TB_top;
   import APB3_pkg_Fig4::*;

   bit CLK;
   always #5 CLK = ~CLK; // clock generator

   APB3_TB_intf apb3_intf(CLK);
/* -----\/----- EXCLUDED -----\/-----
   APB3_device DUT(
		   .PCLK(CLK),
		   .PENABLE(apb3_intf.PENABLE),
		   ...);
 -----/\----- EXCLUDED -----/\----- */
   initial begin : Test_Activity
      APB3_BFM bfm; // abstract BFM reference
      int read_data;
      bfm = apb3_intf.bfm; // reference to BFM
      bfm.write(100, 1234);
      bfm.read(100, read_data);
      if (read_data != 1234)
	$display("error: unexpected read data");
   end
endmodule
```

What 'verilator' command line do we use to run your example?

```
verilator --prefix Vt_interface_class ../obj_vlt/t_interface_class/Vt_interface_class__main.cpp --exe --make gmake --x-assign unique -cc -Mdir obj_vlt/t_interface_class --fdedup --debug-check --comp-limit-members 10 --clk clk  -f input.vc +define+TEST_OBJ_DIR=obj_vlt/t_interface_class +define+TEST_DUMPFILE=obj_vlt/t_interface_class/simx.vcd --timing  t/t_interface_class.v
```

What 'verilator --version' are you using?  Did you try it with the git master version?
```
v5.026-gd5cfe1a37
```
What OS and distribution are you using?
```
Linux 5.14.0-427.20.1.el9_4.x86_64
```

wsnyder, 2024-06-30:
Probably similar fix as to whatever's needed for #5212 - this one might be easier to fix first though.

As always due to large backlog pull requests to fix this are welcome.

solomatnikov, 2024-07-02:
#5212 might hit this bug too after 2 other bugs are fixed

wsnyder, 2025-03-16:
Simplified test:

```
package Pkg;
   virtual class BFM;
      pure virtual task write (int unsigned data);
        pure virtual task read (output int unsigned data);
   endclass
endpackage

interface TB_intf(input bit PCLK);
   import Pkg::*;

   typedef logic [15:0] data_t;

   data_t PDATA;

   class concrete_BFM extends BFM;
      task write(int unsigned data);
         @(posedge PCLK);
         PDATA <= data_t'(data);
      endtask

      task read (output int unsigned data);
         @(posedge PCLK);
         data = int'(PDATA);
      endtask
   endclass

   concrete_BFM bfm = new;
endinterface
module t;
   import Pkg::*;

   bit clk;
   always #5 clk = ~clk;

   TB_intf apb3_intf(clk);

   initial begin
      BFM bfm; // abstract BFM reference
      int read_data;
      bfm = apb3_intf.bfm; // reference to BFM
      bfm.write(1234);
      bfm.read(read_data);
      if (read_data != 1234) $stop;
      $write("*-* All Finished *-*\n");
      $finish;
   end
endmodule
```