verilator/verilator #5210: Non-blocking assignments after `@(posedge clk);` delay don't behave as expected

cousteaulecommandant, 2024-06-28:
### Problem description:
I usually create my testbenches so that the signal generation section uses some idiom for "wait for the next clock rising edge" rather than waiting a fixed amount of time.  This makes the signals appear as if they were generated from a synchronous circuit.
In the case of [System]Verilog, I achieve this by adding a `@(posedge clk) ;` statement inside the `initial begin` block, and then assigning to a signal using `<=`.  The expected result is identical to doing that assignment inside the `@(posedge clk)`, but gives me more freedom to write readable testbenches.  This syntax may seem a bit strange, but it has worked in every other simulator I've tested, and I understand it is a valid use of Verilog.

However, it appears that Verilator treats this `<=` assignment as a blocking `=` assignment, assigning the value immediately, rather than "after the clock edge", and the result is not as expected.

### Example code:
```verilog
`timescale 1ns/100ps

module T_verilator_posedge;

reg clk = 1;
reg [7:0] d = 0;
reg [7:0] q = 0;

// Clock generation
always #0.5 clk = ~clk;

// Input signal generation
initial begin
    // Initialize VCD dump
    $dumpfile("T_verilator_posedge.vcd");
    $dumpvars(0, T_verilator_posedge);

    d <= 0;
    repeat (5) @(posedge clk);
    d <= 1;
    @(posedge clk);
    d <= 2;
    @(posedge clk);
    d <= 3;
    @(posedge clk);
    d <= 4;
    @(posedge clk);
    d <= 0;
    repeat (5) @(posedge clk);
    $finish;
end

// Unit under test (flip-flop)
always @(posedge clk)
    q <= d;

endmodule
```

Compiled with:
```sh
verilator -CFLAGS -fcoroutines --binary --trace -Wno-fatal T_verilator_posedge.v
```

### Expected result:
The `q` signal is delayed one clock cycle after `d`.  This is the case if I simulate the file with Icarus Verilog.
[image]

### Obtained result:
The `q` signal is simultaneous to the `d` signal.  This is the case if I simulate the file with Verilator 5.026.
[image]

### Verilator version:
Verilator 5.026, freshly compiled.
```
# verilator --version
Verilator 5.026 2024-06-15 rev v5.026
```

### Temporary workaround:
For the time being, I'm adding a `#0.1;` after each `@(posedge clk);`.  It works, but it's not a perfect solution.

Kreijstal, 2024-06-30:
looks like #4384 but I agree, this is a problem for verilator as a testbench :) afaik the author says he will accept a pull request, but he won't work on it

cousteaulecommandant, 2024-07-01:
I forgot to mention that Verilator is throwing a warning message that the `<=` is being treated as a `=` because it's in an `initial` block.
________

> looks like #4384

I think in that case there is a race condition, since there is a blocking assignment in one `initial` block, and one nonblocking assignment in the `always` block, so the value of `addr` being non-blockingly assigned to `addr_wtf` may be read before or after it was blockingly assigned to `addr` in the `initial` block.
But in my case there's only one block.

Kreijstal, 2024-07-01:
I get this error lots of times
```
%Warning-INITIALDLY: jtag_master.sv:35:21: Non-blocking assignment '<=' in initial/final block
                                         : ... note: In instance 'intf_tb.jtag'
                                         : ... This will be executed as a blocking assignment '='!
   35 |     ir_stored_valid <= '0;
      |                     ^~
                     intf_tb.sv:10:1: ... note: In file included from 'intf_tb.sv'
                     ... For warning description see https://verilator.org/warn/INITIALDLY?v=5.026
                     ... Use "/* verilator lint_off INITIALDLY */" and lint_on around source to disable this message.
%Warning-INITIALDLY: jtag_master.sv:36:21: Non-blocking assignment '<=' in initial/final block
                                         : ... note: In instance 'intf_tb.jtag'
                                         : ... This will be executed as a blocking assignment '='!
   36 |     ir_stored       <= '0;
      |                     ^~
                     intf_tb.sv:10:1: ... note: In file included from 'intf_tb.sv'

```

cousteaulecommandant, 2024-07-01:
Yes; that seems to be the issue.  Verilator is replacing `<=` with `=` in **`initial`** blocks, and only allows `<=` in **`always`** blocks.

Apparently, the workaround is as easy as **replacing `initial` with `always` in the code,** which will behave the same as long as there's a `$finish` at the end of the block:
```verilog
// Input signal generation
//initial begin // <== doesn't work in Verilator
always begin // <== WORKAROUND FOR VERILATOR
    // Initialize VCD dump
    $dumpfile("T_verilator_posedge.vcd");
    $dumpvars(0, T_verilator_posedge);

    d <= 0;
    repeat (5) @(posedge clk);
    d <= 1;
    @(posedge clk);
    d <= 2;
    @(posedge clk);
    d <= 3;
    @(posedge clk);
    d <= 4;
    @(posedge clk);
    d <= 0;
    repeat (5) @(posedge clk);
    $finish;
end
```

But I think this should be fixed for `initial` blocks as well.

Kreijstal, 2024-07-01:
that would solve for initial, but it is also broken for some tasks
```
task do_transaction();
  integer i;
  // Send request on A channel:
  @(posedge clk_i);
  tl_o.a_valid <= '1;
  $display("setting valid to 1 %t",$realtime);
  @(posedge clk_i);
  if(tl_i.d_valid=='1)
  begin
    $display("d_valid should be 0 not 1: %d",tl_i.d_valid);
    $stop;
  end
  i = '0;
  if(tl_i.a_ready) begin
    $display("this is true in xsim but not in verilator");
  end
 //...
 
 ```
 seems to have the same issue, but not sure if this is the same issue or a different one

kbieganski, 2024-07-02:
This is, in fact, due to Verilator treating `<=` as `=`, but it's not the flip flop. It's the `<=`s in the `initial` block. There's a warning for this:
```
%Warning-INITIALDLY: test.sv:18:7: Non-blocking assignment '<=' in initial/final block
```
You can test this by changing the `initial` to `always`. It'll execute once anyway because of the `$finish`, it won't complain because it's not an `initial`, and the resulting waveform will be correct.

Kreijstal, 2024-07-02:
> This is, in fact, due to Verilator treating `<=` as `=`, but it's not the flip flop. It's the `<=`s in the `initial` block. There's a warning for this:
>
> ```
> %Warning-INITIALDLY: test.sv:18:7: [...]

> This is, in fact, due to Verilator treating `<=` as `=`, but it's not the flip flop. It's the `<=`s in the `initial` block. There's a warning for this:
>
> ```
> %Warning-INITIALDLY: test.sv:18:7: [...]

that is only a workaround for initial, but if you use tasks, you still get the error

plase look a https://github.com/verilator/verilator/issues/4384#issuecomment-2171174001

Kreijstal, 2024-07-02:
> > This is, in fact, due to Verilator treating `<=` as `=`, but it's not the flip flop. It's the `<=`s in the `initial` block. There's a warning for this:
> > ```
> > %Warning-INITIALDLY: test.sv:18:7: Non-blocking assignment '<=' in initial/final block
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > You can test this by changing the `initial` to `always`. It'll execute once anyway because of the `$finish`, it won't complain because it's not an `initial`, and the resulting waveform will be correct.
> 
> > This is, in fact, due to Verilator treating `<=` as `=`, but it's not the flip flop. It's the `<=`s in the `initial` block. There's a warning for this:
> > ```
> > %Warning-INITIALDLY: test.sv:18:7: Non-blocking assignment '<=' in initial/final block
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > You can test this by changing the `initial` to `always`. It'll execute once anyway because of the `$finish`, it won't complain because it's not an `initial`, and the resulting waveform will be correct.
> 
> that is only a workaround for initial, but if you use tasks, you still get the error
> 
> plase look a [#4384 (comment)](https://github.com/verilator/verilator/issues/4384#issuecomment-2171174001)

> > This is, in fact, due to Verilator treating `<=` as `=`, but it's not the flip flop. It's the `<=`s in the `initial` block. There's a warning for this:
> > ```
> > %Warning-INITIALDLY: test.sv:18:7: Non-blocking assignment '<=' in initial/final block
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > You can test this by changing the `initial` to `always`. It'll execute once anyway because of the `$finish`, it won't complain because it's not an `initial`, and the resulting waveform will be correct.
> 
> > This is, in fact, due to Verilator treating `<=` as `=`, but it's not the flip flop. It's the `<=`s in the `initial` block. There's a warning for this:
> > ```
> > %Warning-INITIALDLY: test.sv:18:7: Non-blocking assignment '<=' in initial/final block
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > You can test this by changing the `initial` to `always`. It'll execute once anyway because of the `$finish`, it won't complain because it's not an `initial`, and the resulting waveform will be correct.
> 
> that is only a workaround for initial, but if you use tasks, you still get the error
> 
> plase look a [#4384 (comment)](https://github.com/verilator/verilator/issues/4384#issuecomment-2171174001)

I just notice the task gets called from an initial.... Hmm.

Kreijstal, 2024-07-02:
yes, I just tested it, changing the original initial to always fixed the bug...

Kreijstal, 2024-07-02:
so we can simply change the initial logic to the always one

cousteaulecommandant, 2024-07-03:
> I just notice the task gets called from an initial.... Hmm.

Now that you mention it, the first time I noticed this was on an *interface* task that was being called from `initial`; calling it from `always` instead solved the issue.  I guess it's the same for regular tasks.

For now, I guess the workaround of changing `initial` to `always` will do, but it would be nice if this were fixed for future versions, for compliance with the standard (which I re-read a lot of times to make sure I wasn't doing something illegal with that `<=` in an `initial` block).

Kreijstal, 2024-07-03:
> Now that you mention it, the first time I noticed this was on an _interface_ task that was being called from `initial`; calling it from `always` instead solved the issue. I guess it's the same for [...]

I agree, now we need to find someone that fixes it lol

kbieganski, 2024-07-04:
This patch:

```diff
diff --git i/src/V3Active.cpp w/src/V3Active.cpp
index 6c6070574..4ac69da3d 100644
--- i/src/V3Active.cpp
+++ w/src/V3Active.cpp
@@ -362,6 +362,7 @@ private:

         // Issue appropriate warning
         if (m_check == CT_INITIAL) {
+            if (nodep->fileline()->timingOn()) return;
             nodep->v3warn(INITIALDLY,
                           "Non-blocking assignment '<=' in initial/final block\n"
                               << nodep->warnMore()
```

partially fixes NBAs in initials for `--timing`. Works for this case.

However, it's not enough. For example, if we remove all timing controls from the `initial`, we'll get an internal error. So at the very least, that needs to be resolved. I'm busy with other stuff, so if someone wants to work on this, feel free.

RootCubed, 2024-07-04:
See also https://github.com/verilator/verilator/issues/5216#issuecomment-2205293418, the linked GitHub action run shows under which conditions things fail. It's not something I will try to fix with high priority as it works well enough for me for now, but I might take another look at some point.