verilator/verilator #5212: Verilator internal fault and incorrect error "Usage of cycle delays requires default clocking"

solomatnikov, 2024-06-28:
```
%Error: Verilator internal fault, sorry. Suggest trying --debug --gdbbt
%Error: Command Failed ulimit -s unlimited 2>/dev/null; exec .../bin/verilator_bin --prefix Vt_interface_clocking_class ../obj_vlt/t_interface_clocking_class/Vt_interface_clocking_class__main.cpp --exe --make gmake --x-assign unique -cc -Mdir obj_vlt/t_interface_clocking_class --fdedup --debug-check --comp-limit-members 10 --clk clk -f input.vc +define+TEST_OBJ_DIR=obj_vlt/t_interface_clocking_class +define+TEST_DUMPFILE=obj_vlt/t_interface_clocking_class/simx.vcd --timing t/t_interface_clocking_class.v
%Warning: vlt/t_interface_clocking_class: Exec of perl failed: %Error: Verilator internal fault, sorry. Suggest trying --debug --gdbbt
```

With `--debug` `--gdbbt`:

```
- V3Ast.cpp:1324:     Dumping obj_vlt/t_interface_clocking_class/Vt_interface_clocking_class_014_const.tree
- V3Undriven.cpp:552: undrivenAll: 
- V3AssertPre.cpp:507:assertPreAll: 
%Error: t/t_interface_clocking_class.v:97:7: Usage of cycle delays requires default clocking (IEEE 1800-2023 14.11)
                                           : ... note: In instance 'APB3_TB_top'
-node: DELAY 0x116d560 <e1426> {e97ah} [CYCLE]
   97 |       ##(cycles);
      |       ^~
- V3Ast.cpp:1324:     Dumping obj_vlt/t_interface_clocking_class/Vt_interface_clocking_class_015_assertpre.tree
- V3Assert.cpp:626:   assertAll: 
- V3Ast.cpp:1324:     Dumping obj_vlt/t_interface_clocking_class/Vt_interface_clocking_class_016_assert.tree
- V3LinkLevel.cpp:144:wrapTop: 
- V3Const.cpp:3945:   constifyAllLint: 
- V3Ast.cpp:1324:     Dumping obj_vlt/t_interface_clocking_class/Vt_interface_clocking_class_018_const.tree
- V3SplitVar.cpp:1239:splitVariable: 
- V3Inst.cpp:618:     dearrayAll: 
- V3LinkDot.cpp:3924: linkDotArrayed: 
- V3Fork.cpp:674:     makeDynamicScopes: 
- V3Ast.cpp:1324:     Dumping obj_vlt/t_interface_clocking_class/Vt_interface_clocking_class_023_fork_dynscope.tree
- V3Fork.cpp:680:     makeTasks: 
- V3Ast.cpp:1324:     Dumping obj_vlt/t_interface_clocking_class/Vt_interface_clocking_class_024_fork.tree
- V3Begin.cpp:363:    debeginAll: 
- V3Ast.cpp:1324:     Dumping obj_vlt/t_interface_clocking_class/Vt_interface_clocking_class_025_begin.tree
- V3Tristate.cpp:1834:tristateAll: 
- V3Ast.cpp:1324:     Dumping obj_vlt/t_interface_clocking_class/Vt_interface_clocking_class_026_tristate.tree
- V3Unknown.cpp:532:  unknownAll: 
- V3Ast.cpp:1324:     Dumping obj_vlt/t_interface_clocking_class/Vt_interface_clocking_class_027_unknown.tree
- V3DfgOptimizer.cpp:233:extract: 
- V3Ast.cpp:1324:     Dumping obj_vlt/t_interface_clocking_class/Vt_interface_clocking_class_028_dfg-extract.tree
- V3DfgOptimizer.cpp:240:optimize: 
- V3Ast.cpp:1324:     Dumping obj_vlt/t_interface_clocking_class/Vt_interface_clocking_class_029_dfg-optimize.tree
- V3Inline.cpp:630:   inlineAll: 
- V3Ast.cpp:1324:     Dumping obj_vlt/t_interface_clocking_class/Vt_interface_clocking_class_030_inline.tree
- V3LinkDot.cpp:3924: linkDotArrayed: 
- V3DfgOptimizer.cpp:240:optimize: 
- V3Ast.cpp:1324:     Dumping obj_vlt/t_interface_clocking_class/Vt_interface_clocking_class_032_dfg-optimize.tree
- V3Const.cpp:3988:   constifyAll: 
- V3Ast.cpp:1324:     Dumping obj_vlt/t_interface_clocking_class/Vt_interface_clocking_class_033_const.tree
- V3Dead.cpp:550:     deadifyDTypes: 
- V3Ast.cpp:1324:     Dumping obj_vlt/t_interface_clocking_class/Vt_interface_clocking_class_034_deadDtypes.tree
%Error: Exiting due to 1 error(s)
- V3Ast.cpp:1324:     Dumping obj_vlt/t_interface_clocking_class/Vt_interface_clocking_class_990_final.tree
- V3StatsReport.cpp:231:statsReport: 
%Error: Aborting since under --debug

Program received signal SIGABRT, Aborted.
__pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at pthread_kill.c:44
44	      return INTERNAL_SYSCALL_ERROR_P (ret) ? INTERNAL_SYSCALL_ERRNO (ret) : 0;
#0  __pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at pthread_kill.c:44
#1  0x0000155554e8b9b3 in __pthread_kill_internal (signo=6, threadid=<optimized out>) at pthread_kill.c:78
#2  0x0000155554e3e646 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#3  0x0000155554e287f3 in __GI_abort () at abort.c:79
#4  0x0000000000473017 in V3Error::vlAbort () at ../V3Error.cpp:291
#5  0x00000000004740a0 in V3ErrorGuarded::vlAbortOrExit (this=this@entry=0x1063d20 <V3Error::s()::s_s>) at ../V3Error.cpp:104
#6  0x00000000004754b2 in V3ErrorGuarded::v3errorEnd (this=0x1063d20 <V3Error::s()::s_s>, sstr=..., extra="") at ../V3Error.cpp:241
#7  0x000000000047555d in V3Error::v3errorEnd (sstr=..., extra="") at ../V3Error.cpp:318
#8  0x0000000000475606 in v3errorEndFatal (sstr=...) at ../V3Error.cpp:326
#9  0x0000000000476955 in V3Error::abortIfWarnings () at ../V3Error.cpp:280
#10 0x0000000000597f47 in V3Error::abortIfErrors () at ../V3Error.h:486
#11 0x0000000000595178 in process () at ../Verilator.cpp:318
#12 0x00000000005968be in verilate (argString="--prefix Vt_interface_clocking_class ../obj_vlt/t_interface_clocking_class/Vt_interface_clocking_class__main.cpp --exe --make gmake --x-assign unique -cc -Mdir obj_vlt/t_interface_clocking_class --fde"...) at ../Verilator.cpp:707
#13 0x0000000000597502 in main (argc=<optimized out>, argv=0x7fffffffd6c8) at ../Verilator.cpp:852
```

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

From **"Abstract BFMs Outshine Virtual Interfaces for Advanced SystemVerilog Testbenches"**

```
// DESCRIPTION: Verilator: Verilog Test module
//
// This file ONLY is placed under the Creative Commons Public Domain, for
// any use, without warranty, 2024 by .
// SPDX-License-Identifier: CC0-1.0

package APB3_pkg;
   virtual class APB3_BFM;
      pure virtual task write (int unsigned addr, data);
      pure virtual task read (int unsigned addr, output int unsigned data);
      pure virtual task init ();
      pure virtual task idle (int unsigned cycles);
   endclass
endpackage

module APB3_TB_top;
   import APB3_pkg::*;
   bit CLK;
   always #5 CLK = ~CLK; // clock generator

   APB3_TB_intf apb3_intf(CLK);

/* -----\/----- EXCLUDED -----\/-----
   APB3_device DUT(
      .PCLK(CLK),
      .PENABLE(apb3_intf.PENABLE),
   ...);
 -----/\----- EXCLUDED -----/\----- */

   initial begin : Test_Activity
      APB3_BFM bfm; // abstract BFM reference
      int read_data;
      bfm = apb3_intf.bfm; // reference to BFM
      bfm.init;
      bfm.idle(5);
      bfm.write(100, 1234);
      bfm.read(100, read_data);
      if (read_data != 1234)
	$display("error: unexpected read data");
      bfm.idle(100); $stop;
   end
endmodule

interface APB3_TB_intf(input bit PCLK);
   import APB3_pkg::*;

   typedef logic [15:0] addr_t;
   typedef logic [15:0] data_t;

   logic  PENABLE, PWRITE, PSEL, PREADY;
   addr_t PADDR;
   data_t PWDATA, PRDATA;

   default clocking apb_ck @(posedge PCLK);
      output 	PENABLE, PWRITE, PSEL, PADDR, PWDATA;
      input 	PREADY, PRDATA;
   endclocking

class APB3_concrete_BFM extends APB3_BFM;
   task init;
      apb_ck.PSEL <= '0;
      apb_ck.PENABLE <= '0;
      apb_ck.PWRITE <= '0;
   endtask

   task write(int unsigned addr, data);
//      ##0
	apb_ck.PSEL <= '1;
      apb_ck.PENABLE <= '0;
      apb_ck.PWRITE <= '1;
      apb_ck.PADDR <= addr_t'(addr);
      apb_ck.PWDATA <= data_t'(data);
//      ##1
	apb_ck.PENABLE <= '1;
//      do ##1; while (!apb_ck.PREADY);
      apb_ck.PSEL <= '0;
      apb_ck.PENABLE <= '0 ;
   endtask

   task read(int unsigned addr, output int unsigned data);
//      ##0
	apb_ck.PSEL <= '1;
      apb_ck.PENABLE <= '0;
      apb_ck.PWRITE <= '0;
      apb_ck.PADDR <= addr_t'(addr);
//      ##1
	apb_ck.PENABLE <= '1;
//      do ##1; while (!apb_ck.PREADY);
      apb_ck.PSEL <= '0;
      apb_ck.PENABLE <= '0 ;
      data = int'(apb_ck.PRDATA);
   endtask

   task idle(int unsigned cycles);
      ##(cycles);
   endtask
endclass

   APB3_concrete_BFM bfm = new;
endinterface
```

What 'verilator' command line do we use to run your example?

```
verilator --prefix Vt_interface_clocking_class ../obj_vlt/t_interface_clocking_class/Vt_interface_clocking_class__main.cpp --exe --make gmake --x-assign unique -cc -Mdir obj_vlt/t_interface_clocking_class --fdedup --debug-check --comp-limit-members 10 --clk clk  -f input.vc +define+TEST_OBJ_DIR=obj_vlt/t_interface_clocking_class +define+TEST_DUMPFILE=obj_vlt/t_interface_clocking_class/simx.vcd --timing t/t_interface_clocking_class.v
```

What 'verilator --version' are you using?  Did you try it with the git master version?

```
v5.026-gd5cfe1a37
```

What OS and distribution are you using?

```
Linux 5.14.0-427.20.1.el9_4.x86_64
```

May we assist you in trying to fix this in Verilator yourself?

(Please avoid attaching screenshots that show text - you can convert images to text using e.g. https://ocr.space)

wsnyder, 2024-06-30:
Could you convert the example to a self-checking test?  Also as you submit many issues, please use the test_regress format described in the docs/internals to make it easier to try them, thanks.

wsnyder, 2024-06-30:
Ignore second half of that - see it is in test_regress format, thanks.

closed 2026-10-21 by wsnyder