verilator/verilator #5213: Failed parameter type override in class

esynr3z, 2024-06-30:
Hello,
I have class with some default `parameter type` and cannot override it, which should be legal.

The code:

```module test;
  typedef enum {
    FOO_0,
    FOO_1
  } foo_e;

  typedef enum {
    BAR_0,
    BAR_1
  } bar_e;

  class baz #(parameter type E = foo_e);
    static function void print();
      E enum_item = enum_item.first();
      do begin
        $display("%s", enum_item.name());
        enum_item = enum_item.next();
      end while (enum_item != enum_item.first());
    endfunction
  endclass

  initial begin
    baz#(bar_e)::print();
    $finish;
  end
endmodule
```

Output (`foo_e` was not overridden with `bar_e`):
```
- V e r i l a t i o n   R e p o r t: Verilator 5.027 devel rev v5.026-13-gd6a294b78
- Verilator: Built from 0.012 MB sources in 3 modules, into 0.029 MB in 17 C++ files needing 0.000 MB
- Verilator: Walltime 0.145 s (elab=0.000, cvt=0.003, bld=0.136); cpu 0.000 s on 1 threads; alloced 18.789 MB
FOO_0
FOO_1
- test.sv:24: Verilog $finish
- S i m u l a t i o n   R e p o r t: Verilator 5.027 devel
```

Verilator version: Verilator 5.027 devel rev v5.026-13-gd6a294b78
OS: Ubuntu 22.04.4 LTS

wsnyder, 2024-07-02:
V3Param seems to not be making the parameterized class at all.  I expected this to work around it but it still breaks:

```
  initial begin
    foo#(foo_e)::print();
    baz#(bar_e)::print();
```

Needs more debug, if you have time to look into making a pull to fix it, it would be appreciated, but I will try to get to this once have a chance.

esynr3z, 2024-07-05:
I'd like to help, but unfortunately, I don't have enough skill to do the debug.

rhanqtl, 2024-07-11:
@wsnyder @esynr3z I'll try to make a pull.

rhanqtl, 2024-07-12:
`foo_e` and `bar_e` are considered similar, so `foo_e` is not replaced: 

https://github.com/verilator/verilator/blob/6882f8c55e9ab55f298e5203c193d0ccd94e2a57/src/V3Param.cpp#L749-L754

Trying to fix.

closed 2024-10-03 by wsnyder