verilator/verilator #5215: Why does Verilator report errors for assertions? Is it because it doesn't support property syntax?

ZhenzongGu, 2024-07-01:
property same_entry_write_dirty;
        @(posedge clk) disable iff(~rst_n)
            (write_miss_en ) |-> (&is_dirty);
    endproperty
[image]

wsnyder, 2024-07-01:
Please attach a minimal example that compiles, and don't use screen shots, thanks.  It might end up as unsupported, but should not give a syntax error.

ZhenzongGu, 2024-07-02:
> Please attach a minimal example that compiles, and don't use screen shots, thanks. It might end up as unsupported, but should not give a syntax error.
module
 property same_entry_write_dirty;
        @(posedge clk) disable iff(~rst_n)
            (write_miss_en & |(is_dirty_set&is_dirty))|-> (&is_dirty);
    endproperty
    assert property(same_entry_write_dirty) else $error("write dirty cacheline while clean entries exist");
....
endmodule
  260 |     property same_entry_write_dirty;
      |     ^~~~~~~~
%Error-UNSUPPORTED: /data/memory/NX9031/100A_BE/SRAM/s2ssls0g4u1p1024x262m4b1w0c1p0d0s2rm4rw00zh0h0ms0mg1/compout/views/s2ssl
s0g4u1p1024x262m4b1w0c1p0d0s2rm4rw00zh0h0ms0mg1/tt0p75v85c/s2ssls0g4u1p1024x262m4b1w0c1p0d0s2rm4rw00zh0h0ms0mg1.v:2105:5: Uns
upported: Verilog 1995 deassign

lint option:
(finished - exit code 1) ['/bin/tcsh', '-c', 'verilator --lint-only -Wall -Wno-DECLFILENAME -I''/proj_tmp/scoreboard/verilog'' -sv --assert --default-language "1800-2012"  -f /proj_tmp/scoreboard_buffer_eda.vc -top scoreboard ''/tmp/vwyLoeh/5/
scoreboard_cac.sv''']

wsnyder, 2024-07-02:
That isn't a complete example, please paste one that compiles on another simulator cleanly, thanks.

ZhenzongGu, 2024-07-03:
sorry,
```
module prop_test (/*autoarg*/     
     // Inputs     
     clk,     
     rst_n,     
     wen,     
     wr_hit_way    
     );     
      /*autooutput*/     
      
      /*autoinput*/     
      
      /*autowire*/     
      input clk;     
      input rst_n;     
      
      input wen;     
      input[5:0] wr_hit_way;     
      
     property wr_hit_one_way; > syntax error, unexpected property                                                                                                                                                                                           
          @(posedge clk) disable iff(~rst_n)     
               (wen) |-> ($countones(wr_hit_way)<=1);     
      endproperty     
      assert property(wr_hit_one_way) else $error("write hit multi address!!");     
      
  endmodule     
```

[image]

wsnyder, 2024-07-03:
This code works fine for me. Perhaps you are not using the latest version.

ZhenzongGu, 2024-07-03:
Are my compilation options correct?
[image]

wsnyder, 2024-07-03:
What verilator version are you using?

Please attach text only, not screen shots.

ZhenzongGu, 2024-07-03:
Verilator 4.226 2022-08-31 rev UNKNOWN.REV

Copyright 2003-2022 by Wilson Snyder.  Verilator is free software; you can
redistribute it and/or modify the Verilator internals under the terms of
either the GNU Lesser General Public License Version 3 or the Perl Artistic
License Version 2.0.

wsnyder, 2024-07-03:
You are using a 2 year old version, use the latest.

closed 2024-07-03 by wsnyder

ZhenzongGu, 2024-07-03:
Ok,thanks