verilator/verilator #5216: Warning-INITIALDLY with module or interface tasks, but not with class tasks

RootCubed, 2024-07-01:
While trying out workarounds for #5116, I came across this behavior:
Writing non-blocking assignments in tasks do not yield an `INITIALDLY` warning if the task belongs to a class, but it does if the task belongs to an interface or a module. Examples (compiled with `verilator --binary --build <file>.sv`):

Class tasks with virtual interfaces (Compiles and runs correctly):
```sv
interface INTF ();
    logic [7:0] field1; logic [7:0] field2;
endinterface

class driver;
    virtual INTF intf;

    function new(virtual INTF intf);
        this.intf = intf;
    endfunction

    task automatic set_fields();
        intf.field1 <= #5ns 42;
        #3ns;
        intf.field2 = 12;
    endtask
endclass

module top ();
    INTF intf ();
    driver drv = new(intf);

    initial begin
        $display("%d %d | Time: %t", intf.field1, intf.field2, $time);
        drv.set_fields();
        $display("%d %d | Time: %t", intf.field1, intf.field2, $time);
        #3ns;
        $display("%d %d | Time: %t", intf.field1, intf.field2, $time);
        $finish;
    end
endmodule
```

Interface tasks (`%Warning-INITIALDLY: initialdly_interface.sv:5:21`):
```sv
interface INTF ();
    logic [7:0] field1; logic [7:0] field2;

    task automatic set_fields();
        intf.field1 <= #5ns 42;
        #3ns;
        intf.field2 = 12;
    endtask
endinterface

module top ();
    INTF intf();

    initial begin
        $display("%d %d | Time: %t", intf.field1, intf.field2, $time);
        intf.set_fields();
        $display("%d %d | Time: %t", intf.field1, intf.field2, $time);
        #3ns;
        $display("%d %d | Time: %t", intf.field1, intf.field2, $time);
        $finish;
    end
endmodule
```

Module tasks (`%Warning-INITIALDLY: initialdly_module.sv:3:16`):
```sv
module driver( output logic [7:0] field1, output logic [7:0] field2 );
    task automatic set_fields();
        field1 <= #5ns 42;
        #3ns;
        field2 = 12;
    endtask
endmodule

module top ();
    logic [7:0] f1, f2;

    driver i_drv(f1, f2);

    initial begin
        $display("%d %d | Time: %t", f1, f2, $time);
        i_drv.set_fields();
        $display("%d %d | Time: %t", f1, f2, $time);
        #3ns;
        $display("%d %d | Time: %t", f1, f2, $time);
        $finish;
    end
endmodule
```

Is this behavior intended? If not, how hard would it be to make the non-blocking assignments work in at least `module` tasks as well?

wsnyder, 2024-07-02:
It's intended in that the same task can be used both in initial and not, so it makes sense it doesn't currently know how to warn, but it's certainly not right in that the user isn't aware of the problem.  Of course the real fix is to support this behavior - keeping this open in hopes we get that first.

RootCubed, 2024-07-02:
I understand we don't support non-blocking assignments in initial blocks (and thus by extensions tasks called from an initial block). But I was surprised that calling a class task with non-blocking assignments from an initial block *does* work...

```sv
class driver;
// module driver;
    logic [7:0] test;

    task automatic set_fields();
        test <= #5ns 42;
        #1ns;
    endtask
endclass
// endmodule

module top ();
    driver drv = new();
    // driver drv;

    initial begin
        drv.set_fields();
        $display("%d | Time: %t", drv.test, $time);
        #9ns;
        $display("%d | Time: %t", drv.test, $time);
        $finish;
    end
endmodule
```
I would have expected this to either work both with `class` and `module` - or with neither. However, while the `module` version doesn't work, the `class` version does...

wsnyder, 2024-07-02:
Hmm, @kbieganski?

kbieganski, 2024-07-02:
The check for NBAs in `initial` blocks is very simple – it only sees NBAs directly under `initial`s. Class methods are not inlined, so it would have to peek into each class method call to see if there's an NBA somewhere (but it doesn't do that). NBAs are supported in class methods just because there's one occurrence of an NBA in UVM (see https://github.com/verilator/verilator/pull/4572), however, they require `--timing`.

I suppose we could support NBAs in `initial`s the same way? It's certainly not worth it to traverse the call tree just to see if there's an NBA hidden somewhere.

RootCubed, 2024-07-02:
I simply added an early return at `V3Active.cpp:365` and it seems to be working correctly now! The three examples in the initial report now have the same behavior.
I think all it needs is a check there to see if `--timing` is active, as without there is an internal error thrown later on (`Internal Error: initialdly_module.sv:3:9: ../V3Delayed.cpp:820: <= assignment in non-clocked block, should have been converted in V3Active`) instead of a nicer error like the one for class methods (`%Error-NOTIMING: initialdly_class.sv:13:21: Delayed assignment in a non-inlined function/task requires --timing`).

RootCubed, 2024-07-03:
While it seems to work for the use cases I need, there definitely still are many cases where it doesn't work (see test suite ran on this patch https://github.com/RootCubed/verilator/actions/runs/9772966644)

It seems that delayed NBAs work, but non-delayed ones usually don't. Example:
```sv
module top ();
    logic f1, f2;
    logic clk;

    initial begin
        f1 <= 1; // f1 <= #4ns 1; works
        f2 <= 1;
    end

    initial begin
        #5ns;
        if (!f1 || !f2) $stop();
        $finish;
    end
endmodule
```

Here I get `%Error: Internal Error: initialdly_issue.sv:6:9: ../V3Delayed.cpp:820: <= assignment in non-clocked block, should have been converted in V3Active`

I am not really familiar enough with the internals to know what the next step from here is. From what I could gather, a fork is created to handle delayed NBAs - and non-inlined functions like class methods also do this for non-delayed NBAs too. So that logic probably needs to be extended to do the same for initial/final blocks (create forks for "normal" NBAs), but I'm not quite sure where that needs to go....

wsnyder, 2026-08-28:
INITIALDLY was deprecated, and initial delays now supported, so no longer relevant.

closed 2026-08-28 by wsnyder