verilator/verilator #5218: no verilog.h file when trying to build the rocket-chip C emulator

nhuynh368, 2024-07-02:
Hey, I'm trying to build the C simulator for the rocket-chip project, but apparently the Verilator version the build tool fetched did not include a verilog.c or verilog.h in the src folder. For reference the build script fetches Verilator from Veripool. Any way to get verilator to build properly?

make -C verilator/src/verilator-4.028 verilator_bin
make[1]: Entering directory '/home/nam-huynh/rocket-chip/emulator/verilator/src/verilator-4.028'
------------------------------------------------------------
making verilator in src
make -C src 
make[2]: Entering directory '/home/nam-huynh/rocket-chip/emulator/verilator/src/verilator-4.028/src'
make -C obj_dbg -j 1  TGT=../../bin/verilator_bin_dbg VL_DEBUG=1 -f ../Makefile_obj serial
make[3]: Entering directory '/home/nam-huynh/rocket-chip/emulator/verilator/src/verilator-4.028/src/obj_dbg'
make[3]: Nothing to be done for 'serial'.
make[3]: Leaving directory '/home/nam-huynh/rocket-chip/emulator/verilator/src/verilator-4.028/src/obj_dbg'
make -C obj_dbg       TGT=../../bin/verilator_bin_dbg VL_DEBUG=1 -f ../Makefile_obj
make[3]: Entering directory '/home/nam-huynh/rocket-chip/emulator/verilator/src/verilator-4.028/src/obj_dbg'
      Compile flags:  g++ -I/home/nam-huynh/rocket-chip/emulator/verilator/install/include -I/home/nam-huynh/rocket-chip/emulator/verilator/install/include -std=gnu++14 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow -I/home/nam-huynh/rocket-chip/emulator/verilator/install/include -MMD -I. -I.. -I.. -I../../include -I../../include -DYYDEBUG -ggdb -DVL_DEBUG -D_GLIBCXX_DEBUG -MP -faligned-new -Wno-unused-parameter -Wno-shadow -fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc -fno-builtin-free -DDEFENV_SYSTEMC="" -DDEFENV_SYSTEMC_ARCH="" -DDEFENV_SYSTEMC_INCLUDE="" -DDEFENV_SYSTEMC_LIBDIR="" -DDEFENV_VERILATOR_ROOT="/home/nam-huynh/rocket-chip/emulator/verilator/install/share/verilator"
g++ -I/home/nam-huynh/rocket-chip/emulator/verilator/install/include -std=gnu++14  -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow -I/home/nam-huynh/rocket-chip/emulator/verilator/install/include -MMD -I. -I.. -I.. -I../../include -I../../include -DYYDEBUG 	 -ggdb -DVL_DEBUG -D_GLIBCXX_DEBUG -MP  -faligned-new -Wno-unused-parameter -Wno-shadow -fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc -fno-builtin-free -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/home/nam-huynh/rocket-chip/emulator/verilator/install/share/verilator\" -Wno-char-subscripts -Wno-unused -c ../V3ParseGrammar.cpp
In file included from ../V3ParseGrammar.cpp:27:
verilog.c:308:10: fatal error: verilog.h: No such file or directory
compilation terminated.
make[3]: *** [../Makefile_obj:298: V3ParseGrammar.o] Error 1
make[3]: Leaving directory '/home/nam-huynh/rocket-chip/emulator/verilator/src/verilator-4.028/src/obj_dbg'
make[2]: *** [Makefile:70: ../bin/verilator_bin_dbg] Error 2
make[2]: Leaving directory '/home/nam-huynh/rocket-chip/emulator/verilator/src/verilator-4.028/src'
make[1]: *** [Makefile:231: verilator_bin] Error 2
make[1]: Leaving directory '/home/nam-huynh/rocket-chip/emulator/verilator/src/verilator-4.028'
make: *** [Makefrag-verilator:42: verilator/src/verilator-4.028/bin/verilator] Error 2

wsnyder, 2024-07-02:
First, 4.028 is very old, maybe there's a reason rocket-chip needs this version, but this means we won't be fixing any issues related to it.  Try the latest stable Verilator and build using our instructions, then, or first, ask rocket-chip if unsure how to use it with rocket-chip.

As to the error this means bison didn't build correctly. It looks like you didn't include the logfile starting from a clean area with configure and build (but rather only that from typing "make" again).  See why bison isn't building correctly.

If you have no idea what I'm talking about, ask rocket-chip for advice since they are wrapping Verilator's build somehow.

Closing but feel free to follow up here.

closed 2024-07-02 by wsnyder

nhuynh368, 2024-07-03:
I deleted everything in the emulator folder and tried to build yet again. This might be a rocket-chip build script issue but I want to be prudent about this just in case. Weirdly enough, the source doesn't seem to be broken, it's just that the make tool just did not make the .c and .h files for some reason.

mkdir -p verilator/
wget http://www.veripool.org/ftp/verilator-4.028.tgz -O verilator/verilator-4.028.tar.gz
--2024-07-03 09:33:30--  http://www.veripool.org/ftp/verilator-4.028.tgz
Resolving www.veripool.org (www.veripool.org)... 173.230.151.124
Connecting to www.veripool.org (www.veripool.org)|173.230.151.124|:80... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://www.veripool.org/ftp/verilator-4.028.tgz [following]
--2024-07-03 09:33:31--  https://www.veripool.org/ftp/verilator-4.028.tgz
Connecting to www.veripool.org (www.veripool.org)|173.230.151.124|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 2448209 (2.3M) [application/x-gzip]
Saving to: ‘verilator/verilator-4.028.tar.gz’

verilator/verilator-4.028.tar 100%[=================================================>]   2.33M  1.72MB/s    in 1.4s    

2024-07-03 09:33:33 (1.72 MB/s) - ‘verilator/verilator-4.028.tar.gz’ saved [2448209/2448209]

rm -rf verilator/src/verilator-4.028/
mkdir -p verilator/src/verilator-4.028/
cat verilator/verilator-4.028.tar.gz | tar -xz --strip-components=1 -C verilator/src/verilator-4.028/
touch verilator/src/verilator-4.028/configure
mkdir -p verilator/src/verilator-4.028/
cd verilator/src/verilator-4.028/ && ./configure --prefix=/home/nam-huynh/rocket-chip/emulator/verilator/install
configuring for Verilator 4.028 2020-02-06
checking whether to perform partial static linking of Verilator binary... yes
checking whether to use tcmalloc... check
checking whether to use hardcoded paths... yes
checking whether to show and stop on compilation warnings... no
checking whether to run long tests... no
checking whether allow pre-C++11... no
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for g++... g++
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking for a BSD-compatible install... /usr/bin/install -c
compiler is g++ --version = g++ (Ubuntu 13.2.0-23ubuntu4) 13.2.0
checking that C++ compiler can compile simple program... yes
checking for perl... /usr/bin/perl
checking for flex... /usr/bin/flex
checking for bison... /usr/bin/bison
checking how to run the C++ preprocessor... g++ -E
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking for size_t... yes
checking for size_t... (cached) yes
checking for inline... inline
checking whether g++ accepts -std=gnu++14... yes
checking whether g++ accepts -std=c++03... yes
checking whether g++ accepts -Wextra... yes
checking whether g++ accepts -Wfloat-conversion... yes
[... 156 lines cut ...]
g++ -I/home/nam-huynh/rocket-chip/emulator/verilator/install/include -std=gnu++14  -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow -I/home/nam-huynh/rocket-chip/emulator/verilator/install/include -MMD -I. -I.. -I.. -I../../include -I../../include -DYYDEBUG 	 -ggdb -DVL_DEBUG -D_GLIBCXX_DEBUG -MP  -faligned-new -Wno-unused-parameter -Wno-shadow -fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc -fno-builtin-free -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/home/nam-huynh/rocket-chip/emulator/verilator/install/share/verilator\" -c ../V3String.cpp
g++ -I/home/nam-huynh/rocket-chip/emulator/verilator/install/include -std=gnu++14  -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow -I/home/nam-huynh/rocket-chip/emulator/verilator/install/include -MMD -I. -I.. -I.. -I../../include -I../../include -DYYDEBUG 	 -ggdb -DVL_DEBUG -D_GLIBCXX_DEBUG -MP  -faligned-new -Wno-unused-parameter -Wno-shadow -fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc -fno-builtin-free -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/home/nam-huynh/rocket-chip/emulator/verilator/install/share/verilator\" -c ../V3Subst.cpp
g++ -I/home/nam-huynh/rocket-chip/emulator/verilator/install/include -std=gnu++14  -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow -I/home/nam-huynh/rocket-chip/emulator/verilator/install/include -MM [... cut ...]
make[3]: Leaving directory '/home/nam-huynh/rocket-chip/emulator/verilator/src/verilator-4.028/src/obj_dbg'
make[2]: *** [Makefile:70: ../bin/verilator_bin_dbg] Error 2
make[2]: Leaving directory '/home/nam-huynh/rocket-chip/emulator/verilator/src/verilator-4.028/src'
make[1]: *** [Makefile:231: verilator_bin] Error 2
make[1]: Leaving directory '/home/nam-huynh/rocket-chip/emulator/verilator/src/verilator-4.028'
make: *** [Makefrag-verilator:42: verilator/src/verilator-4.028/bin/verilator] Error 2

wsnyder, 2024-07-03:
Perhaps is #2505 which was fixed years ago, but I not going to debug - please use a newer version.