verilator/verilator #5220: C++ compile errors: operands to ‘?:’ have different types, base operand of ‘->’ is not a pointer, etc.

solomatnikov, 2024-07-03:
```
In file included from Vt_complex__ALL.cpp:5:
Vt_complex___024root__DepSet_h4bf5e420__0.cpp: In function ‘void Vt_complex___024root___nba_sequent__TOP__0(Vt_complex___024root*)’:
Vt_complex___024root__DepSet_h4bf5e420__0.cpp:93:42: error: operands to ‘?:’ have different types ‘VlQueue<VlClassRef<Vt_complex___024unit__03a__03az_txn_class> >’ and ‘CData’ {aka ‘unsigned char’}
   91 |                 if (VL_LTS_III(32, 0U, ((9U >= (0xfU
      |                                         ~~~~~~~~~~~~
   92 |                                                 & top__DOT__testbench__DOT__tile__DOT__mmio_z_bfm__DOT__unnamedblk1__DOT__unnamedblk2_1__DOT__i))
      |                                                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   93 |                                          ? vlSelf->top__DOT__testbench__DOT__tile__DOT__mmio_z_bfm__DOT__z_txn_qs
      |                                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   94 |                                         [(0xfU & top__DOT__testbench__DOT__tile__DOT__mmio_z_bfm__DOT__unnamedblk1__DOT__unnamedblk2_1__DOT__i)]
      |                                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   95 |                                          : vlSelf->top__DOT__testbench__DOT__tile__DOT__mmio_z_bfm__DOT____Vxrand_h8df6f50b__0).size())) {
      |                                          ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Vt_complex___024root__DepSet_h4bf5e420__0.cpp:119:47: error: base operand of ‘->’ is not a pointer
  119 |                                               ->__PVT__cid))
      |                                               ^~
Vt_complex___024root__DepSet_h4bf5e420__0.cpp:134:47: error: base operand of ‘->’ is not a pointer
  134 |                                               ->__PVT__req_opcode))
      |                                               ^~
Vt_complex___024root__DepSet_h4bf5e420__0.cpp:145:28: error: base operand of ‘->’ is not a pointer
  145 |                            ->__PVT__address);
      |                            ^~
```

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

```
// DESCRIPTION: Verilator: Verilog Test module
//
// This file ONLY is placed under the Creative Commons Public Domain, for
// any use, without warranty, 2024 by .
// SPDX-License-Identifier: CC0-1.0

package z_pkg;
   localparam int OPCODEW=3;
   localparam int CIDW=4;
   localparam int SIDW=10;
   localparam int CTAGW=7;
   localparam int STAGW=9;
   localparam int STATEW=3;
   localparam int ADDRW=37;
   localparam int DATAW=256;
   localparam int MASKW=DATAW/8;
   localparam int SIZEW=4;
  typedef enum logic [OPCODEW-1:0] {
    A   = 3'h 0,
    B   = 3'h 1
  } a_op_t;
  typedef enum logic [OPCODEW-1:0] {
    C   = 3'h 0,
    D   = 3'h 1
  } b_op_t;
  typedef enum logic [OPCODEW-1:0] {
    E   = 3'h 0,
    F   = 3'h 1
  } c_op_t;
  typedef enum logic [OPCODEW-1:0] {
    G       = 3'h 0,
    H       = 3'h 1
  } d_op_t;
  typedef enum logic [OPCODEW-1:0] {
    I = 3'h 4
  } e_op_t;
  typedef logic [CIDW-1:0]      cid_t;
  typedef logic [SIDW-1:0]      sid_t;
  typedef logic [CTAGW-1:0]     ctag_t;
  typedef logic [STAGW-1:0]     stag_t;
  typedef logic [STATEW-1:0] state_t;
  typedef logic [ADDRW-1:0]     address_t;
[... 405 lines cut ...]
  (
   input bit clk /*verilator clocker*/ ,
   input bit reset_l
   );
   testbench_core_wrapper testbench( .* );
endmodule
```

What 'verilator' command line do we use to run your example?

```
verilator --prefix Vt_complex ../obj_vlt/t_complex/Vt_complex__main.cpp --exe --make gmake --x-assign unique -cc -Mdir obj_vlt/t_complex --fdedup --debug-check --comp-limit-members 10 --clk clk  -f input.vc +define+TEST_OBJ_DIR=obj_vlt/t_complex +define+TEST_DUMPFILE=obj_vlt/t_complex/simx.vcd t/t_complex.v
```

What 'verilator --version' are you using?  Did you try it with the git master version?

```
Verilator 5.027 devel rev v5.026-14-g72993ec3d
```

What OS and distribution are you using?

```
Linux 5.14.0-427.20.1.el9_4.x86_64
```

wsnyder, 2024-07-03:
Can you please shorten the example to only that needed to show the issue? Thanks.

solomatnikov, 2024-07-03:
> Can you please shorten the example to only that needed to show the issue? Thanks.

No, I cannot.

When I tried, it compiled but failed at runtime:

```
z_bfm_sender.sv:85: Null pointer dereferenced
```

wsnyder, 2024-07-03:
Hmm, try if `-fno-inline` lets it get smaller?

solomatnikov, 2024-07-03:
@wsnyder this is a real showstopper - I tried to find so many workarounds for Verilator bugs but every time I get a new kind of failure.

Clearly, the strategy of having only minimal one feature tests doesn't work - almost any combination of features I tried results in a new failure (and I am not even talking about `--timing` - it fails in various ways so often). Tests should cover not only single features but also combinations.

solomatnikov, 2024-07-03:
> Hmm, try if `-fno-inline` lets it get smaller?

Just tried - it doesn't make a difference - slightly smaller test case doesn't fail the same way.

wsnyder, 2024-07-03:
Having larger tests in the suite is reasonable. To debug an issue however the way I debug it's generally much more efficient the smaller the tests case.  E.g. for UVM each problem found typically had to have a small test extracted to debug.  If you want to submit pull requests that also fix issues there's much more latitude since you can use your own debugging style.

I appreciate you have a large number of issues; there's not much we can do about resolving those quickly unless you/your organization contributes directly or indirectly fixes.

Anyhow as to this issue, thanks for trying to shrink, if you can't I'll debug with what's given.

solomatnikov, 2024-07-03:
Unfortunately, I could not reduce the above test case any further.

From all attempts so far it looks like it must be large enough to hit this issue.

E.g. just removing `CH_Y2` from above (-2 lines) results in a different failure (which I could not re-produce in `test_regress` format so far).

wsnyder, 2024-07-05:
Unfortunately unrelated to 5221.

closed 2024-07-08 by wsnyder