verilator/verilator #5221: CPP ERROR: Request for a member that is non class type

deepans88, 2024-07-03:
I am using verilator to create a binary executable of our system verilog design . We are running into some errors on the cpp files generated.  Please find one such example below 

 

 

**ERROR : Vmod_sram_wrapper__DepSet_h01a03603__0__Slow.cpp:103:46: error: request for member \u2018__PVT__BIST\u2019 in \u2018vlSelf->Vmod_sram_wrapper:`_i_ram_mbist`_[7]\u2019, which is of non-class type \u2018CData\u2019 {aka \u2018unsigned char\u2019}
  103 |                                         [7U].__PVT__BIST))));**

Here the tool is inferring structure instance` i_ram_mbist` is of type character rather than class/struct type.

 

 This structure is defined as a package :-

```
package dft_mbist;

  typedef struct   {
    logic         BIST;
    logic [38:0]  WEB; // MBIST bit write enable
    logic         CEB;
  } t_i_mbist_top_mem;

endpackage
```

In our system Verilog RTL , this structure is instantiated as:-

 

```
 import dft_mbist::*;

 input      t_i_mbist_top_mem i_ram_mbist [7:0];

mem_wrapper  wrapper_0(

.bist_0 (i_ram_mbist[0].MBIST);

.bist_1(i_ram_mbist[1].MBIST);

.bist_2(i_ram_mbist[2].MBIST);
.

.

.

.

);
```

 

**Verilator command used :- Verilator - --binary -j 0 -Wall --timescale 1ps/1fs --bbox-unsup --bbox-sys --assert --Wpedantic -sv -Wno-lint -Wno-style -Wno-BLKANDNBLK -Wno-BLKLOOPINIT -Wno-MULTIDRIVEN -Wno-UNOPTFLAT -Wno-COMBDLY -Wno-NOLATCH +systemverilogext+  -Wno-EOFNEWLINE -Wno-style -Wno-lint -f  <file_list.f>  -top top_module** 

 

****Verilator version :- Verilator 5.027 devel rev v5.026-2-gd5cfe1a**

**GCC version:-gcc (GCC) 7.3.0****

deepans88, 2024-07-03:
**OS type  : CentOS Linux 7 (Core)******

wsnyder, 2024-07-03:
I suspect there's a WIDTH warning you have suppressed in this code - likely near the instantiation?  (I don't see it in your verilator flags, but maybe you have metacomments in code disabling it?) If so, fix that warning as a workaround.  If not, try using a temporary wire of same type as the instance's lower pin, instead of the cross-module dotted assignment hooking directly to the pin.

Can you please provide a complete minimal standalone test that shows the problem?

solomatnikov, 2024-07-03:
This looks like a duplicate of #5220

deepans88, 2024-07-04:
[Verilator_test.zip](https://github.com/user-attachments/files/16103182/Verilator_test.zip)
Hi,

Using a temporary wire did not solve the issue. And I don't have any meta comments disabling warnings.

I am attaching a folder with the sample test case and file list .
  
Verilator command used :-  **Verilator - --binary -j 0 -Wall --timescale 1ps/1fs --bbox-unsup --bbox-sys --assert --Wpedantic -sv -Wno-lint -Wno-style -Wno-BLKANDNBLK -Wno-BLKLOOPINIT -Wno-MULTIDRIVEN -Wno-UNOPTFLAT -Wno-COMBDLY -Wno-NOLATCH +systemverilogext+  -Wno-EOFNEWLINE -Wno-style -Wno-lint -f  list.f  -top ram_wrapper** 

Failing with this error:-
**[Verilator_test.zip](https://github.com/user-attachments/files/16103180/Verilator_test.zip)**

wsnyder, 2024-07-05:
This was caused by having a top-level port having an unpacked structure data type.

closed 2024-07-05 by wsnyder