verilator/verilator #5223: How to dump the DfgGraph of a Verilog/SystemVerilog hardware?

enochliang, 2024-07-04:
I need to get DFG of a circuit.
The circuit that I'm working on is a SystemVerilog 64-bit counter "counter.sv".
I'm trying to get the DFG with following commands and flags.

`verilator --cc --dump-dfg --debug rtl/counter.sv`
`verilator --xml-only --dump-dfg --debug rtl/counter.sv`

Because the DfgGraph is for debugging so I tryed the flag `--debug`

As you mentioned in the document of verilator, `--dump-dfg` enables dumping DfgGraph `.dot` debug files with dumping level 3.
But after I tryed the command above, I only get some `.tree` files, they seem to be the AST after DFG optimization, not DFG itself.

So my questions are:
"How should I give the command to get the `.dot` file of DFG?",
"Are there any other way that I can try to dump the DfgGraph?"

The picture below are the files under `./obj_dir` after I run the first command.
[image]

gezalore, 2024-07-04:
Try `--dumpi-dfg 9`, or a smaller number if you want less verbosity. See dumpDfgLevel() in the code for what the various numbers give you. You should not need `--debug`.

As the docs say, `--dump-dfg` should be equivalent to `--dumpi-dfg 3`, a PR to fix that would be appreciated.

wsnyder, 2024-07-04:
Closing as answered but feel free to follow up.

closed 2024-07-04 by wsnyder

enochliang, 2024-07-06:
Thank you very much. It works.

Final command: 
`verilator --cc --dumpi-dfg 9 {design.sv} --top {design_name}`