verilator/verilator #5225: VPI difference between Verilator v4 and v5

fredrequin, 2024-07-04:
Hello,
in an effort to improve the support of force/release in cocotb, I was surprised to discover that Verilator v4.218 exposes the ``__VforceEn``, ``__VforceVal`` signals in the VPI interface but Verilator 5.012 does not.
I have started looking at the source code (V3Force.cpp, Verilator.cpp) to spot the differences between the two versions but, nothing obvious was visible.
I would like to access the ``__VforceEn``, ``__VforceVal`` signals with the VPI interface in Verilator v5, how can I get this "feature" back ?
Thank you,
Frédéric

wsnyder, 2024-07-04:
I suspect this line: https://github.com/verilator/verilator/blob/dace2c0a9a8f9d0559c4e4cbdb56ebdcae93b097/src/V3EmitCSyms.cpp#L370

See if making that condition true always fixes it - if so we need to set isSigUserRdPublic when creating the force variables.  If so might you make a pull with a test for this, and with that fix?

fredrequin, 2024-07-08:
The force variables are already flagged with ``sigUserRWPublic = true``, so I changed line 370 of V3EmitCSyms.cpp :
```
        if ((nodep->isSigUserRdPublic() || nodep->isSigUserRWPublic())
            && !m_cfuncp) m_modVars.emplace_back(m_modp, nodep);
```
I now have the ``__VforceVal`` and ``__VforceEn`` in the VPI interface and it does not seem to break the visitor method of V3EmitCSyms.
That's great !
Should I do a PR ?

wsnyder, 2024-07-08:
Great, yes, please make a pull & include updating a test to check the case, thanks.

closed 2025-02-15 by wsnyder