verilator/verilator #5226: `--Mdir` does not play well with user c/cpp files

FanShupei, 2024-07-04:
* Issue explained

It's best explained by running the following example. The command and error message is given below. The key point is that user c file `test.c` and a "complicated" Mdir option co-exists. `--Mdir obj_dir` or `--Mdir obj_dir_other` is OK, but `--Mdir build/obj_dir` is not OK.

```
$ verilator --cc --build test.sv dpi.c -Mdir build/obj_dir
make: Entering directory '/home/fsp/my/test/build/obj_dir'
/usr/sbin/python3 /usr/share/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include Vtest.cpp Vtest___024root__DepSet_h9b2d61ff__0.cpp Vtest___024root__DepSet_hc07518e4__0.cpp Vtest___024root__Slow.cpp Vtest___024root__DepSet_hc07518e4__0__Slow.cpp Vtest__Syms.cpp > Vtest__ALL.cpp
g++ -Os  -I.  -MMD -I/usr/share/verilator/include -I/usr/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-overloaded-virtual -Wno-shadow -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable      -c -o Vtest__ALL.o Vtest__ALL.cpp
make: *** No rule to make target 'dpi.c', needed by 'dpi.o'.  Stop.
make: Leaving directory '/home/fsp/my/test/build/obj_dir'
%Error: make -C build/obj_dir -f Vtest.mk -j 1 exited with 2
%Error: Command Failed ulimit -s unlimited 2>/dev/null; exec /usr/bin/verilator_bin --cc --build test.sv dpi.c -Mdir build/obj_dir
```

This is a Makefile issue, the contents of the two files are irrelevant. A minimal `test.sv` like below and an empty `dpi.c` is sufficient

```sv
module top;
    initial $display("hello");
endmodule
```

* Investigation

It seems Verilator assumes `--Mdir` must specify a direct child of current directory. `VPATH=..` is hardcoded in `verilated.mk`. When it is not the case, e.g. given "--Mdir build/obj_dir", Makefile does not know where to find user c/cpp files.

https://github.com/verilator/verilator/blob/dace2c0a9a8f9d0559c4e4cbdb56ebdcae93b097/include/verilated.mk.in#L94

* Verilator version

`Verilator 5.026 2024-06-15 rev UNKNOWN.REV` on ArchLinux, reproduced on master branch.

* Proposed fix

We may remove the line `VPATH=..` in `verilated.mk`. Instead, we can record the absolute path of working directory of verilator in `Vprefix.mk`. As an example,

```Makefile
# contents of Vtest.mk
...
# Include list of all generated classes
include Vtest_classes.mk
# Include global rules
include $(VERILATOR_ROOT)/include/verilated.mk
VPATH += /home/fsp/my/test # absolute path of where verilator runs
...
```

Any comments on this strategy?

wsnyder, 2024-07-04:
Appreciate the debugging. 

Absolute paths break compile caching when __FILE__ is used, so I prefer relative paths on the command line remain relative.  We could instead record the initial path, and convert it to a relative path from the build_directory.  (e.g. it would be computed to become ../.. in your example.)

Might you make a pull with a test to do that?

FanShupei, 2024-07-05:
>  We could instead record the initial path, and convert it to a relative path from the build_directory.

That sounds reasonable.

> Might you make a pull with a test to do that?

Yes, I plan to work on a PR later.

wsnyder, 2024-09-26:
Fix in master towards 5.030.

closed 2024-09-26 by wsnyder

ethanuppal, 2025-01-12:
Could this version be added to the homebrew formula?

wsnyder, 2025-01-12:
You need to ask the homebrew maintainers.