verilator/verilator #5228: Rasberry Pi compilation problems

fbonardi, 2024-07-05:
I was trying to compile the latest version on a Rasberry Pi 4 since the packaged version is too old
I followed the Detailed Build Instructions section and the compilation step complete correctly but when I ran 

make test 

It failed with:

echo "" > Vt_a1_first_cc__ALL.verilator_deplist.tmp
g++    Vt_a1_first_cc__main.o verilated.o verilated_vcd_c.o verilated_threads.o Vt_a1_first_cc__ALL.a    -pthread -lpthread -latomic   -o Vt_a1_first_cc
rm Vt_a1_first_cc__ALL.verilator_deplist.tmp
driver: Leaving directory '/home/cino/verilator/test_regress/obj_vlt/t_a1_first_cc'
	obj_vlt/t_a1_first_cc/Vt_a1_first_cc    > obj_vlt/t_a1_first_cc/vlt_sim.log
Vt_a1_first_cc: /home/cino/verilator/test_regress/../include/verilated_imp.h:97: VerilatedEvalMsgQueue::VerilatedEvalMsgQueue(): Assertion `atomic_is_lock_free(&m_depth)' failed.
%Warning: vlt/t_a1_first_cc: Exec of obj_vlt/t_a1_first_cc/Vt_a1_first_cc failed: Vt_a1_first_cc: /home/cino/verilator/test_regress/../include/verilated_imp.h:97: VerilatedEvalMsgQueue::VerilatedEvalMsgQueue(): Assertion `atomic_is_lock_free(&m_depth)' failed.

vlt/t_a1_first_cc: %Error: Exec of obj_vlt/t_a1_first_cc/Vt_a1_first_cc failed: Vt_a1_first_cc: /home/cino/verilator/test_regress/../include/verilated_imp.h:97: VerilatedEvalMsgQueue::VerilatedEvalMsgQueue(): Assertion `atomic_is_lock_free(&m_depth)' failed.
vlt/t_a1_first_cc: FAILED: Exec of obj_vlt/t_a1_first_cc/Vt_a1_first_cc failed: Vt_a1_first_cc: /home/cino/verilator/test_regress/../include/verilated_imp.h:97: VerilatedEvalMsgQueue::VerilatedEvalMsgQueue(): Assertion `atomic_is_lock_free(&m_depth)' failed.
==SUMMARY: Passed 0  Failed 1  Time 0:27
==SUMMARY: Passed 0  Failed 1  Time 0:27

Googleing around the error msg I found something related to ARM architecture and they suggest to pass 
-mcpu=cortex-a53
to the compiler. I tried to pass that to the configure step using CXXFLAGS=-mcpu=cortex-a53 but it didn't seems to change anything

Any idea?

Thx

wsnyder, 2024-07-05:
A Google on the error shows: https://stackoverflow.com/questions/64242067/stdatomicbool-lock-free-inconsistency-on-arm-raspberry-pi-3

which I presume is why you tried -mcpu=cortex-a53

I suggest looking for other similar sources, I don't know the fix myself.

fbonardi, 2024-07-05:
Good finding that did the trick: 
without -CFLAGS '-mcpu=cortex-a53'
verilator  --cc --exe --build -j 2 --main -Wno-lint  --trace-fst --trace-structs --timing -f flist.f
I get
Vset_data_addr: /usr/local/share/verilator/include/verilated_imp.h:97: VerilatedEvalMsgQueue::VerilatedEvalMsgQueue(): Assertion `atomic_is_lock_free(&m_depth)' failed.
Aborted

While if I add -mcpu=cortex-a53 or -march=armv8-a either way
verilator -CFLAGS '-mcpu=cortex-a53' --cc --exe --build -j 2 --main -Wno-lint  --trace-fst --trace-structs --timing -f flist.f
It worked as expected:
- example.sv:42: Verilog $finish
- S i m u l a t i o n   R e p o r t: Verilator 5.027 devel
- Verilator: $finish at 150ps; walltime 0.002 s; speed 0.000 s/s
- Verilator: cpu 0.000 s on 1 threads; alloced 157 MB

Thx a lot

closed 2024-07-05 by wsnyder